crankyoldgit/IRremoteESP8266 #2: Implement receiving and receiving sample

probonopd, 2015-06-27:
On the main page the description says "send and receive infrared signals" but I can only find examples for sending, which works great. Please also implement receiving.

markszabo, 2015-06-28:
Yes, the receiver part was not implemented yet. I edited the description for now, and I will implement it later

probonopd, 2015-06-28:
Cool. Looking forward to it. Do you think it will be hard to implement, and where do you see the challenges? 
Would you be interested in some more complex example sketches, like supporting sending Pronto Hex codes which are popular on the web? (I am working on it.)

markszabo, 2015-06-28:
Well I have no experience with timers and interrupts on ESP8266, so that will be a pretty big challenge for me :D Also someone else started to implement it, and got issues with the receiving part: http://www.esp8266.com/viewtopic.php?f=33&t=3357#p19122 So it probably won't be easy

probonopd, 2015-06-28:
Yes, PaulDG's comment reads like he got receiving working by using https://sui77.wordpress.com/2011/05/27/use-any-arduino-pin-for-transmission-with-the-irremote-libary-by-soft-pwm/ plus some changes he describes on the page you linked. Did you try that?

markszabo, 2015-06-28:
Not yet, I don't have too much time right now. And originally I only needed the sending part (I recorded the codes with Arduino previously), that's why I did not implemented it back then.

probonopd, 2015-07-15:
This works for me, it gives the raw timings. From there it could be hooked into the existing decoding algorithms.

```
#define receiver_pin 2 
#define max_pulses 700

unsigned int raw_array[max_pulses];
volatile bool is_on; // volatile = set by ISR
volatile unsigned long this_change_micros; // volatile = set by ISR
bool was_on;
unsigned long this_loop_micros, previous_loop_micros, previous_change_micros = 0;
unsigned int i_pulses, i = 0;

// Keep isr() as short as possible
void isr() { 
  this_change_micros = micros();
  is_on = digitalRead(2);
}

void setup() {
  Serial.begin(115200);
  pinMode(receiver_pin, INPUT);
  was_on = digitalRead(receiver_pin);
  is_on = was_on = previous_loop_micros = 0;
  previous_change_micros = this_change_micros = micros();
  attachInterrupt(receiver_pin, isr, CHANGE);
}

void loop() {
  this_loop_micros = micros();
  if (was_on != is_on && i_pulses < max_pulses) {
    raw_array[i_pulses++] = (this_change_micros - previous_change_micros);
    previous_change_micros = this_change_micros;
    previous_loop_micros = this_loop_micros;
    was_on = is_on;
  }
  if (is_on && i_pulses > 0 && (i_pulses == max_pulses ||  (this_loop_micros - previous_loop_micros) > 100000)) { // If we have received the maximum pulses or its 100ms since last one
    for (i = 1; i < i_pulses; i++) {
      if (raw_array[i - 1] > 10000) // Assuming that everything above 10ms indicates a repeat
        Serial.println("");
      if (i % 2 == 0)
        Serial.print("-");
      else
        Serial.print("+");
      Serial.print(raw_array[i]);
      Serial.print(" ");
    }
    Serial.println("-99999"); // Otherwise IrScrutinizer does not decode properly
    i_pulses = 0;
  }
}
```

markszabo, 2015-07-16:
Thanks for the code. I tried it yesterday, but it didn't worked for me. It seemed as loop() would stop after a couple of minute and I didn't have time to debug it yet. But the interrupt part certainly works, so I  will try to rewrite the receiving part of the library using it.

samehhady, 2015-09-16:
Guys anything new regarding the receive part?

ildr3him, 2015-09-16:
Same here. Any progress on enabling the receiver part? Would be amazing!

ildr3him, 2015-09-16:
So I tried @probonopd sketch. instead of GPIO2 I used GPI14 on an ESP-12.

It is working now but everything i get doesn't look like distinct IR codes form my remote:

>+4384 -630 +491 -629 +491 -629 +494 -626 +491 -628 +493 -629 +491 -628 +493 -627 +492 -628 +1610 -627 +1610 -627 +1606 -628 +1611 -627 +1607 -629 +1607 -628 +1610 -627 +1609 -626 +495 -626 +492 -627 +1609 -628 +1608 -628 +492 -628 +496 -623 +496 -624 +494 -628 +1610 -626 +1610 -626 +492 -628 +493 -627 +1611 -624 +1612 -626 +1608 -629 +1610 -625 +38854 
-9044 +2152 -629 -99999

>+9016 -4389 +624 -497 +624 -496 +623 -497 +623 -497 +622 -498 +624 -495 +624 -497 +623 -499 +621 -1611 +625 -1613 +622 -1613 +623 -1613 +623 -1613 +623 -1614 +624 -1611 +625 -1612 +624 -495 +624 -496 +625 -495 +625 -1612 +623 -1613 +625 -496 +624 -496 +622 -495 +626 -1613 +624 -1611 +625 -1612 +624 -498 +623 -496 +625 -1614 +623 -1610 +625 -1613 +623 -99999

>+9041 -4384 +630 -493 +626 -494 +626 -491 +629 -490 +630 -491 +627 -493 +627 -493 +627 -492 +628 -1607 +629 -1609 +627 -1606 +628 -1610 +625 -1609 +628 -1606 +627 -1608 +630 -1607 +627 -493 +627 -1608 +628 -1608 +629 -1606 +628 -1608 +627 -493 +628 -1607 +628 -491 +628 -1607 +628 -492 +627 -493 +627 -494 +625 -493 +626 -1609 +629 -491 +628 -1607 +627 -38854 
+9042 -2151 +629 -99999

themindfactory, 2015-09-16:
If your expecting the same EXACT bunch of #'s to appear then this will not happen, I do not know the coding of your remote, but all these #'s do look good, they are all +- a few counts from one another, you need to feed this data into a decoder and you will I think get real keypresses out!

ildr3him, 2015-09-17:
Alright. This helps me out a lot since I know now where to go. The problem is, as of now, there doesn't seem to be any compatible IR decoder for the ESP8266 as they all rely on the AVR/io.h and AVR/interrupt.h. And this is exactly the same problem why IRremote.h doesn't work in the first place.

So, I'm receiving IR signals but  sadly I can't do anything with them on the ESP8266.

Isn't there a way to get the io.h and interrupt.h working on the ESP8266?

probonopd, 2015-09-18:
@ildr3him the fact that you got these codes shows that they were received properly without the need for AVR/io.h or AVR/interrupt.h. It is not simply a matter of feeding these timings into the decoding software (which should not need AVR/io.h or AVR/interrupt.h) in order to get it decoded. 

However, note that markszabo mentioned the receiving code stops working after a couple of minutes which could be an indication for a leak somewhere. So I haven't given up hope yet for markszabo to find some time and implement the receiving :-)

sebastienwarin, 2015-09-23:
Hi everyone !
I'm happy to announce that I have just finished to implement the IR decoder on ESP8266 Arduino ! It works perfectly with my Panasonic & Samsung remote controls !
I need to clean my code & finish to test the IR sending and I'll publish the code this week !
For now, I must go to sleep , it's late in France ;)
[ir decode on esp](https://cloud.githubusercontent.com/assets/6674275/10035221/16c2933a-6198-11e5-8964-1b2157fbf57f.PNG)

samehhady, 2015-09-23:
@sebastienwarin these are great news!! I can't wait to try it out :)

sebastienwarin, 2015-09-23:
I just publish my library with the receiving part : https://github.com/sebastienwarin/IRremoteESP8266
Have fun ;)

samehhady, 2015-09-23:
@sebastienwarin you are them man! :D

markszabo, 2015-09-23:
Sorry guys, I didn't have the time to contribute to this, but great job, thank you :+1: 
I'm new to github, so is there any way to merge your work to my repo with giving you the credits? I think you should submit a pull request, but I'm not sure

closed 2016-07-08 by markszabo