crankyoldgit/IRremoteESP8266 #5: LG code port problem

direk, 2015-09-14:
Hi

Library is great, but I have strange problem. I ported LG code send function from here: https://github.com/z3t0/Arduino-IRremote/blob/master/ir_LG.cpp
but it is not working. I am sending ir codes to my LG airconditioner from ESP8266. I also have second device based on arduino as ir-receiver for troubleshooting. Both codes, sent from ESP8266 and from oryginal ir remote looks identically in arduino code grabber, but remote is working fine and codes from ESP are being ignored. What am I doing wrong? Maybe I ported code in a wrong way? I just copied function code and declarations...

markszabo, 2015-09-14:
Well can you send the code from the Arduino? I think, maybe your LED has a slightly different frequency then the receiver of the airconditioner

direk, 2015-09-14:
The same LED and the same code on oryginal arduino board and oryginal IR library works fine. I have no idea what is wrong.

direk, 2015-09-15:
One more thing - when I am sending NEC codes to my TV from ESP it works fine, so it looks like code need some tweaks after porting, or aircon is very sensitive for every disruption (length of signal for zeros and ones).

chaeplin, 2015-09-16:
@direk Thank you for your tip.

This module works with my esp8266
http://www.dfrobot.com/wiki/index.php?title=DIGITAL_IR_Transmitter_Module_(SKU:DFR0095)

gorec2005, 2015-10-16:
Hi!
this code work for me:
in IRremoteESP8266.cpp
...
void IRsend::sendNEC(unsigned long data, int nbits)
{
  enableIROut(38);
  mark(NEC_HDR_MARK);
  space(NEC_HDR_SPACE);
  for (int i = 0; i < nbits; i++) {
    if (data & TOPBIT) {
      mark(NEC_BIT_MARK);
      space(NEC_ONE_SPACE);
    }.
    else {
      mark(NEC_BIT_MARK);
      space(NEC_ZERO_SPACE);
    }
    data <<= 1;
  }
  mark(NEC_BIT_MARK);
  space(0);
}

void  IRsend::sendLG (unsigned long data,  int nbits)
{
    enableIROut(38);
    mark(LG_HDR_MARK);
    space(LG_HDR_SPACE);
    mark(LG_BIT_MARK);
    for (unsigned long  mask = 1UL << (nbits - 1);  mask;  mask >>= 1) {
        if (data & mask) {
            space(LG_ONE_SPACE);
            mark(LG_BIT_MARK);
        } else {
            space(LG_ZERO_SPACE);
            mark(LG_BIT_MARK);
        }
    }
    space(0);
}

void IRsend::sendWhynter(unsigned long data, int nbits) {
...

and off course in IRremoteESP8266.h

  void sendNEC(unsigned long data, int nbits);
  void sendLG(unsigned long data, int nbits);
  void sendSony(unsigned long data, int nbits);

markszabo, 2015-10-17:
Ok, thanks, I added your function. Next time you may issue a pull request, which would make my work easier and give you proper credits (but I still wrote your name in the commit message).

closed 2016-07-08 by markszabo