crankyoldgit/IRremoteESP8266 #7: Pinout

Atzingen, 2015-10-13:
Can you tell me if the pin conection is GPIO0 ( irsend(0) ) to GND or to 3.3V ? This two possibilits would invert the led status/
In Arduino is generaly is positive GPIO to GND but because the esp8266 can give low current most blink examples use gpio to 3.3V

markszabo, 2015-10-13:
To GND by default

mocheffendi, 2016-01-11:
When we put LED IR to GPIO0 ESP 8266 ESP 12E always at programming mode, can I change to another GPIO ?

markszabo, 2016-01-12:
Sure, just use eg. irsend(1) instead of irsend(0) and connect your ir led to GPIO1

xCite1986, 2016-01-25:
if connect the gpio0 to the led the esp would go in flashmode, change to gpio2 the blue light on the esp-01 stay bright (uart-mode?) 

someone a simple idea how to solve that problem??

http://www.esp8266.com/viewtopic.php?f=13&t=1730

mocheffendi, 2016-01-25:
Use irsend(5) to use gpio05

xCite1986, 2016-01-25:
on an ESP8266-ESP01??
there are only 2 gpio pins, or am I wrong?

mocheffendi, 2016-01-26:
Sorry 
I think, you need to use esp12E, there is more pin to use.
Gpio00 and gpio02 can't be used for ir send.

Atzingen, 2016-01-26:
If you are not using Serial, there are two more pins available (rx and tx) in esp8266-01 that can be used like regular gpio.

xCite1986, 2016-01-26:
thanks for the hint :)

i would try this solution first
[image]

if this won't work i'll try to get out another gpio directly from the chip.
[image]

markszabo, 2016-01-26:
I think you may be able to use GPIO0 and GPIO2 for irsend, if you connect the led in reverse order, and change the library a bit.
1.) Connect the led to any pin and the other leg of the led to Vcc (instead of normal GND). Use resistor as usual.
2.) Open IRremoteESP8266.cpp in text editor and go to line 305, to function IRsend::mark(int time). You need to change the following lines:
310.  digitalWrite(IRpin, HIGH);
312.  digitalWrite(IRpin, LOW);
321.  digitalWrite(IRpin, LOW);
Change these lines to the opposition (LOW to HIGH and HIGH to LOW) like this:
310.  digitalWrite(IRpin, LOW);
312.  digitalWrite(IRpin, HIGH);
321.  digitalWrite(IRpin, HIGH);
I think this should work, since now the led is by default connected to Vcc, so ESP8266 can boot normally. Let me know if it works, and if does, I may write an article on Wiki, because this could help others too.

xCite1986, 2016-01-26:
hey mark.

yes, now they boots up perfectly... but the ir-led is always "bright"

markszabo, 2016-01-26:
hm could you try this simple sketch:
void setup() {
  pinMode(0, OUTPUT);
  digitalWrite(0, HIGH); 
}
void loop() {
  delay(1000);
}

Change 0 to 2 if you are using that pin. This should leave your led turned off. Now change from digitalWrite(0, HIGH)  to digitalWrite(0,LOW), and run it. It should turn on the led and keep it that way.
Is it doing this? You can use a normal led instead of the IR led to test it. The important thing is, to use the previous setup: connect the led to the pin and to Vcc.

xCite1986, 2016-01-26:
void setup() {
pinMode(2, OUTPUT);
digitalWrite(2, HIGH); 
}
void loop() {
delay(1000);
}

= dark LED

void setup() {
pinMode(2, OUTPUT);
digitalWrite(2, LOW); 
}
void loop() {
delay(1000);
}

= bright LED

VCC connected to LED+, GPIO2 on LED-

markszabo, 2016-01-26:
This is exactly the expected behavior. Now if you edited the library correctly, it should set the pin HIGH in the end of the transmission, thus it should be dark. 

Are you sure you are editing the code of the library which is used by the arduino IDE? Try to make some obvious mistakes, like write a non-existing command in the library, save it, and then try to compile some code in the IDE using the library. It should throw an error message. 

Maybe you have to restart the IDE to recognize the changed library, but for me I don't have to do that.

xCite1986, 2016-01-26:
set ...

-----
void IRsend::mark(int time) {
  // Sends an IR mark for the specified number of microseconds.
  // The mark output is modulated at the PWM frequency.
  long beginning = micros();
  while(micros() - beginning < time){
    digitalWrite(IRpin, LOW);
    delayMicroseconds(halfPeriodicTime);
    digitalWrite(IRpin, HIGH);
    delayMicroseconds(halfPeriodicTime); //38 kHz -> T = 26.31 microsec (periodic time), half of it is 13
  }
}

-----

...to ...
-----

void IRsend::mark(int time) {
  // Sends an IR mark for the specified number of microseconds.
  // The mark output is modulated at the PWM frequency.
  long beginning = micros();
  while(micros() - beginning < time){
    digitalWrite(IRpin, xxx);
    delayMicroseconds(halfPeriodicTime);
    digitalWrite(IRpin, xxxx);
    delayMicroseconds(halfPeriodicTime); //38 kHz -> T = 26.31 microsec (periodic time), half of it is 13
  }
}

-----

...and save the file gets errors in compiling 
-----

C:\Program Files (x86)\Arduino\libraries\IRremoteESP8266\IRremoteESP8266.cpp: In member function 'void IRsend::mark(int)':
C:\Program Files (x86)\Arduino\libraries\IRremoteESP8266\IRremoteESP8266.cpp:310:25: error: 'xxx' was not declared in this scope
     digitalWrite(IRpin, xxx);
                         ^
C:\Program Files (x86)\Arduino\libraries\IRremoteESP8266\IRremoteESP8266.cpp:312:25: error: 'xxxx' was not declared in this scope
     digitalWrite(IRpin, xxxx);
                         ^
Fehler beim Kompilieren.

-----

so the IDE works correctly. 

But hey.. sending IR codes still works.

if the code/library to be used in battery-powered devices, it would be better the LED would be off while "standbying"

markszabo, 2016-01-26:
Well, if sending IR codes works, then you can simply turn of the led after sending with digitalWrite(2, HIGH).

Btw are you sure, that you also changed the library in the 321. line? It's in the function IRsend::space(int time) and it should be digitalWrite(IRpin, HIGH) for you (the default is LOW).

xCite1986, 2016-01-26:
okay, i will try :+1: 

thanks for help and quick response

yes, liine 321 was also HIGH

[image]

rajasekarsarangapani, 2016-04-02:
good day all, 
i like to say thank you for this library creator and supporting members,
its decoding perfectly but ESP8266 Restarting, If  receive continuous  IR Signal ,

iam using samsung TV remote
[image]
[image]

rajasekarsarangapani, 2016-04-02:
Iam using gpio13 (TSOP1738)

closed 2016-07-08 by markszabo

bragasil, 2018-01-26:
you can use esp-01 without problems (pin 2 = irsend(2)). just change irinverter to true (irsend.cpp) and add the lines below in setup (.ino) and use vcc default for IRemiter.
void setup() {
pinMode(2, OUTPUT);
digitalWrite(2, HIGH);
}

crankyoldgit, 2018-01-27:
@bragasil Do you have a circuit diagram for your ESP-01 circuit that we can have so we can show ESP-01 users how to do this?

crankyoldgit, 2018-01-28:
If you use the code in the _master_ branch (or any release **after** _v2.3.2_)
You can do what @bragasil suggested by using the library as normal, but use/change _your_ code to:
```
IRsend irsend(2, true);  // Use gpio 2 and invert the signal. i.e. Normally HIGH.
//  See https://github.com/markszabo/IRremoteESP8266/blob/master/src/IRsend.cpp#L27 for more details.
```

bragasil, 2018-01-28:
remembering that at the time of uploading the firmware the led should not be connected. Another issue is that if the infrared led is low power, it is not necessary to use a drive or even a resistor.