crankyoldgit/IRremoteESP8266 #8: Samsung codes

DiegoDVG, 2015-10-18:
Nice job, congratulations!!!

My question is if I can send samsun codes an knws the encoding, because the ir recv prints this:

E0E0D02F (32 bits)
Raw (67): 4500 -600 1700 -600 1700 -600 1700 -600 550 -600 550 -600 550 -600 600 -600 600 -600 1650 -650 1700 -600 1700 -600 600 -600 550 -600 550 -650 500 -650 600 -600 1700 -600 1650 -650 600 -600 1700 -600 600 -600 600 -600 600 -600 600 -600 550 -600 550 -600 1650 -650 600 -600 1700 -600 1650 -650 1650 -650 1650 -650 

and decoded type is 11

markszabo, 2015-10-18:
Decode type 11 means Samsung (as in IRremoteESP8266.h), so if you want to send the code above, you can use something like this: 

#include <IRremoteESP8266.h>
IRsend irsend(0); //an IR led is connected to GPIO pin 0
void setup()
{
  irsend.begin();
}
void loop() {
  irsend.sendSAMSUNG(0xE0E0D02F, 32); // the decoded Samsung code, length of code (32 bits in this case)
  delay(2000);
}

I don't have an ESP8266 nearby, so I can not test it now, but it should work this way. If it doesn't, please try to receive what it sends with an Arduino/other ESP8266, and post the results. 

Usually if it doesn't work it mainly because the TV/device require more IR power. That case getting an other led, or using some transistor (like here: https://github.com/markszabo/IRremoteESP8266/issues/5) may help.

DiegoDVG, 2015-10-18:
perfect!

you know a databse o ir code like LIRC

in mysql, because i can integrate the data

closed 2016-07-08 by markszabo