crankyoldgit/IRremoteESP8266 #11: RC5 possible issue

axwell, 2016-01-04:
Hello,
i have attached an IR LED (http://goo.gl/m9HZkR) to my ESP module.
I am tring to control my tv from internet sending RC5 signals. (read RC5 and codes with an receiver)
TV accepts the IR signal only after 10-15 tries. Read somewhere that RC5 is very sensitive to timing.

Do you have any idea why ?

Thanks.

markszabo, 2016-01-05:
Hello,

can you try it with a normal arduino and the original library?
 
You may also try to change the constants RC5_T1 and RC5_RPT_LENGTH in IRremoteInt.h to get better results. I would start with decreasing their value a bit, but I'm not 100% sure.

Mark

axwell, 2016-01-05:
Tested with arduino uno and original library works fine.

markszabo, 2016-01-12:
Wow, that's kind a new for me, and no, sorry, I have no idea why it has two codes. Does it work if you send both codes after each other?

renno-bih, 2016-02-19:
That is how RC5 works. You need to put toggle bit into code to send it...
That is how I implemented it:
```
else if (type_json == RC5 || type_json == RC6) {  //Implementation of sending RC5/RC6 codes

    // Flip the toggle bit for a new button press
    if (repeat) {
      toggle = 1 - toggle;
    }

    // Put the toggle bit into the code to send
    codeValue_json = codeValue_json & ~(1 << (bits_json - 1));
    codeValue_json = codeValue_json | (toggle << (bits_json - 1));
    if (type_json == RC5) {
#if DEBUG
      Serial.print("Sent RC5 ");
      Serial.println(codeValue_json, HEX);
#endif
      irsend.sendRC5(codeValue_json, bits_json);
    }
    else {
      irsend.sendRC6(codeValue_json, bits_json);
#if DEBUG
      Serial.print("Sent RC6 ");
      Serial.println(codeValue_json, HEX);
#endif
    }
  }
```
I also store last code value to check if it is the same as new value. If it is I set repeat to 1.

markszabo, 2016-02-22:
Wow nice. If you think it's better than the actual one in the library, you are more than welcome to issue a pull request :)

renno-bih, 2016-02-23:
This is something that is implemented in executing sketch, not in library. 
In official library there is also example that shows how to implement it. Here it is:
[https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecord/IRrecord.ino](url)

closed 2016-07-08 by markszabo