crankyoldgit/IRremoteESP8266 #13: Protocol with more than 32bit?

gorec2005, 2016-02-13:
Hi!
I want user protocol like this:
`Encoding  : UNKNOWN
Code      : ED1215A9 (32 bits)
Timing[99]: 
     +3050, -9000     + 600, -1500     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, -1550     + 550, - 550
     + 550, - 550     + 600, - 500     + 550, - 550     + 550, -1550
     + 550, -1550     + 550, -1550     + 550, -1550     + 550, -1550
     + 550, -1550     + 550, - 550     + 550, - 550     + 550, - 550
     + 600, - 500     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 500
     + 600, - 500     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, -1550     + 600, - 500
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, - 550     + 550, - 550     + 550, - 550     + 550, - 550
     + 550, - 550     + 550
unsigned int  rawData[99] = {3050,9000, 600,1500, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,1550, 550,550, 550,550, 600,500, 550,550, 550,1550, 550,1550, 550,1550, 550,1550, 550,1550, 550,1550, 550,550, 550,550, 550,550, 600,500, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,500, 600,500, 550,550, 550,550, 550,550, 550,550, 550,550, 550,1550, 600,500, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550,550, 550};  // UNKNOWN ED1215A9
`
This is air conditioner samsung... - how correct do this?

gorec2005, 2016-02-13:
48 bit long...

markszabo, 2016-02-22:
Hello,
sorry for the late answer. I don't know if that is possible with the actual state of the library. I'm actually thinking about changing the whole code structure to be similar to the newest version of the Arduino IR library (https://github.com/z3t0/Arduino-IRremote). That way it would have all the different protocols in separate file thus making it easier to add new protocols. I will try to do it during the next weeks, but I can't promise anything for sure.
Mark

gorec2005, 2016-03-25:
Hi!
this project have realized many long IR packet type devices.... https://github.com/ToniA/arduino-heatpumpir

gorec2005, 2016-06-06:
Hi Mark!
do you have any news about long-block protocols ?

markszabo, 2016-06-08:
Sorry I don't have time to work on this project right now :/

crankyoldgit, 2017-03-03:
Hey, there are a few protocols we now support with more than 32bits.
e.g. Daikin, Kelvinator, MitsubishAC, & Panasonic.
None, however, use an 'unsigned long long' yet.
Just beware, that some AC remote & units have complex transmission packets. I say that, as the decode you got, may only be a small portion of what is being sent. Due to their nature, the decode routines often report only a fragment of these longer codes as there are significant breaks in them.

With a 54 bit protocol as you seem to indicate, you could code it as either a procedure with a single unsigned long long data parameter(<= 64bit), or similar to the Panasonic implementation with two 32bit parameters (32 + 32 = 64 bits).

The other A/Cs have chosen to do this as a char/byte array as they are much longer than 64 bits.

Your fall back (The lazy/easy way) is to use GC or the Raw methods to send arbitrary sized codes.

gorec2005, 2017-04-17:
ok... so close issue...

closed 2017-04-17 by gorec2005