crankyoldgit/IRremoteESP8266 #14: ESP resets if delay is used...

renno-bih, 2016-02-19:
When I press and hold button on my remote for about 10 seconds my ESP resets. It only happens when I use some delay in loop. Even example sketch IRrecvDumo contains 100ms delay in loop resulting ESP to reset.
Any ideas to identify source of problem?

markszabo, 2016-02-22:
If I understand correctly you are holding the button down, which means constantly sending codes. I think the ESP just can not keep up the pace receiving all of them, and that is the reason of the reset.
Also I believe that there isn't too many use-cases where you want to hold down the button on the remote for more then 1-2 seconds, and even then, you can simply write your code without delay. So in my opinion this is not something where we should put much effort to investigate the real cause. (Although it is kinda strange behavior. Have you tried it with really small delay, let's say 1 ms?)

renno-bih, 2016-02-23:
Instead of using delay to block some parts of my code (for example for LED to blink when receiving code) I used elapsedMillis library. Now it works.
btw, it works if delay is under 40ms.

roidayan, 2017-03-02:
issue not reproducing with current code.

closed 2017-03-02 by roidayan