crankyoldgit/IRremoteESP8266 #17: irsend.sendNEC problem

goobering, 2016-03-09:
I had some problems getting IRServer to send the right codes to my Toshiba Regza TV. I captured all of the remote codes from the manufacturer's remote and plugged them into the IRServer demo. No dice - definitely LED activity, everything was firing fine, just no reaction from the TV. Working my way through troubleshooting, I captured a set of codes from the infrared LED attached to my ES-01 and compared them to the manufacturer's remote:

Manufacturer remote code: 
9050,4500, 650,600, 600,600, 600,600, 600,600, 600,600, 600,600, 600,1700, 600,600, 600,1700, 600,1700, 600,1700, 600,1650, 600,1650, 600,1650, 600,600, 600,1650, 650,600, 600,1700, 600,600, 600,600, 600,1650, 600,600, 600,600, 600,600, 600,1700, 600,600, 600,1700, 600,1700, 600,600, 600,1650, 600,1650, 600,1650, 600

Reproduced code:
9000,4550, 600,600, 600,600, 600,600, 600,600, 600,600, 600,600, 600,1700, 600,600, 600,1700, 600,1700, 600,1700, 600,1700, 600,1700, 600,1700, 600,600, 600,1700, 600,600, 600,1700, 600,600, 600,600, 600,1700, 600,600, 600,600, 600,600, 600,1700, 600,600, 600,1700, 600,1700, 600,600, 600,1700, 600,1700, 600,1700, 600,**_600, 600,600, 600,600, 600,600, 600_**

The additional eight pulses at the end caused problems. Comparing your example code to Shirrif's example code: 

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino#L30]
`irsend.sendNEC(code, 36);`

https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRtest/IRtest.ino#L158
`testNEC(0x00000000, 32);`

I amended 36 to 32, which removed the spare pulses and got everything working.

markszabo, 2016-03-10:
Thank you for reporting the issue, I fixed it :)

closed 2016-03-10 by markszabo

carlostorrez, 2017-04-07:
Felicidades estimado Markszabo, 
Quiza puedas ayudarme tengo un problema, estoy tratando de controlar un TV Samsung he decodificado correcatamente con el Sketch IrRecvDumpV2, Encoding  : SAMSUNG  Code      : E0E040BF (32 bits). Al tratar de usar este codigo con el Sketch IRServer, en la linea donde esta esta instruccion *irsend.sendNEC*, yo cambio por esta otra instruccion irsend.sendSAMSUNG, pero no obtengo resultados.
Finalmente Pruebo con el Sketch IRSendDemo con esta instruccion irsend.sendSAMSUNG(3772793023, 32); y funciona correctamente. Por lo que intuyo que el problema esta lineas arriba de esta instruccion, por lo que realizo otra prueba y los comento con //

void handleIr(){
  //for (uint8_t i=0; i<server.args(); i++){   **Comentado
    //if(server.argName(i) == "code")          **Comentado  
    //{                                                        **Comentado   
      //unsigned long code = server.arg(i).toInt();**Comentado
      irsend.sendSAMSUNG(0xE0E040BF, 32);                 **Aqui cambio *code* por el numero HEX
    //}                                                        **Comentado    
  }
  handleRoot();
}

Y funciona correctamente, pero por supuesto no diferencia el origen del click en el href.

Necesito tu ayuda muchas gracias.

Saludos

crankyoldgit, 2017-04-07:
Translating via google:

> Congratulations dear Markszabo,
Maybe you can help me I have a problem, I am trying to control a Samsung TV I have decoded correctly with the Sketch IrRecvDumpV2, Encoding: SAMSUNG Code: E0E040BF (32 bits). When trying to use this code with the Sketch IRServer, in the line where this instruction is irsend.sendNEC, I change for this other instruction irsend.sendSAMSUNG, but I do not get results.
Finally I try with the Sketch IRSendDemo with this instruction irsend.sendSAMSUNG (3772793023, 32); And works correctly. So I intuit that the problem is above this instruction, so I make another test and I comment with //
> ```
> Void handleIr () {
> // for (uint8_t i = 0; i <server.args (); i ++) {** Reviewed
> //if(server.argName(i) == "code") ** Reviewed
> // {** Reviewed
> // unsigned long code = server.arg (i) .toInt (); ** Reviewed
> Irsend.sendSAMSUNG (0xE0E040BF, 32); ** Here I change code by the number HEX
> //} ** Reviewed
> }
> HandleRoot ();
> }
> ```
> And it works fine, but of course it does not differentiate the source of the click on the href.
>
> I need your help thank you very much.
>
> regards

crankyoldgit, 2017-04-07:
@carlostorrez This appears not to be the same issue. Can you please create a new issue to track your request?
(Spanish via translate: Esto no parece ser el mismo problema. ¿Puede crear un nuevo número para realizar el seguimiento de su solicitud?)