crankyoldgit/IRremoteESP8266 #18: ES-01 pin usage

goobering, 2016-03-11:
I'm just finishing up my ES-01 build of an IRServer. The IRServer code currently uses GPIO 0 to flash the infrared LED. I found this inconvenient due to the pin being pulled low by the attached LED on every boot. This causes the ES-01 to go into flash mode, preventing normal running. I could get the thing to work but I had to disconnect the LED, reboot the ES-01, then reconnect the LED after boot. 

I managed to get everything working smoothly on an ES-01 using GPIO 3/RX as the LED pin. This works almost without modification, but it did require me to disable the RX pin in the serial.begin() function. 

https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRServer/IRServer.ino#L54
`Serial.begin()`

Revised version:
`Serial.begin(115200,SERIAL_8N1,SERIAL_TX_ONLY);`

This approach might be worth mentioning in a comment somewhere.

darshkpatel, 2016-03-25:
Else one can just use a pull up resistor. 

None the less your solution is a more elegant one.

bphermansson, 2016-10-20:
@goobering  But you have to disconnect the led when flashing new software?
@darshkpatel What value do you use for the pull up?

penfold42, 2016-10-31:
I would add another IR led in series (and change the resistor to suit)

They each have about 1.5v forward voltage and two of them gives you 3v which shouldn't read as a low.

closed 2017-07-14 by crankyoldgit