crankyoldgit/IRremoteESP8266 #21: RC-MM Protocol For IRremoteESP8266

darshkpatel, 2016-04-02:
I came across a remote which uses RC-MM protocol (probably also known as Nokia 24 ) , I needed help adding that to IRremoteESP8266 library. 

here is a sample for the same

This is the dump of the **same button **being pressed 10 times

``Encoding : UNKNOWN
Code : 49EF3073 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 300 + 200, - 550 + 150, - 600
+ 200, - 250 + 200, - 750 + 200, - 550 + 200, - 250
+ 200, - 250 + 150, - 600 + 200, - 550 + 200, - 450
+ 150, - 600 + 150, - 300 + 150, - 300 + 150, - 250
+ 200, - 250 + 150
unsigned int rawData[35] = {450,250, 150,300, 200,550, 150,600, 200,250, 200,750, 200,550, 200,250, 200,250, 150,600, 200,550, 200,450, 150,600, 150,300, 150,300, 150,250, 200,250, 150}; // UNKNOWN 49EF3073

Encoding : UNKNOWN
Code : 6E277A3 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 250 + 200, - 600 + 150, - 600
+ 200, - 250 + 150, - 800 + 150, - 600 + 150, - 300
+ 150, - 300 + 150, - 250 + 200, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 150, - 300 + 150, - 250
+ 200, - 250 + 200
unsigned int rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200}; // UNKNOWN 6E277A3

Encoding : UNKNOWN
Code : B70D49E4 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 300 + 150, - 600 + 150, - 600
+ 200, - 250 + 200, - 750 + 150, - 600 + 200, - 250
+ 200, - 250 + 150, - 600 + 200, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 150, - 300 + 150, - 250
+ 200, - 250 + 200
unsigned int rawData[35] = {450,250, 150,300, 150,600, 150,600, 200,250, 200,750, 150,600, 200,250, 200,250, 150,600, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200}; // UNKNOWN B70D49E4

Encoding : UNKNOWN
Code : 25D6E01D (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 250 + 200, - 600 + 150, - 600
+ 200, - 250 + 150, - 800 + 150, - 600 + 200, - 250
+ 150, - 300 + 150, - 250 + 200, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 150, - 300 + 150, - 300
+ 150, - 250 + 200
unsigned int rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 200,250, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,300, 150,250, 200}; // UNKNOWN 25D6E01D

Encoding : UNKNOWN
Code : 40033D4D (32 bits)
Timing[35]: 
+ 450, - 250 + 200, - 250 + 150, - 600 + 200, - 600
+ 150, - 300 + 150, - 750 + 200, - 600 + 150, - 250
+ 200, - 250 + 200, - 600 + 150, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 200, - 250 + 150, - 300
+ 150, - 250 + 200
unsigned int rawData[35] = {450,250, 200,250, 150,600, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,600, 150,600, 150,450, 150,600, 200,250, 200,250, 150,300, 150,250, 200}; // UNKNOWN 40033D4D

Encoding : UNKNOWN
Code : 6E277A3 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 250 + 200, - 600 + 150, - 600
+ 200, - 250 + 150, - 800 + 150, - 600 + 150, - 300
+ 150, - 300 + 150, - 250 + 200, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 150, - 300 + 150, - 250
+ 200, - 250 + 200
unsigned int rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,300, 150,250, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200}; // UNKNOWN 6E277A3

Encoding : UNKNOWN
Code : 9A2A2562 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 250 + 200, - 600 + 150, - 600
+ 200, - 250 + 150, - 800 + 150, - 600 + 150, - 300
+ 150, - 250 + 200, - 600 + 150, - 600 + 200, - 400
+ 200, - 600 + 150, - 300 + 150, - 250 + 200, - 250
+ 200, - 250 + 150
unsigned int rawData[35] = {450,250, 150,250, 200,600, 150,600, 200,250, 150,800, 150,600, 150,300, 150,250, 200,600, 150,600, 200,400, 200,600, 150,300, 150,250, 200,250, 200,250, 150}; // UNKNOWN 9A2A2562

Encoding : UNKNOWN
Code : DF6C1994 (32 bits)
Timing[35]: 
+ 450, - 250 + 200, - 250 + 150, - 600 + 200, - 600
+ 150, - 300 + 150, - 750 + 200, - 600 + 150, - 250
+ 200, - 250 + 200, - 250 + 150, - 600 + 200, - 400
+ 200, - 600 + 150, - 300 + 150, - 250 + 200, - 250
+ 200, - 250 + 150
unsigned int rawData[35] = {450,250, 200,250, 150,600, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,250, 150,600, 200,400, 200,600, 150,300, 150,250, 200,250, 200,250, 150}; // UNKNOWN DF6C1994

Encoding : UNKNOWN
Code : 2EA1533E (32 bits)
Timing[35]: 
+ 450, - 200 + 200, - 250 + 200, - 600 + 150, - 600
+ 150, - 300 + 150, - 750 + 200, - 600 + 150, - 300
+ 150, - 250 + 200, - 600 + 150, - 600 + 200, - 400
+ 200, - 600 + 150, - 250 + 200, - 250 + 200, - 250
+ 150, - 300 + 150
unsigned int rawData[35] = {450,200, 200,250, 200,600, 150,600, 150,300, 150,750, 200,600, 150,300, 150,250, 200,600, 150,600, 200,400, 200,600, 150,250, 200,250, 200,250, 150,300, 150}; // UNKNOWN 2EA1533E

Encoding : UNKNOWN
Code : 915E804D (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 300 + 150, - 600 + 200, - 550
+ 200, - 250 + 200, - 750 + 150, - 600 + 200, - 250
+ 200, - 250 + 150, - 300 + 150, - 600 + 200, - 400
+ 200, - 550 + 200, - 250 + 200, - 250 + 200, - 250
+ 150, - 300 + 150
unsigned int rawData[35] = {450,250, 150,300, 150,600, 200,550, 200,250, 200,750, 150,600, 200,250, 200,250, 150,300, 150,600, 200,400, 200,550, 200,250, 200,250, 200,250, 150,300, 150}; // UNKNOWN 915E804D

Encoding : UNKNOWN
Code : 7B255A09 (32 bits)
Timing[35]: 
+ 400, - 250 + 200, - 250 + 200, - 550 + 200, - 600
+ 150, - 300 + 150, - 750 + 200, - 600 + 150, - 250
+ 200, - 250 + 200, - 600 + 150, - 600 + 150, - 450
+ 200, - 550 + 200, - 250 + 200, - 250 + 150, - 300
+ 150, - 300 + 150
unsigned int rawData[35] = {400,250, 200,250, 200,550, 200,600, 150,300, 150,750, 200,600, 150,250, 200,250, 200,600, 150,600, 150,450, 200,550, 200,250, 200,250, 150,300, 150,300, 150}; // UNKNOWN 7B255A09

Encoding : UNKNOWN
Code : AD737632 (32 bits)
Timing[35]: 
+ 450, - 250 + 150, - 250 + 200, - 600 + 150, - 600
+ 200, - 250 + 200, - 750 + 150, - 600 + 200, - 250
+ 150, - 300 + 150, - 250 + 200, - 600 + 150, - 450
+ 150, - 600 + 200, - 250 + 200, - 250 + 150, - 300
+ 150, - 250 + 200`

crankyoldgit, 2017-03-18:
The protocol is documented here: http://www.sbprojects.com/knowledge/ir/rcmm.php

Seems fairly straight forward, yet a little odd compared to other common protocols.

crankyoldgit, 2017-03-18:
@darshkpatel Care to give https://github.com/crankyoldgit/IRremoteESP8266/tree/philips-rc-mm a try?

crankyoldgit, 2017-03-22:
Ping.

crankyoldgit, 2017-03-30:
Ping.

darshkpatel, 2017-03-30:
Sure, let me see

darshkpatel, 2017-03-30:
@crankyoldgit 
Sorry for the long day
I checked it on my bench and the lib 
Could identify the protocol , but still gives different codes for the same button

darshkpatel, 2017-03-30:
I enquired with a few colleagues , came to know it's not the protocol, but the wrong receiver for the frequency the IR transmitter is transmitting at. 

I suppose someone with the right receiver can try it out ? 
The 38KHz receiver doesn't work well

crankyoldgit, 2017-03-30:
Um, let me check/confirm. Are you trying to send RC-MM or receive it?

I only added sending btw. If you are getting different results from something for each send, can you share them please? It should produce the same signal when sending the same data., but seeing the different outputs might help me work out what's not working.

crankyoldgit, 2017-03-30:
Ah, yeah. 38kHz vs 36kHz might be an issue. Don't know. I've only got 38kHz ones myself. :-(

darshkpatel, 2017-03-30:
Same here, guess we'll have to wait for someone else to confirm

crankyoldgit, 2017-03-30:
I might add what I've got to the library as is, and label it untested/experimental etc. That way someone might be able to check it.

Or, you could probably order a 36kHz receiver module off ebay for a couple of dollars etc.

crankyoldgit, 2017-03-30:
Actually, can you describe what testing setup is? I didn't add a decode for rc-mm, just the sender.

> but still gives different codes for the same button
the 'unknown' code values should be taken with a grain of salt. It is a hash of the raw data, not an indication of the actual code signal. A minute change in a timing length will produce a very different 'unknown' code even if the underlying pattern is the same. The underlying pattern in your earlier examples however looks fine/the same. So, I can probably try to add a decode routine we might have some luck.

crankyoldgit, 2017-03-30:
Decoding one of your captures by hand:

> unsigned int rawData[35] = {450,250, 150,300, 150,600, 150,600, 200,250, 200,750, 150,600, 200,250, 200,250, 150,600, 200,600, 150,450, 150,600, 200,250, 150,300, 150,250, 200,250, 200}; 

HEADER + 00 + 10 + 10 + 00 + 11 + 10 + 00 + 00 + 10 + 10 + 01 + 10 + 00 + 00 + 00 + 00 + FOOTER
== B00101000111000001010011000000000 == 32 bits of data (which the protocol says should only be 12 or 24 bits???) == 0x28e0a600

crankyoldgit, 2017-03-30:
Hey @darshkpatel, care to try the latest update to my 'philips-rc-mm' branch?
Based on the data you provided, I think I now understand the RC-MM protocol better than the limited reference docs I could find. i.e. There is a footer mark that wasn't documented.

It also has my very first attempt at writing a decode routine. No promises etc. Completely untested. No warranties etc etc.

darshkpatel, 2017-03-31:
Sure, I'll check it as soon as I get home on 15th
 Apologies for the delay

tejipal, 2017-04-06:
I have got it working with 38khz tsop38338, tested this with stb remote with success, was looking for this for a long! thanks and kudos

tejipal, 2017-04-06:
I have tested the decoding (receiving), will confirm the send too tomorrow.

tejipal, 2017-04-06:
[image]

crankyoldgit, 2017-04-06:
Many thanks for that @tejipal 
However, I think I have a bug. Decoding by hand, that middle dump should decode to:
B00100110000000000010011100000010 (0x26002702)
yet my code is getting:
B00010101000000000001011000000001 (0x15001601)

I think it is not matching the spaces well, in that it is matching a lower 2-bit value when it should be matching a higher 2-bit value per mark/space pair.

Looking at the code MATCH code, the tolerances for it matching are too wide. We need to reduce the tolerances for this to accurately match values received.
e.g.
a space of 550us (which should equal to B10 (0x2) for which we expect has a space of 611us) has a lower tolerance of 611\*0.75 = ~458us and an upper of 611\*1.25 = ~764ms. This also matches the recorded space of 750us (which should be B11 (0x3) which has an expected value of 777us).

I'll need to work out a nice way to reduce the tolerances for this protocol and not for all the others for this to work correctly.

tejipal, 2017-04-07:
the tolerance thingy looks promising, I will test it again and get back with results. I also ordered 36khz decoder, will share results after testing.

tejipal, 2017-04-07:
hatts off to your quick remedy :)

crankyoldgit, 2017-04-07:
@tejipal I'm currently developing in my personal fork under this branch https://github.com/crankyoldgit/IRremoteESP8266/tree/rework-match

But beware, I haven't tested it at all yet. It's flat out compiling. ;-) i.e. Expect things to break.

crankyoldgit, 2017-04-23:
@tejipal & @darshkpatel Care to test the current _master_ branch? Should be better/working, I think. (famously wrong/last words.)

darshkpatel, 2017-04-23:
I gotta wait until another ESP is shipped.

crankyoldgit, 2017-05-16:
@darshkpatel Did your next ESP arrive? Care to test master and/or v2.0-dev branch?

darshkpatel, 2017-05-16:
I'm sorry, It just slipped out of my mind
I'll test it this week for sure

crankyoldgit, 2017-05-16:
No worries. I'm writing some test-cases for the code now. So, in a day or so I should be more confident if it will work or not, but nothing beats real-world tests.

crankyoldgit, 2017-05-17:
@darshkpatel & @tejipal  The [v2.0-dev](https://github.com/markszabo/IRremoteESP8266/tree/v2.0-dev) tree should have an updated & working version for both sending and decoding.

I think I've ironed out the remaining bugs when I wrote the uinit tests for them, but real hardware is different etc.

Note: You'll need to call it via `sendRCMM(data, 32);` if you want to send your 32-bit messages you've listed in this issue. It defaults to 24-bits.

Feedback etc most welcome. Good or bad. :)

I'm hoping to have a release candidate out for v2.0 by the end of the month.

darshkpatel, 2017-05-17:
I tried it with my existing receiver tsop1738 but it seems like 38khz
TSOP38338  is needed, it still gives different codes for same button presses

crankyoldgit, 2017-05-18:
Out of curiosity, any chance you can include the dumps here please?

I would have thought the tsop1738 would have worked fine as it is also a 38kHz receiver.

darshkpatel, 2017-05-18:
Next time I set it up I'll dump it here

darshkpatel, 2017-05-18:
This week maybe

crankyoldgit, 2017-05-18:
Just checking. You used the latest v2.0-dev branch in your latest tests?

darshkpatel, 2017-05-18:
Yeah, i pulled the v2.0dev

darshkpatel, 2017-05-18:
@crankyoldgit  I am attaching the dump for my Tatasky remote , looks like this 
[image]

First two are pressing the same button and the third is for another button on the number pad.

also the last two cases are while sliding the finger actoss the touchpad ( not sure what it's called , but remote blasts out a lot of codes while sliding the finger , similar to scrolling with fingers )

```
Encoding  : MITSUBISHI
Code      : 0 (0 bits)
Timing[35]: 
     + 450, - 250     + 200, - 300     + 200, - 650     + 200, - 650
     + 200, - 300     + 200, - 800     + 200, - 650     + 200, - 300
     + 200, - 300     + 200, - 300     + 200, - 650     + 200, - 450
     + 200, - 650     + 200, - 300     + 200, - 300     + 200, - 300
     + 200, - 450     + 200
uint16_t  rawData[35] = {450,250, 200,300, 200,650, 200,650, 200,300, 200,800, 200,650, 200,300, 200,300, 200,300, 200,650, 200,450, 200,650, 200,300, 200,300, 200,300, 200,450, 200};  // MITSUBISHI 0
uint64_t  data = 0x0;

Encoding  : MITSUBISHI
Code      : 0 (0 bits)
Timing[35]: 
     + 450, - 250     + 200, - 300     + 200, - 650     + 200, - 650
     + 200, - 300     + 200, - 800     + 200, - 650     + 200, - 300
     + 200, - 300     + 200, - 650     + 200, - 650     + 200, - 450
     + 200, - 650     + 200, - 300     + 200, - 300     + 200, - 300
     + 200, - 450     + 200
uint16_t  rawData[35] = {450,250, 200,300, 200,650, 200,650, 200,300, 200,800, 200,650, 200,300, 200,300, 200,650, 200,650, 200,450, 200,650, 200,300, 200,300, 200,300, 200,450, 200};  // MITSUBISHI 0
uint64_t  data = 0x0;

Encoding  : MITSUBISHI
Code      : 0 (0 bits)
Timing[35]: 
     + 500, - 250     + 200, - 300     + 200, - 650     + 200, - 650
     + 200, - 300     + 200, - 800     + 200, - 650     + 200, - 300
     + 200, - 300     + 200, - 300     + 200, - 650     + 200, - 450
     + 200, - 650     + 200, - 300     + 200, - 300     + 200, - 300
     + 200, - 650     + 200
uint16_t  rawData[35] = {500,250, 200,300, 200,650, 200,650, 200,300, 200,800, 200,650, 200,300, 200,300, 200,300, 200,650, 200,450, 200,650, 200,300, 200,300, 200,300, 200,650, 200};  // 
 MITSUBISHI 0
uint64_t  data = 0x0;

Encoding  : UNKNOWN
Code      : EA0E5282 (4 bits)
Timing[7]: 
     + 400, -3700     + 150, -1300     + 150, -2950     + 200
uint16_t  rawData[7] = {400,3700, 150,1300, 150,2950, 200};  // UNKNOWN EA0E5282

Encoding  : UNKNOWN
Code      : A8B55377 (9 bits)
Timing[17]: 
     + 450, - 300     + 200, - 300     + 150, - 700     + 150, - 700
     + 150, -2100     + 150, - 500     + 200, -1550     + 150, -1000
     + 150
uint16_t  rawData[17] = {450,300, 200,300, 150,700, 150,700, 150,2100, 150,500, 200,1550, 150,1000, 150};  // UNKNOWN A8B55377
```

crankyoldgit, 2017-05-19:
Nice. Thanks for the data.
Not sure about the UNKNOWN's, but your results have highlighted a bug in how I've re-done decodeMitsubishi(). It shouldn't be grabbing and claiming those messages.

crankyoldgit, 2017-05-19:
@darshkpatel Once #214 is merged into v2.0-dev, I think it should decode RC-MM for you,

darshkpatel, 2017-05-19:
Sure, I'll try that

crankyoldgit, 2017-05-19:
You could try https://github.com/markszabo/IRremoteESP8266/tree/v2.0-dev-mitsubishi-unittests in the meantime.

crankyoldgit, 2017-05-28:
This should now be working in [v2.0-RC0](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.0-RC0). Marking this closed. Please either re-open if there are issues, or create a new issue.

closed 2017-05-28 by crankyoldgit