crankyoldgit/IRremoteESP8266 #22: Restarting if continues ir received

rajasekarsarangapani, 2016-04-06:
i like to say thank you for this library creator and supporting members,
its decoding perfectly but ESP8266 Restarting, If receive continuous IR Signal ,
Iam using gpio13 to connect ir receiver(TSOP1738)

iam using samsung TV remote
[image]
[image]

darshkpatel, 2016-04-07:
Don't encode the values in hex and print, mine worked pretty well without
encoding the output in hex

rajasekarsarangapani, 2016-04-07:
kindly explain why don't encode in to hex value,
how to use switch case without convert hex,have any other way to use switch case

darshkpatel, 2016-04-07:
You can directly use the values, by assigning results.value to a variable , i have given a example snipped below
`
code = results.value;
switch(code){
    //case 1
    case 3778927144:
     //do something

}
`

rajasekarsarangapani, 2016-04-07:
Thanks for your code, I tested but still restarting, can you please share your hardware schematic diagram

darshkpatel, 2016-04-08:
I am replying form my mobile device so i cannot upload a schematic ,
I am using a ESP-12E and have attached my tsop1738 to Gpio 5 , and have
provided it with a independent power supply , and is not sharing it with
the esp , as it creates issues sometimes.

rajasekarsarangapani, 2016-04-08:
Thanks,I will check my connection, if you send schematic diagram, this is my diagram
[image]

rajasekarsarangapani, 2016-04-15:
It's restart again and again if receive few ir codes

sivanandareddy, 2016-04-19:
TSOP1738 is working very fine with 3.3V directly connect Vin to 3.3V, GND to GND, and output to some GPIO pin. I'm able to decode the my Videcon DTH remote. The problem with me is, I'm not able to send the messages using IRsend.sendNEC(). If you have tried it please reply.

markszabo, 2016-04-19:
@sivanandareddy Try connecting a normal led to the output pin while sending. It should blink for a short time. If it does, then try decoding it with an other arduino/esp8266. Usually the receiver will pick up the signal from a normal led, though you need to put them close to each other. Then try with the IR led. Also (if possible) try sending the IR codes from an arduino.

sivanandareddy, 2016-04-19:
I'm sure it is blinking something. I'll try with another arduino board and decode it. Another strange thing I have found is that carrier freqency of IR transmission is only 34.5kHz instead of 38kHz. I'm using NodeMCU with ESP12-E

sivanandareddy, 2016-04-19:
@rajasekarsarangapani  See section Timing and delays in the esp8266/arduino reference link https://github.com/esp8266/Arduino/blob/master/doc/reference.md. They say not to use delay as it blocks the some wifi routines which will reset the chip, if wifi is enabled.

rajasekarsarangapani, 2016-04-21:
Thanks for the info, I will try

igrr, 2016-07-21:
FYI, a lot of these `Exception (0)` issues reported here have to do with the fact that GPIO ISR handler is placed into flash. ISR handler should instead be marked with `IRAM_ATTR` for placement into RAM. Otherwise if the interrupt happens while flash cache is disabled, ISR code will not be accessible, causing exception 0.

lwalpurgis, 2016-07-22:
Is any Solution for this problem available? My ESP restarts with the demo rcv example if delay() is used ....

tekker, 2016-07-25:
As **igrr** suggested, changing the ISR method signatures has resolved all of the issues with receiving repetitive IR signals causing WDT resets for me.

Changes applied:

(_IRemoteESP8266.cpp, line 478_)

 **static void  ICACHE_FLASH_ATTR gpio_intr(void *arg) {**

to:

**static void ICACHE_RAM_ATTR gpio_intr(void *arg) {**

and (_IRemoteESP8266.cpp, line 470_):

**static void ICACHE_FLASH_ATTR read_timeout(void *arg) {**

to

**static void ICACHE_RAM_ATTR read_timeout(void *arg) {**

AndrewMohawk, 2016-11-23:
@tekker  / @igrr : Thank you. I've spent ages trying to figure out the problem and these fixes solved it for me. If you guys are ever in Cape Town, South Africa I will gladly buy you a beverage of your choosing!

closed 2017-03-02 by roidayan