crankyoldgit/IRremoteESP8266 #24: air conditioners no work

josep112, 2016-04-23:
I have successfully used the irsend.sendRaw in most remote controls, but does not work in air conditioners, how do I work because the codes seem to be alike see an example of the power of an air conditioner sanyo: unsigned int  rawData[99] = {4450,4450, 550,1700, 550,600, 550,1650, 550,1700, 550,600, 550,600, 550,1700, 550,600, 550,600, 550,1700, 550,600, 550,600, 550,1700, 550,1700, 550,600, 550,1700, 550,1700, 550,600, 550,1700, 550,1650, 550,1700, 550,1700, 550,1650, 550,1650, 550,600, 550,1700, 550,600, 550,600, 550,600, 550,600, 550,600, 550,600, 550,1700, 550,1700, 550,600, 550,1700, 550,600, 550,600, 550,600, 550,600, 550,600, 550,600, 550,1700, 550,600, 550,1700, 550,1700, 550,1700, 550,1700, 500};  // UNKNOWN 70B6FC0A  
Any help is welcome thank you

markszabo, 2016-04-30:
Sometimes it's not a software issue, but a hardware: some receivers are filtered to only a given infrared frequency, and won't work with other emitters. I had a similar problem in the past and buying a different IR led solved the issue, so you might want to look up if there is any specification about the frequency of the IR used in your appliance, and try buying a led which matches it.

josep112, 2016-05-02:
Thanks for the feedback!
I'm using the frequency of 38khz, the receiver he reads, as you can see above in the code I posted, but when sending does not work

chnchg, 2016-05-16:
The code can have more than 99 intervals for remotes of air conditioner. You can increase the RAWBUF value in IRremoteESP8266.h and try again.

josep112, 2016-05-16:
Actually the problem was the amount of greater than 99 ram air control has forced 199 problem solved thanks chnchg

closed 2016-05-16 by josep112