crankyoldgit/IRremoteESP8266 #25: How to process UNKNOWN encoding / NEC can only turn on cannot turn off

casper-gh, 2016-04-30:
First I'd like to thank you for your awesome work.

I have 2 issues right now:

- For UNKNOWN encoding, can I still be able to use any send() function, or it isn't implemented yet?
```
Encoding  : UNKNOWN
Code      : 6D23589D (32 bits)
Timing[93]: 
     +1300, - 450     +1300, - 450     + 500, -1250     +1300, - 450
     +1300, - 450     + 450, -1300     + 500, -1250     + 450, -1300
     + 500, -1250     + 450, -1300     + 450, -1300     +1350, -7200
     +1350, - 400     +1350, - 400     + 500, -1250     +1300, - 450
     +1300, - 450     + 450, -1300     + 500, -1250     + 450, -1300
     + 450, -1300     + 500, -1250     + 450, -1300     +1300, -7200
     +1300, - 450     +1300, - 450     + 450, -1300     +1350, - 400
     +1300, - 450     + 450, -1300     + 500, -1250     + 450, -1300
     + 450, -1300     + 450, -1300     + 450, -1300     +1300, -7200
     +1300, - 450     +1300, - 450     + 450, -1300     +1350, - 400
     +1300, - 400     + 450, -1300     + 450, -1300     + 500, -1250
     + 450, -1300     + 450, -1300     + 500
unsigned int  rawData[93] = {1300,450, 1300,450, 500,1250, 1300,450, 1300,450, 450,1300, 500,1250, 450,1300, 500,1250, 450,1300, 450,1300, 1350,7200, 1350,400, 1350,400, 500,1250, 1300,450, 1300,450, 450,1300, 500,1250, 450,1300, 450,1300, 500,1250, 450,1300, 1300,7200, 1300,450, 1300,450, 450,1300, 1350,400, 1300,450, 450,1300, 500,1250, 450,1300, 450,1300, 450,1300, 450,1300, 1300,7200, 1300,450, 1300,450, 450,1300, 1350,400, 1300,400, 450,1300, 450,1300, 500,1250, 450,1300, 450,1300, 500};  // UNKNOWN 6D23589D

Encoding  : UNKNOWN
Code      : F6D1C6CE (32 bits)
Timing[99]: 
     +1300, - 450     +1300, - 450     + 450, -1300     +1300, - 450
     +1300, - 450     + 450, -1300     + 450, -1300     + 450, -1300
     + 450, -1300     + 450, -1300     + 450, -1300     +1300, -7200
     +1300, - 450     +1300, - 450     + 450, -1300     +1300, - 450
     +1300, - 450     + 450, -1300     + 450, -1300     + 450, -1300
     + 450, -1300     + 450, -1300     + 450, -1300     +1300, -7200
     +1300, - 450     +1300, - 450     + 500, -1250     +1300, - 450
     +1300, - 450     + 450, -1300     + 450, -1300     + 450, -1300
     + 450, -1300     + 450, -1300     + 500, -1250     +1300, -7200
     +1300, - 450     +1300, - 450     + 450, -1300     +1300, - 450
     +1300, - 450     + 450, -1300     + 450, -1300     + 450, -1300
     + 450, -1300     + 450, -1300     + 500, -1250     +1300, -7200
     +1300, - 450     +1300
unsigned int  rawData[99] = {1300,450, 1300,450, 450,1300, 1300,450, 1300,450, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 1300,7200, 1300,450, 1300,450, 450,1300, 1300,450, 1300,450, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 1300,7200, 1300,450, 1300,450, 500,1250, 1300,450, 1300,450, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 500,1250, 1300,7200, 1300,450, 1300,450, 450,1300, 1300,450, 1300,450, 450,1300, 450,1300, 450,1300, 450,1300, 450,1300, 500,1250, 1300,7200, 1300,450, 1300};  // UNKNOWN F6D1C6CE
```

- For a NEC encoding, I could turn (TV) on, but when I trigger it again, it won't turn off despite the dump signal came out exactly the same:

```
Encoding  : NEC
Code      : 210704FB (32 bits)
Timing[67]: 
     +8950, -4400     + 650, - 600     + 600, - 550     + 650, -1650
     + 650, - 600     + 600, - 600     + 600, - 600     + 600, - 550
     + 650, -1650     + 650, - 600     + 600, - 550     + 650, - 550
     + 600, - 550     + 650, - 550     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, - 550     + 650, - 550     + 650, - 550
     + 650, - 550     + 650, - 550     + 650, -1650     + 650, - 550
     + 600, - 600     + 600, -1650     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, -1650     + 650, - 550     + 650, -1650
     + 650, -1650     + 650
unsigned int  rawData[67] = {8950,4400, 650,600, 600,550, 650,1650, 650,600, 600,600, 600,600, 600,550, 650,1650, 650,600, 600,550, 650,550, 600,550, 650,550, 650,1650, 650,1650, 650,1650, 650,550, 650,550, 650,550, 650,550, 650,550, 650,1650, 650,550, 600,600, 600,1650, 650,1650, 650,1650, 650,1650, 650,1650, 650,550, 650,1650, 650,1650, 650};  // NEC 210704FB
unsigned int  data = 0x210704FB;

Encoding  : NEC
Code      : 210704FB (32 bits)
Timing[67]: 
     +8950, -4400     + 650, - 550     + 650, - 550     + 650, -1650
     + 650, - 600     + 600, - 550     + 650, - 600     + 600, - 550
     + 650, -1650     + 650, - 550     + 650, - 600     + 600, - 600
     + 600, - 600     + 600, - 550     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, - 550     + 650, - 600     + 600, - 550
     + 650, - 550     + 600, - 600     + 600, -1650     + 650, - 600
     + 600, - 600     + 600, -1650     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, -1650     + 650, - 600     + 600, -1650
     + 650, -1650     + 650
unsigned int  rawData[67] = {8950,4400, 650,550, 650,550, 650,1650, 650,600, 600,550, 650,600, 600,550, 650,1650, 650,550, 650,600, 600,600, 600,600, 600,550, 650,1650, 650,1650, 650,1650, 650,550, 650,600, 600,550, 650,550, 600,600, 600,1650, 650,600, 600,600, 600,1650, 650,1650, 650,1650, 650,1650, 650,1650, 650,600, 600,1650, 650,1650, 650};  // NEC 210704FB
unsigned int  data = 0x210704FB;

Encoding  : NEC
Code      : 210704FB (32 bits)
Timing[67]: 
     +8950, -4400     + 650, - 600     + 600, - 550     + 650, -1650
     + 650, - 600     + 600, - 550     + 650, - 550     + 650, - 550
     + 650, -1650     + 650, - 600     + 600, - 600     + 600, - 600
     + 600, - 600     + 600, - 550     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, - 550     + 650, - 600     + 600, - 550
     + 650, - 550     + 650, - 600     + 600, -1650     + 650, - 600
     + 600, - 550     + 650, -1600     + 650, -1650     + 650, -1650
     + 650, -1650     + 650, -1650     + 650, - 600     + 600, -1650
     + 650, -1650     + 650
unsigned int  rawData[67] = {8950,4400, 650,600, 600,550, 650,1650, 650,600, 600,550, 650,550, 650,550, 650,1650, 650,600, 600,600, 600,600, 600,600, 600,550, 650,1650, 650,1650, 650,1650, 650,550, 650,600, 600,550, 650,550, 650,600, 600,1650, 650,600, 600,550, 650,1600, 650,1650, 650,1650, 650,1650, 650,1650, 650,600, 600,1650, 650,1650, 650};  // NEC 210704FB
unsigned int  data = 0x210704FB;
```

markszabo, 2016-04-30:
There is a sendRaw() function which can be used to send the rawData array, but that doesn't always work.

casper-gh, 2016-04-30:
Thanks @markszabo, but what value should I use for Hz parameter?

markszabo, 2016-04-30:
It's usually 38 by other encodings (simply check how they call enableIROut()), so that would be my guess

casper-gh, 2016-05-01:
Much appreciated, it works perfect with value 38.

However I have a last question, kind of out of scope of this app, but I'm new to Arduino:

   How do I set up the ESP8266 Arduino so the sketch starts automatically when the system reboots? When I disconnect the power from system, the sketch is gone.

markszabo, 2016-05-01:
Okay, just to make sure we are on the same page: you have an ESP8266 microcontroller (hardware), and you are using the Arduino IDE (software) to program it, right? In this case if you restart your ESP8266 (or just simply disconnect the power) the sketch should automatically restart.
However when ESP8266 starts it checks the GPIO0 pin. If that pin is LOW, then it will go to bootloader mode and wait for a new sketch to be uploaded. If it receives the new sketch, it stores it and then starts it (regardless of the state of the GPIO0 pin). But if your GPIO0 pin is still LOW, then when you restart it, it will go to bootloader mode again and wait for a new sketch. So check if your GPIO0 pin is LOW at startup :)

casper-gh, 2016-05-01:
@markszabo Yes we're on the same page, and I'm using the IRServer so the pin is set to 0 by its default. However after I disconnect and reconnect the power, I couldn't access the web server anymore. 

Sorry for being dumb, but what does is mean when you say "pin is LOW", and how can I set it to HIGH or whatever it is for the autostart to work?

Thanks a lot

casper-gh, 2016-05-01:
Okay I figured it out, changed `IRsend irsend(0);` to `IRsend irsend(4);` in IRServer

gorec2005, 2016-06-06:
Hi Mark!
do you have any news about long-block protocols ?

skyforcetw, 2016-11-25:
I'm processing UNKOWN IR encoding  recently.
I found there are a bit difference in IRremote between avr & ESP8266 version:

avr version: A long space will create a STOP state in IR receive.
ESP8266 version: Continued receive until read timeout.

This difference may cause avr receive one IR packet and STOP, ESP8266 receive more packets and signal length is non stable.

Is any chance to improve this?

Joypacharya, 2016-12-19:
I  thought you are decoding AC remote. 
You should increase your Buffer size.If you do this thing then you will get definitely result. 
In such case this is happyend. So,try it and reply me back.

crankyoldgit, 2017-03-18:
Closing this issue due to inactivity and it appears the original request has been addressed.

closed 2017-03-18 by crankyoldgit