crankyoldgit/IRremoteESP8266 #26: Problem air conditioner

josep112, 2016-06-03:
Hello
The air conditioner uses a raw 199, this has to change the IRremote.h -> #define RAWBUF 100 to 200 and change IRremoteInt.h #define _GAP 5000 change to 50000, the problem is that there are over 300 air and the string only accepts 255 to solve this?

annhan, 2016-06-06:
i can change 
`typedef struct {
  uint8_t recvpin;           // pin for IR data from detector
  uint8_t rcvstate;          // state machine
  unsigned int timer;     // state timer, counts 50uS ticks.
  unsigned int rawbuf[RAWBUF]; // raw data
  uint8_t rawlen;         // counter of entries in rawbuf
} 
irparams_t;`

to
`typedef struct {
  uint8_t recvpin;           // pin for IR data from detector
  uint8_t rcvstate;          // state machine
  unsigned long timer;     // state timer, counts 50uS ticks.
  unsigned long rawbuf[RAWBUF]; // raw data
  long rawlen;         // counter of entries in rawbuf
} 
irparams_t;`

it can leaning IR code over 300 ???

josep112, 2016-06-08:
Hi Annhan,
Thanks for the reply, I changed the code but did not work my code is below

#include <IRremoteESP8266.h>
long rawlen;
uint8_t RECV_PIN = 2; //an IR detector/demodulatord is connected to GPIO pin 2

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}

josep112, 2016-06-08:
Sorry I had not seen the link, I could change my raw and went from 254 to 347 will test the air thanks for the help

annhan, 2016-06-08:
hi josep112 ,
i change library for raw >255, That requires changing the data type in dozens of places throughout the source code, however. I have Test recv and send raw=278 air with F=38K OK.
Can you help me decode raw -> frequency IR.
thanks you

josep112, 2016-06-08:
Hi Annhan, 
Not quite understand what you need, you had said that does not speak in English which his country talves can talk in another language

safari2, 2016-10-16:
Hi all.

I've just solved the similar problem.  The issue is only with long IR reading. 
There are good solution with hardware interrupt driving reading: http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/
It works almost out of the box for ESP8266. Just change this code from 0 to your pin number (I used 12 ([D4 on nodemcu aka lolin](http://esp8266.ru/forum/resources/raspinovka-pinout-esp8266-nodemcu-v3-lolin.35/[/url])):
```
   attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE);//set up ISR for receiving IR signal
   detachInterrupt(0);//stop interrupts & capture until finshed here
   attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE);//re-enable ISR for receiving IR signal
```

So I've successfully got unsigned int  rawData  [439] for my Panasonic HVAC.  
Than you can use irsend.sendRaw() from this library.

helio58, 2017-05-19:
>So I've successfully got unsigned int rawData [439] for my Panasonic HVAC.
Than you can use irsend.sendRaw() from this library.

Please safari2 where is the irsend.sendRaw() ? and how to send the obtained code?
Thanks

safari2, 2017-05-22:
helio58, see link below.

crankyoldgit, 2017-07-04:
Closing this issue. Versions of the library later than v2.0 now support larger (256+) capture buffers.

closed 2017-07-04 by crankyoldgit