crankyoldgit/IRremoteESP8266 #28: Library not properly decoding Samsung remote

cimd, 2016-06-28:
I'm trying to use the library with a Samsung remote, and this is the data I get for a volume down key on the remote:
     **Code:** 83B19366
     **Raw:** rawData[67] = {4550,4550, 600,1700, 550,1750, 550,1750, 550,600, 550,600, 550,600, 550,600, 550,600, 550,1750, 550,1750, 550,1750, 550,600, 550,600, 550,600, 550,600, 550,600, 550,1750, 550,1750, 550,600, 550,1750, 550,600, 550,600, 550,600, 550,600, 550,600, 550,600, 550,1750, 550,600, 550,1750, 550,1750, 550,1750, 550,1750, 550};

If I try to send that signal using the hex code and the sendSamsung function, nothing happens. I am able to make it work though by sending the raw data.

I moved back into the original IRremote - which this library was based on - with a regular arduino, and I got a different hex code for the same volume down key: **E0E0D02F**. And if I use this code with the IRremoteESP8266 library and the sendSamsung function, it does work.

Is there a reason for the IRremoteESP8266 library not properly decoding - or at least not generating the right hex code - for the Samsung remote? Can someone provide me guidance on how to fix this?

Thanks,

Christian

chaeplin, 2016-06-29:
Check this https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteInt.h#L119-L134

cimd, 2016-06-29:
Looks like I had an older version of the library. I updated it based on this revision [https://github.com/markszabo/IRremoteESP8266/commit/b8521daea2e0a1e2810eefd476f1d7ca66f72f8d](url) and it worked.

Chaeplin, I have also revised Samsung pulse parameters but as per Chris Cyborg's IRLib. He uses values that are multiple of 8 bits. At the end of the day, it is very close to Sebastien's parameters.

Now I need to make it work with my LG home theater... :-)

closed 2016-06-29 by cimd