crankyoldgit/IRremoteESP8266 #29: IRsend.sendSony() problem

jordan-public, 2016-06-29:
Sony TV does not see the transmission from sendSony().  I simply use the IRSendDemo to send the following code to the Sony TV:
irsend.sendSony(0xC90, 12); // Volume Down

In order to debug the issue I did the following:
- Built another ESP8266 with IR Receiver and the sample IRrecvDumpV2.  It recognized the codes from both the original Sony Remote Control, as well as from my ESP8266 transmitter described above.  Below are 3 samples of each.  
- Previously tried the same IR LED on an Arduino build and saw that the Sony TV reads it, so the LED has the correct wavelength (940).  

I presume this may be some timing tolerance issue, but I am not savvy enough in the IR codes to spot the problem.  Please help.

Thanks,
Jordan
 P.S. 3 samples of readings by IRrecvDumpV2 from IRSendDemo and original Sony Remote:

1. Sent by IRSendDemo with irsend.sendSony(0xC90, 12); // Volume Down:

Encoding  : SONY
Code      : C90 (12 bits)
Timing[27]: 
     +2550, - 500     +1350, - 550     +1350, - 550     + 750, - 500
     + 750, - 500     +1350, - 550     + 700, - 550     + 750, - 550
     +1300, - 550     + 750, - 550     + 700, - 550     + 750, - 550
     + 700, -14550     + 200
unsigned int  rawData[27] = {2550,500, 1350,550, 1350,550, 750,500, 750,500, 1350,550, 700,550, 750,550, 1300,550, 750,550, 700,550, 750,550, 700,14550, 200};  // SONY C90
unsigned int  data = 0xC90;

Encoding  : SONY
Code      : C90 (12 bits)
Timing[25]: 
     +2550, - 500     +1350, - 550     +1300, - 500     + 750, - 500
     + 750, - 550     +1300, - 550     + 750, - 550     + 700, - 550
     +1350, - 550     + 700, - 600     + 650, - 550     + 700, - 550
     + 750
unsigned int  rawData[25] = {2550,500, 1350,550, 1300,500, 750,500, 750,550, 1300,550, 750,550, 700,550, 1350,550, 700,600, 650,550, 700,550, 750};  // SONY C90
unsigned int  data = 0xC90;

Encoding  : SONY
Code      : C90 (12 bits)
Timing[27]: 
     +2550, - 500     +1350, - 500     +1350, - 550     + 750, - 500
     + 750, - 500     +1350, - 500     + 800, - 500     + 750, - 550
     +1300, - 550     + 750, - 550     + 700, - 600     + 750, - 550
     + 700, -1800     + 250
unsigned int  rawData[27] = {2550,500, 1350,500, 1350,550, 750,500, 750,500, 1350,500, 800,500, 750,550, 1300,550, 750,550, 700,600, 750,550, 700,1800, 250};  // SONY C90
unsigned int  data = 0xC90;

----------------------------------

2. Sent by Original Sony Remote Control:

Encoding  : SONY
Code      : C90 (12 bits)
Timing[25]: 
     +2550, - 500     +1300, - 600     +1250, - 600     + 700, - 550
     + 650, - 600     +1250, - 600     + 650, - 600     + 650, - 550
     +1300, - 600     + 700, - 550     + 650, - 600     + 650, - 550
     + 650
unsigned int  rawData[25] = {2550,500, 1300,600, 1250,600, 700,550, 650,600, 1250,600, 650,600, 650,550, 1300,600, 700,550, 650,600, 650,550, 650};  // SONY C90
unsigned int  data = 0xC90;

Encoding  : SONY
Code      : C90 (12 bits)
Timing[25]: 
     +2550, - 500     +1300, - 550     +1250, - 600     + 650, - 550
     + 650, - 550     +1300, - 600     + 650, - 600     + 650, - 600
     +1250, - 600     + 650, - 600     + 700, - 550     + 700, - 550
     + 700
unsigned int  rawData[25] = {2550,500, 1300,550, 1250,600, 650,550, 650,550, 1300,600, 650,600, 650,600, 1250,600, 650,600, 700,550, 700,550, 700};  // SONY C90
unsigned int  data = 0xC90;

Encoding  : SONY
Code      : C90 (12 bits)
Timing[25]: 
     +2650, - 450     +1350, - 550     +1250, - 600     + 650, - 600
     + 650, - 600     +1250, - 600     + 650, - 550     + 650, - 600
     +1250, - 600     + 650, - 600     + 650, - 600     + 650, - 600
     + 650
unsigned int  rawData[25] = {2650,450, 1350,550, 1250,600, 650,600, 650,600, 1250,600, 650,550, 650,600, 1250,600, 650,600, 650,600, 650,600, 650};  // SONY C90
unsigned int  data = 0xC90;

jordan-public, 2016-07-19:
I figured out the problem.  Here is the (workaround): 
- Sony expects the code to be sent 3 times.  So, sending it 3 times and pausing about 40ms (using delay(40);) fixed the problem for me.  Now the Sony TV responds.  

In addition, if someone knows about guidelines which devices need repeated transmit like this, how many times and at what interval, please let us know.  

And Mark, thank you for contributing this valuable library.  

Cheers,
Jordan

GabeHC, 2017-01-24:
I had send it three times, But still not working.

 sendSony works fine on UNO. 
Compare two version's differences.
>in Arduino-IRRemote
```
void  IRsend::sendSony (unsigned long data,  int nbits)
{
	// Set IR carrier frequency
	enableIROut(40);

	// Header
	mark(SONY_HDR_MARK);
	space(SONY_HDR_SPACE);

	// Data
	for (unsigned long  mask = 1UL << (nbits - 1);  mask;  mask >>= 1) {
		if (data & mask) {
			mark(SONY_ONE_MARK);
			space(SONY_HDR_SPACE);
		} else {
			mark(SONY_ZERO_MARK);
			space(SONY_HDR_SPACE);
  	}
  	}

	// We will have ended with LED off
}
``` 
>IRremoteESP8266
```
void IRsend::sendSony(unsigned long data, int nbits) {
  // Set IR carrier frequency
  enableIROut(40);
  // Header
  mark(SONY_HDR_MARK);
  // Data
  for (unsigned long mask = 1UL << (nbits - 1); mask; mask >>= 1) {
    if (data & mask) {  // 1
      mark(SONY_ONE_MARK);
      space(SONY_HDR_SPACE);
    } else {  // 0
      mark(SONY_ZERO_MARK);
      space(SONY_HDR_SPACE);
    }
  }
  // Footer
}
```
Does that space needed after mark in header setction?
It makes no different when adding it in, still not working.

marcosamarinho, 2017-01-24:
Please try out the code at the pull request 1 marcosamarinho/IRremoteESP8266 that I made many changes including  sony, note that this code already send more than I time , them you not need loop anymore.

lukecorkill, 2017-02-16:
Thanks - this helped me get sending to a Sony amplifier working from an ESP8266.  Keep up the good work!

crankyoldgit, 2017-02-24:
@GabeHC Is correct. There was a missing space() after the initial header mark(). I've added it.
Can someone with a Sony device please confirm this addresses the issue?

MopheusDG, 2017-03-06:
I updated my library on Arduino but still no luck getting my Sony AMP to work. Tried single "irsend.sendSony(0x240C, 15);" and triple using delay 40 and nothing... Any ideas ?

crankyoldgit, 2017-03-07:
@MopheusDG Yes, a simple download per the instructions on the main page should be fine. Is your setup working with other devices at all? If not, try using a digital camera (on a phone etc) to see if you can see the IR LED is flashing when you send a signal.
Have you checked what IR code your existing remote is sending via one of the decoding or dumping programs?

crankyoldgit, 2017-03-07:
Re: the repeating problem. Yes, there should be a delay after each command is sent. See https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.cpp#L644 Currently the irsend routines do not enforce this. It's something I want to add but finding the info for each protocol is a fair bit of work. Depending on the protocol, I'd *guess* you should approx delay(40) or so before beginning the next IR command/packet transmission.

If you do find out from a datasheet or similar, what the minimum sequence length is for your protocols, please let us know, as I think we should support it.

Now, to the Sony problem. I've rechecked the code, it appears to be functionally identical to the Arduino IRremote library we are based off. (e.g. http://www.sbprojects.com/knowledge/ir/sirc.php)

Are you able to use another ESP8266 or Arduino board with an IR decoder/raw dump to read what the ESP is sending at all?

Details on the SONY protocol seem to be here:
http://www.sbprojects.com/knowledge/ir/sirc.php

I note that that page says it sends the LSB first, where we(IRremote) seem to be sending the MSB first which has me scratching my head.

I hand decoded the raw data you presented, it indeed decodes to 0x540C/B101010000001100. AFAICT, that's what we should be transmitting too.

I've visually inspected another implementation of the Sony IR protocol, and it seems we have it correct.

Are you transmitting the code 3 times in sequence, with a delay(40) between each? Some references say sony devices need to have the code repeated at least 3 times.

MopheusDG, 2017-03-07:
I have an Arduino and the ESP, let me do some more testing... I'm gonna try using RAW mode too. When I first started this project I was using an Arduino and Sony using RAW (Didn't know about SendSONY at the time), so, gonna do some more testing an get back to you guys.

Thanks a lot for your help.

crankyoldgit, 2017-03-07:
Thanks. Looking forward to your results. I'd like to get to the bottom of the Sony issue and fix what ever needs done to get it working etc.

MopheusDG, 2017-03-09:
Ok. This are my findings so far. 1) I found a problem on my RAW Codes from SONY cause my receiver was only 38 hz, so looked for std one's on the LIRC DB. Now, in Arduino UNO works perfectly using CODE + Delay 40 (3 times). So that's for sure, I need to send it 3 times.
Now tested the same on ESP8266 but there's still some problem, and I think it's related to something on my code ? Not sure... this is what I'm doing:

Created a Switch CASE for Fauxmo On Message based on device_id, so far.. works on everything except when I want to send MORE than 1 send on the same CASE... like this:

    case 2: // 'TV ESPN'
        irsend.sendSAMSUNG(0xE0E020DF, 32); // This is remote #1
        Serial.println("PARTE 1");
        delay(500);
        irsend.sendSAMSUNG(0xE0E030CF, 32); // This is remote #7
        Serial.println("PARTE 2");
    break;

I tested on Samsung after not having luck with Sony, just to see if my TV that works fine on other cases, can handle 2 irsend lines, but for some reason, that code gives me:

PARTE1
PARTE2

on the monitor, but only send a #7 on my TV ! It's driving me crazy.... and I think that's why SONY is not working too, only the first send works probably. Any ideas ? I tried using brackets, using a FOR inside the case, changing DELAY time... I have no more ideas.

MopheusDG, 2017-03-09:
I think a found the problem #2, I was handling the irsends inside the onmessage code of fauxmo, I moved it to the Loop and I think it's working now. Sony needs to be send 3 times.

crankyoldgit, 2017-03-10:
Okay. So to summarise, the library's sendSony() works as intended, but for a Sony device to accept the code/command, the IR code has to be sent 3 times in total, with a small delay between them.

e.g. A sequence of the following should work:

```
irsend.sendSony(0x540C, 15);
delay(40);
irsend.sendSony(0x540C, 15);
delay(40);
irsend.sendSony(0x540C, 15);
```

Obviously, that's not how we'd implement it, but is that correct summation of your results?
I want to confirm before I make the changes to the lib.

MopheusDG, 2017-03-10:
That is correct. At least on my Sony AMP only sending 3 times with delay works. Be aware that same goes for the Arduino Library, at least on my case, I also have to send 3 times on Arduino  (both libraries are the same, right ?), so that was expected too.

Thanks for your help.

crankyoldgit, 2017-03-10:
I'm glad there was nothing 'wrong' with the code, because I sure couldn't see it.

Unfortunately, lots of existing code that uses us will expect the current behaviour, but I'll add some notes and an easy method to invoke it the correct number of times.

MopheusDG, 2017-03-10:
That's great. What really got me lost was Sony using 40 hz instead of 38 hz, cause most of the standard IR Receivers work only 38 hz, with senders can do anything in most cases. So when I started the project, captured all my devices, some Samsungs, some Unknown and Sony got in as Unknown because of that, and didn't realize. Then when I found the problem, searched the LIRC DB for Sony and found some matching codes from other Sony's, but again not working.... and finally, remembered what someone said before about the 3 times, and that was it.

Maybe explaining that (in good English) could help other ppl with same problem. By the way, I also learn (that I didn't know) that sometimes you have "codes" for specific functions on TV's or whathever, that are NOT in the actual remote. For example, my TV Remote does not have a HDMI button, just SOURCE that switches between all sources, playing around with the codes found a code, that will do TV/HDMI switching, ignoring the rest of the sources, and that's pretty usefull for my Alexa project. So, that could be a nice help for someppl, again, explained better probably :)

crankyoldgit, 2017-03-11:
With that commit, I think I can close this issue.

closed 2017-03-11 by crankyoldgit

kiralikbeyin, 2017-03-16:
```
Encoding  : SONY
Code      : 240C (15 bits)
Timing[31]: 
     +2450, - 550     + 700, - 550     +1300, - 550     + 650, - 550
     + 650, - 600     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1250, - 550     +1250, - 600     + 700, - 550     + 700
unsigned int  rawData[31] = {2450,550, 700,550, 1300,550, 650,550, 650,600, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1250,550, 1250,600, 700,550, 700};  // SONY 240C
unsigned int  data = 0x240C;
```
I cant transmit and tried 3 times...
````
  irsend.sendSony(dondur , pirbit.toInt());
        delay(40);
        irsend.sendSony(dondur , pirbit.toInt());
        delay(40);
        irsend.sendSony(dondur , pirbit.toInt());
````
Any suggestion?

crankyoldgit, 2017-03-17:
@kiralikbeyin Please open a new issue if you are having a problem with sendSony.

Things you should include in the _new_ issue?
Are you using a version of the library post #101? e.g. the latest/current master.
As of #101 you shouldn't need the delay(40)'s in your code. That's now done in sendSony().

Also, what is your above decode from? i.e. Is it of your physical remote, or from your ESP8266?
More info is the best way for us to be able to help you.

kiralikbeyin, 2017-03-17:
#118 ok