crankyoldgit/IRremoteESP8266 #31: Bug in read_timeout method, "eating" the initial mark duration and producing result different from Arduino...

Nca78, 2016-07-07:
Hello,

 I think there is a bug at line 487 in IRremoteESP8266.cpp, the duration of the mark is not saved and then the data received/interpreted is not the same than the one we get when using iRemote with and arduino board. It also fails to decode most of the protocols I have been able to try.

`if (irparams.rcvstate == STATE_IDLE) {
		irparams.rcvstate = STATE_MARK;
		irparams.rawbuf[irparams.rawlen++] = 20;		
	}
    else if (irparams.rawlen < RAWBUF) {
		irparams.rawbuf[irparams.rawlen++] = (now - start) / USECPERTICK + 1;
}`

When state is STATE_IDLE then  _(now - start) / USECPERTICK + 1;_ should be executed too so that we save the duration of the mark in the buffer.

I changed the code to this and now the protocols I try (Samsung, LG, ...) are successfully decoded, and unknown protocols return me the exact same value that I have when I use the arduino version of iRemote.

I am not sure if moving the test  _irparams.rawlen < RAWBUF_ outside is ok, do we have a case where the state can be STATE_IDLE and we have _irparams.rawlen_ not 0 ?

`	if (irparams.rawlen < RAWBUF) { 
if (irparams.rcvstate == STATE_IDLE) {
irparams.rcvstate = STATE_MARK;
}
irparams.rawbuf[irparams.rawlen++] = (now - start) / USECPERTICK + 1;
}`

markszabo, 2016-07-08:
Thank you for the bug report. For me, and for others the library way working the original way, so it's an interesting issue to investigate. As I understood right now it saves the constant 20 instead of `(now - start) / USECPERTICK + 1` when `irparams.rcvstate == STATE_IDLE`. Can you please send a pull request, so I can review and test your code?

Nca78, 2016-07-08:
I kept the adding of constant 20 also.
I don't know how to make a pull request at the moment, I'm all new to github and never used git before, please leave me a little time to learn :)
I will provide you some examples also if you want.

crankyoldgit, 2017-03-03:
Just chasing up this. If you don't know how to a pull request, can you send us a 'diff' from the current source code to your version that encompases your changes?

roidayan, 2017-03-03:
I'm following this as well. Would be the first mark in your case a constant 1 ? As start and now with be the same? Is that what makes the difference?

roidayan, 2017-03-04:
sorry my mistake. start is 0 first time but now will be the current micros. if we'll use the calc also for when state is idle then we'll get different result for the first item in rawbuf each time we reset to idle state.  that's the reason for the constant for first item. Now sure why constant 20 was chosen.

crankyoldgit, 2017-03-23:
Can this issue be closed now?

crankyoldgit, 2017-04-11:
Marking this issue closed. If you disagree, please re-open.

closed 2017-04-11 by crankyoldgit