crankyoldgit/IRremoteESP8266 #33: Esp resets in GC example

mohitverma123, 2016-07-08:
In the IRGCTCPserver example, 

I am getting following error and resets

IR TCP Server
192.168.1.117

Exception (0):
epc1=0x40106b71 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffeff40 end: 3fff01c0 offset: 01a0

>>>stack>>>
3fff00e0:  3ffef0b8 00000015 00000015 4010020c  
3fff00f0:  3ffeef94 ffffffff 00000084 4010068c  
3fff0100:  3ffeef94 00000006 00000001 3fff0160  
3fff0110:  3ffeef94 ffffffff 00000084 40201ff2  
3fff0120:  00000000 00000000 00000000 40208609  
3fff0130:  00000193 3fff1868 3fff0160 40208710  
3fff0140:  00000000 3ffeef98 3ffeef98 00000384  
3fff0150:  00000001 3ffeef88 3ffeef98 40202242  
3fff0160:  3fff14b4 0000019f 00000197 402082a0  
3fff0170:  00000000 3fff10cc 3fff130c 0000019f  
3fff0180:  00000197 7501a8c0 3ffef0dc 40208278  
3fff0190:  00000000 00000000 00000001 3ffef198  
3fff01a0:  3fffdad0 00000000 3ffef190 40208fec  
3fff01b0:  feefeffe feefeffe 3ffef1a0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

annhan, 2016-07-08:
it can funtion realloc and malloc. When i change esp not reset

crankyoldgit, 2017-04-12:
Is this still happening for you with the current revision of the library?

crankyoldgit, 2017-05-14:
Marking this closed. It should be fixed now, and re-written in the v2.0 release

PR #195 (2017-05-14): Cleanup IRGCTCPserver.ino
- [bug] Actually free up the dynamically allocated memory.
- Simplify the mess of malloc/reallocs. Make a single malloc instead.
- Add check for malloc failing. Report & reboot if failed.
- Change variables to match style guide.
- Basically re-write the parsing code.
- Add detailed instructions on how to use this sketch.

Fixes #33
commit 790be5ee (2017-05-14): Cleanup IRGCTCPserver.ino (#195)

- [bug] Actually free up the dynamically allocated memory.
- Simplify the mess of malloc/reallocs. Make a single malloc instead.
- Add check for malloc failing. Report & reboot if failed.
- Change variables to match style guide.
- Basically re-write the parsing code.
- Add detailed instructions on how to use this sketch.

Fixes #33
closed 2017-05-14 by crankyoldgit