crankyoldgit/IRremoteESP8266 #34: Having issues with sendGC

androiduser99, 2016-07-13:
I'm using sendGC as I'm converting from a iTach. I have it working with 3 other devices but will not control an Onkyo AMP. Exact same codes I'm using through the iTach so I know the codes are good. Any ideas what to try or how to analyse the IR signal?

Thank you!

hishamk, 2016-07-13:
Please post a few working codes that worked (indicate device and function) for you plus non-working Onkyo codes.

Cheers

androiduser99, 2016-07-13:
Thanks for the follow up. I was able to do a little more testing today and as it turns out this is working fine on the standalone ESP8266! Sorry, I should have stated more details with the setup that is not working. I also have an Ardunio + ESP8266 setup that runs some auto amp power code plus direct IR. I used grabbed the sendGC code for the existing IR library I'm using on that device and 3 devices are working great but the Onkyo is not for some odd reason. The IR emitter blinks as expected but the IR code is not recognized on the Onkyo. I take the exact same device with same emitter and trigger an IR code using the same method and it works great. I use that same GC IR code via the iTach and it works fine. Seems like there may be a timing issue on the Ardunio that the Onkyo is sensitive to but the other devices are not.

Here is one of the codes I'm using. To be clear no codes on the Onkyo work via sendGC on the Arduio Nano but it works with other devices and this IR code works with the Onkyo with the iTach.

38479,1,1,342,171,21,21,21,63,21,21,21,21,21,63,21,21,21,63,21,63,21,63,21,21,21,63,21,63,21,21,21,63,21,63,21,21,21,21,21,21,21,63,21,21,21,21,21,21,21,21,21,21,21,63,21,63,21,21,21,63,21,63,21,63,21,63,21,63,21,1484,342,85,21,3655

Note: for the heck of it I also tried 38000 for the the khz. If it's a timing issue I suppose I could keep trying different values in increments/decrements of 1000...

Thank you!

androiduser99, 2016-07-13:
I tried incrementing the khz and sure enough it worked at 41000 on the Ardunio whereas the same code works at 38000 directly on the ESP8266.

Maybe this will help someone else having challenges with this library on an Ardunio.

crankyoldgit, 2017-02-24:
Thanks for the update and info. I'm going to mark this closed for now.

closed 2017-02-24 by crankyoldgit