crankyoldgit/IRremoteESP8266 #42: Issue sending NEC codes, but raw works fine

robbiet480, 2016-08-31:
Hey,

So I know my setup is working since sending raw data controls my Seville Classics fan. The IRRecvDemo says that it's getting NEC codes, and some existing code on Github makes me think it's NEC, but whenever I send NEC out the fan won't respond. If I send the raw data it works fine.

Here's example output from the receiver:

```
Encoding  : NEC
Code      : 10203040 (32 bits)
Timing[131]: 
     +9400, -4800     + 600, - 500     + 600, - 500     + 600, - 550
     + 600, -1600     + 600, - 500     + 600, - 550     + 600, - 500
     + 550, - 550     + 600, - 550     + 550, - 550     + 600, -1600
     + 600, - 500     + 600, - 550     + 550, - 550     + 600, - 500
     + 600, - 550     + 600, - 500     + 600, - 550     + 550, -1650
     + 600, -1600     + 600, - 500     + 550, - 550     + 600, - 550
     + 600, - 500     + 600, - 500     + 600, -1650     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, - 500     + 600, - 550
     + 550, - 550     + 600, - 500     + 600, -1600     + 600, - 550
     + 550, -1650     + 600, - 500     + 600, - 550     + 550, - 550
     + 600, - 500     + 600, - 550     + 600, - 500     + 550, - 550
     + 600, - 550     + 600, - 500     + 550, - 550     + 600, - 550
     + 600, - 500     + 600, - 500     + 600, -1600     + 600, -1650
     + 550, - 550     + 600, - 500     + 600, - 550     + 600, - 500
     + 600, - 500     + 600, - 550     + 550, -1650     + 550, -1650
     + 550, -1650     + 550, - 550     + 600, - 550     + 600, - 500
     + 600, - 500     + 600
unsigned int  rawData[131] = {9400,4800, 600,500, 600,500, 600,550, 600,1600, 600,500, 600,550, 600,500, 550,550, 600,550, 550,550, 600,1600, 600,500, 600,550, 550,550, 600,500, 600,550, 600,500, 600,550, 550,1650, 600,1600, 600,500, 550,550, 600,550, 600,500, 600,500, 600,1650, 550,550, 600,500, 600,550, 600,500, 600,550, 550,550, 600,500, 600,1600, 600,550, 550,1650, 600,500, 600,550, 550,550, 600,500, 600,550, 600,500, 550,550, 600,550, 600,500, 550,550, 600,550, 600,500, 600,500, 600,1600, 600,1650, 550,550, 600,500, 600,550, 600,500, 600,500, 600,550, 550,1650, 550,1650, 550,1650, 550,550, 600,550, 600,500, 600,500, 600};  // NEC 10203040
unsigned int  data = 0x10203040;
```

I try sending with
```c++
irsend.sendNEC(0x10203040, 32);
```

Can anyone help?

robbiet480, 2016-08-31:
Quick note that irdb.tk/decode/ says the following for the code above:

> Protocol Gap-650-1650-64?, device 8, subdevice 4, OBC 14, misc = 8.4.C.2.A.0.6.E

The formatted version of the raw data needed for irdb is 

```
+9450 -4700 +650 -500 +600 -550 +600 -500 +600 -1550 +650 -500 +600 -550 +600 -500 +600 -500 +650 -500 +600 -500 +600 -1600 +600 -500 +650 -500 +600 -500 +600 -550 +600 -500 +600 -500 +600 -550 +600 -1550 +650 -1550 +650 -500 +600 -500 +650 -500 +600 -500 +600 -550 +600 -1550 +650 -500 +600 -500 +650 -500 +600 -500 +600 -500 +650 -500 +600 -500 +600 -1600 +600 -550 +600 -1550 +650 -500 +600 -500 +600 -550 +600 -500 +600 -500 +650 -500 +600 -500 +600 -500 +650 -500 +600 -500 +600 -500 +650 -500 +600 -500 +600 -1600 +600 -1600 +650 -500 +600 -500 +600 -500 +650 -500 +600 -500 +600 -500 +650 -1550 +650 -1550 +650 -1550 +650 -500 +600 -500 +600 -550 +600 -500 +600 -127976
```

AllAboutEE, 2016-12-29:
Hi, I'm not user of this library but I know a little bit about NEC. The code doesn't look like Extended NEC (16 bit address) or regular NEC (8-bit address).  If it was either of these two protocols the signal (last two bytes) would be the inverse of each other. If it was 8-bit the first two bytes (from left to write), the address bytes, would also be the inverse of each other.  So, maybe your controller is not really NEC? even though it kind of looks like it because of the first two raw number (the NEC 9ms and 4.5ms pulses).

Also, if we decoded the entire raw data the NEC code would be: 0x1020304050002070, which means you are missing 0x050002070.

crankyoldgit, 2017-03-03:
@AllAboutEE Is correct. This is not an NEC code. It's just been mis-decoded/detected as such.
NEC codes are 32bit. The rawData[131] code is correct, and way bigger than 32bit. Before anyone asks, sendNEC only send (up to) 32 bit.

closed 2017-03-03 by crankyoldgit