crankyoldgit/IRremoteESP8266 #43: Issue sending RC6 36bit code

nonflammable, 2016-09-02:
MCE Remote dump:
`
Encoding  : RC6
Code      : 800F8416 (36 bits)
Timing[65]: 
     +2650, - 850     + 450, - 450     + 400, - 450     + 450, - 850
     + 450, - 900     +1300, - 900     + 450, - 400     + 450, - 450
     + 450, - 400     + 450, - 450     + 400, - 450     + 450, - 450
     + 400, - 450     + 450, - 450     + 450, - 400     + 450, - 450
     + 850, - 450     + 450, - 450     + 400, - 450     + 450, - 450
     + 450, - 850     + 400, - 450     + 450, - 450     + 400, - 450
     + 900, - 850     + 450, - 450     + 400, - 450     + 450, - 450
     + 400, - 450     + 900, - 850     + 900, - 450     + 400, - 900
     + 450
unsigned int  rawData[65] = {2650,850, 450,450, 400,450, 450,850, 450,900, 1300,900, 450,400, 450,450, 450,400, 450,450, 400,450, 450,450, 400,450, 450,450, 450,400, 450,450, 850,450, 450,450, 400,450, 450,450, 450,850, 400,450, 450,450, 400,450, 900,850, 450,450, 400,450, 450,450, 400,450, 900,850, 900,450, 400,900, 450};  // RC6 800F8416
unsigned int  data = 0x800F8416;
`
Output from ESPEasy IRTX plugin
http://192.168.1.xxx/control?cmd=IRSEND,RC6,800F8416,36
`
Encoding  : RC6
Code      : 0 (36 bits)
Timing[73]: 
     +2700, - 900     + 450, - 900     + 900, - 450     + 450, -1300
     + 900, - 450     + 450, - 450     + 450, - 450     + 450, - 450
     + 450, - 450     + 450, - 450     + 450, - 450     + 450, - 450
     + 450, - 450     + 450, - 450     + 450, - 450     + 450, - 450
     + 450, - 450     + 450, - 400     + 500, - 400     + 500, - 450
     + 450, - 400     + 500, - 400     + 500, - 450     + 450, - 450
     + 450, - 400     + 500, - 400     + 500, - 400     + 500, - 400
     + 500, - 400     + 500, - 400     + 500, - 450     + 450, - 400
     + 450, - 450     + 450, - 450     + 450, - 450     + 450, - 450
     + 450
unsigned int  rawData[73] = {2700,900, 450,900, 900,450, 450,1300, 900,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,450, 450,400, 500,400, 500,450, 450,400, 500,400, 500,450, 450,450, 450,400, 500,400, 500,400, 500,400, 500,400, 500,400, 500,450, 450,400, 450,450, 450,450, 450,450, 450,450, 450};  // RC6 0
unsigned int  data = 0x0;
`

crankyoldgit, 2017-03-03:
Um, just FYI. The RC6 protocol isn't 36 bits long, according to the specs I can find.
That's not to say you technically couldn't send 36 bits. The implementation here, and in downstream IRremote (arduino) library only supports up to 32 bits. (unsigned long).

Do you have a device that uses RC6 that requires more than 32 bits?
e.g. [Xbox](http://www.righto.com/2010/12/64-bit-rc6-codes-arduino-and-xbox.html)

nonflammable, 2017-03-05:
[sample remote dump.txt](https://github.com/markszabo/IRremoteESP8266/files/819717/sample.remote.dump.txt)

I have Asus EB1501 MCE remote. Under the backside battery cover it says RC6.
Dump of Power button shows "Encoding : RC6 Code : 800F840C (36 bits)"
but when I try to send it with ESPEasy (IRSEND,RC6,800F840C,36) it is sending zero code "Encoding : RC6 Code : 0 (36 bits)"

[http://kodi.wiki/view/How-to:Install_Ubuntu_and_XBMC_on_Asus_EeeBox_PC_EB1501#ASUS_Remote_Control](Remote picture)

crankyoldgit, 2017-03-06:
Okay, I'll try to put something together to support > 32bit RC6 codes soon. I don't have any RC6 devices to check/test with so, I'll have to rely on yourself for testing. We'll only be able to speak for the IRremote8266 library naturally, how ESPEasy uses/calls us is out of scope.

crankyoldgit, 2017-03-06:
@nonflammable Can you try out the [rc6](https://github.com/markszabo/IRremoteESP8266/tree/rc6) branch and see if that works any better? I've taken a stab at getting it to support up to 64 bit codes by converting the obvious things to long longs.

nonflammable, 2017-03-06:
Unfortunately, it seems that stopped working completely ... but RC6 only. Other protocols are working properly.

crankyoldgit, 2017-03-07:
Thanks for the feedback on that. I'll take a deeper look.

marcosamarinho, 2017-03-16:
I already implemented It with 36 bits , and tested it  on my changes ,
please take a look at  my code .It Just have an additional change related
the caller functions that I implemented additional signature to allow
string , this simplify the New implementation as Will  not need anymore
additional changes at the Future as It can accept larger "numbers", as I
remember by Heart If you changed the signature to long long you need to
change your caller program too and long long not is so easy to work as long
or integer .

crankyoldgit, 2017-03-16:
@marcosamarinho I've had a look at [your implementation](https://github.com/marcosamarinho/IRremoteESP8266/blob/master/ir_RC5_RC6.cpp#L164) and I can't seem to see how that is different from the way I tried implementing it in the 'RC6' branch [here](https://github.com/markszabo/IRremoteESP8266/blob/rc6/IRremoteESP8266.cpp#L307). If you can spot what I'm missing that would be great.

Can you take a look and see why it isn't working for @nonflammable? Or as you intone, maybe it is the way he is calling it. ie. with an 'unsigned long long'.

crankyoldgit, 2017-05-04:
@nonflammable I believe this should now work in the v2.0-dev branch. Can you take a look and give it a test?

crankyoldgit, 2017-05-23:
@nonflammable In the v2.0-dev branch you should be able to make the following call:
`sendRC6(0xC800F740C, RC6_36_BITS);`

That will send the RC-6 36bit Xbox power on message.

Note: as with all RC-5 & RC-6 protocols, you may need to flip the toggle bit when sending.
The toggle bit indicates when a remote's button has been pressed or released.
To help you flip the correct bit, you can use the new toggleRC6(cmd, size) function. e.g.:

`irsend.sendRC6(irsend.toggleRC6(0xC800F740C, RC6_36_BITS), RC6_36_BITS);  // Xbox 360 On code (toggled)`

crankyoldgit, 2017-05-28:
The [first candidate for v2.0](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.0.0-RC0) has been released, which supports RC6 36 bits. So I am going to mark this issue closed

closed 2017-05-28 by crankyoldgit