crankyoldgit/IRremoteESP8266 #44: Decode IR with long IR data

ngminhduong, 2016-09-14:
I trying to send IR signal from my phone (IR blaster) with raw length about : 72
8973,4552,500,631,500,631,500,631,500,631,500,631,500,631,500,631,500,631,500,1736,500,631,500,1736,500,1736,500,1736,500,1736,500,631,500,1736,500,1736,500,631,500,1736,500,631,500,1736,500,631,500,631,500,631,500,631,500,1736,500,631,500,1736,500,631,500,1736,500,1736,500,1736,500,42447,8973,2289,500,500
And i try to decode that signal from my ESP.
But i can not receive correct data.
This is what i recieved.
{"Code": "6E61198F","CodeType": -1,"Raw":"9150,4650,500,700,500,700,500,700,500,700,500,700,500,700,500,700,650,700,500,1800,500,700,500,1800,500,1800,500,1800,500,1900,500,700,650,1800,500,1800,500,750,500,1900,500,700,500,1800,500,700,500,700,500,700,500,750,500,1800,500,700,500,1800,500,700,500,1800,500,1800,600,1800,500","RawLength":68,"Bits":32}
{"Code": "FFFFFFFF","CodeType": 1,"Raw":"9000,2400,500","RawLength":4,"Bits":0}

I don't know how to learn  correct IR signal if that signal is too long.
Sorry because my english not good.

dotorg, 2016-12-14:
If you change RAWBUF at the bottom of IRemoteESP8266.h, you can increase the size of codes it'll pick up. I haven't figured out why, yet, but if you set it higher than 255, it will cause the code to stop working, but you can set it to 255 safely, as far as I can tell. That's enough for 127 bits, or just under 16 bytes.

roidayan, 2017-03-08:
you can change rawbuf to 512 and also need to change rawlen from uint8 to uint16.
as for dumping the code since the array is now pretty big you need to add a call yo yield() inside loops going over the rawbuf in the ir recv dump V2 example.

roidayan, 2017-03-08:
related to #91

closed 2017-03-08 by roidayan