crankyoldgit/IRremoteESP8266 #47: Not able to send raw data using dynamic arrays

akshitgandhi, 2016-10-26:
I'm using this library to send ir data to a device, the raw data is sent to esp via a server and esp receives the raw data in a char array which then I convert to unsigned int array as it is required by irsend.sendraw function. But I'm not able to see any results with this method.Whereas when I hardcode the array and then send that, everything works fine. So can anyone help me? Where I'm going wrong?

GeorgeIoak, 2017-01-16:
I'm not sure if I have the same issue as you but I'm also using irsend.sendRaw and I'm not getting the results. I tested this first with an Arduino code which did work but it doesn't work with my ESP-12E module.

Here's my code:

```
#include <IRremoteESP8266.h>
//#include <IRremoteInt.h>

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include "fauxmoESP.h"

//#include "credentials.h" can use this 
//to set WiFi SSID and password

#define WIFI_SSID "iotech-2.4GHz"
#define WIFI_PASS "Athena&George"
#define SERIAL_BAUDRATE                 115200
#define LED                             16  // LED on Board
#define RELAY_PIN                       4   // 
#define OPen2Pin                        12 // Open Blinds pin
#define Close2Pin                       14 // Close Blinds pin
#define khz                             40 // 40kHz carrier frequency for the Hunter Douglas

IRsend irsend(0); //

unsigned int OPen2Signal[] = {2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050,
                              2800, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 1050, 1050, 1975
                             }; //Open2

unsigned int Close2Signal[] = {2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050,
                               1050, 2800, 2800, 1050, 2800, 1050, 1050, 1050, 2800, 1050, 1975
                              }; //Close the lower blinds

// variables will change:
int butOpen2State = 0;         // variable for reading the pushbutton status
int butClose2State = 0;        // variable for reading the pushbutton status

fauxmoESP fauxmo;

...
void loop() {

  // Since fauxmoESP 2.0 the library uses the "compatibility" mode by
  // default, this means that it uses WiFiUdp class instead of AsyncUDP.
  // The later requires the Arduino Core for ESP8266 staging version
  // whilst the former works fine with current stable 2.3.0 version.
  // But, since it's not "async" anymore we have to manually poll for UDP
  // packets
  fauxmo.handle();

  // read the state of the pushbutton value:
  butOpen2State = digitalRead(OPen2Pin);
  butClose2State = digitalRead(Close2Pin);
  //Serial.println(butOpen2State);

  if (butOpen2State == HIGH) {
    // send IR code:
    irsend.sendRaw(OPen2Signal, sizeof(OPen2Signal) / sizeof(OPen2Signal[0]), khz);
    Serial.println("Open2 Button Pushed");
    digitalWrite(RELAY_PIN, HIGH);
  }

  if (butClose2State == HIGH) {
    // send IR code:
    irsend.sendRaw(Close2Signal, sizeof(Close2Signal) / sizeof(Close2Signal[0]), khz);
    Serial.println("Close2 Button Pushed");
    digitalWrite(RELAY_PIN, LOW);
  }
}
```

For testing I inserted a standard LED and the LED is always ON. I haven't hooked up the scope to check the signals but this is different than what worked on the Arduino setup.

Maybe I'll try hard coding the array size and length and see if that works like it did with you but that doesn't explain why GPIO0 is always high (3.3V).

ptbw, 2017-01-16:
Try using a different gpio rather than 0.  Gpio 0 is use by the ESP8266 and requires special handling...

http://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html

GeorgeIoak, 2017-01-16:
Yes, I've tried 0 and also 4 and I confirmed on the scope that nothing is going out on those pins. I added the code to toggle another pin when the button is pushed and that works so I know that I'm entering that portion of the code.

I can't seem to figure out what's wrong

GeorgeIoak, 2017-01-16:
I got the code to work but I added ```pinMode(4, OUTPUT)``` which makes sense but none of the examples configured the pin so I assumed that was done in the IR code

crankyoldgit, 2017-02-24:
It is done in the code/library, but you need to call the begin() method. From yourexample above:
IRsend irsend(0);

You should have had a irsend.begin(); in your setup() routine.

That was initialising the IR sender on Pin 0. If you wanted pin 4, then use:

IRsend irsend(4);
setup() {
  irsend.begin();
}

loop() {
  ...
}

closed 2017-02-24 by crankyoldgit