crankyoldgit/IRremoteESP8266 #48: Reset Exceptino(0)

eduardogi, 2016-10-27:
I'm using the IRrecvDemo.ino with nodeMCU V1. But when i'm pressing the botton by 2 or more seconds. The esp8266 reboot and say me it:

`Exception (0):
epc1=0x40203324 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40106fb0 0000004e 00000000 3ffe8d50  
3ffffde0:  ffffffff 00000020 3ffe8dfc 00000030  
3ffffdf0:  00000006 00000006 42d60000 00000022  
3ffffe00:  3fffc200 40106f78 3fffc258 4000050c  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 00000000 80000000  
3ffffe30:  20000000 3fff0640 80000000 203fc040  
3ffffe40:  80000000 3fffc6fc 3ffe91a4 3fff0644  
3ffffe50:  000002b4 003fc040 60000600 00000030  
3ffffe60:  7fffffff 3ffe9c2c 3ffe9c2c 00000001  
3ffffe70:  40107014 00000002 00000000 40106ffe  
3ffffe80:  ffffffff 00000020 3ffefe8c 00000000  
3ffffe90:  00000000 00000000 0000001f 401014a1  
3ffffea0:  4000050c 40106f78 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 00000011 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000007 4021b770 3ffe91a4 bfffffff  
3ffffee0:  ffffffff 3ffefe8c 3ffe91a4 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe88c0 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff0604 000002f4 003fc000  
3fffff20:  401023d2 3ffe88b0 3ffe9050 40107450  
3fffff30:  4021b23d 3ffe9050 3ffe88b0 0243c6fd  
3fffff40:  3fff0604 00001000 4021b6d6 00000008  
3fffff50:  4021d658 3ffe91a4 4021b783 3ffe9104  
3fffff60:  3ffe88b0 00676911 3ffe88b0 60000600  
3fffff70:  40203d05 3ffe9104 3ffe88b0 0243b314  
3fffff80:  40203d4a 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe88c8 3fffdad0 3ffee52c 40202b9f  
3fffffa0:  40000f49 0001822d 3fffdab0 40000f49  
<<<stack<<<
`

reynolga, 2016-11-09:
I am having the same issue with the V1 version also. I thought maybe the issue was tied to a couple of different things so I tried

1) Adding a bunch of capacitors (400 uF) to the VCC - GND.
2) Powering from the wall rather than the computer usb.
3) Changing different pins (I thought maybe the interrupts were maybe causing issues?)
4) Stripping everything else out of the code (Even the basic IRreceiveDemo caused this issue)

Here is my output...
Exception (0):
epc1=0x402032ec epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  40106fb0 80af1999 3ffe8e3c 00000030  
3ffffde0:  ffffffff 00000020 40240000 ffffff1a  
3ffffdf0:  00000006 00000006 42d60000 00000022  
3ffffe00:  3fffc200 40106f78 3fffc258 4000050c  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 7c037c02 80000000  
3ffffe30:  20000000 3fff07c0 80000000 203fc180  
3ffffe40:  80000000 3fffc6fc 3ffe8d5c 3fff07c4  
3ffffe50:  00000174 003fc180 60000600 00000030  
3ffffe60:  7fffffff 3ffe9c30 3ffe9c30 00000001  
3ffffe70:  40107014 00000002 00000000 40106ffe  
3ffffe80:  ffffffff 00000020 3ffefecc 00000000  
3ffffe90:  00000000 00000000 0000001f 401014a1  
3ffffea0:  4000050c 40106f78 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 0000001c ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000013 4021b740 3ffe8d5c 00000000  
3ffffee0:  ffffffff 3ffe91e4 3ffe8d5c 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffe8900 00000030  
3fffff00:  00000000 400042db 00000064 60000600  
3fffff10:  40004b31 3fff0644 000002f4 003fc000  
3fffff20:  401023d2 3ffe88f0 3ffe9090 40107450  
3fffff30:  4021b20d 3ffe9090 3ffe88f0 00deea8a  
3fffff40:  3fff0644 00001000 4021b6a6 00000008  
3fffff50:  401027c4 00000000 4021b753 3ffe9144  
3fffff60:  3ffe88f0 004577a7 3ffe88f0 60000600  
3fffff70:  40203ccd 3ffe9144 3ffe88f0 00decaea  
3fffff80:  40203d12 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffe8908 3fffdad0 3ffee56c 40202b93  
3fffffa0:  40000f49 00017403 3fffdab0 40000f49  
<<<stack<<<

`/*
 * IRremoteESP8266: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * An IR detector/demodulator must be connected to the input RECV_PIN.
 * Version 0.1 Sept, 2015
 * Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009, Copyright 2009 Ken Shirriff, http://arcfn.com
 */

#include <IRremoteESP8266.h>

int RECV_PIN = 2; //an IR detector/demodulatord is connected to GPIO pin 2

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}`

reynolga, 2016-11-09:
From reading other threads here.. it seems the "delay()" command causes issues with the interrupts. I removed that from my code and it is no longer crashing. Hope it helps!

crankyoldgit, 2017-02-24:
The delay() in IRrecvDemo.ino should not be causing this issue. This sounds more like bad power.
Is this issue still happening?

crankyoldgit, 2017-03-18:
No response after 3 weeks. Closing this issue out.

closed 2017-03-18 by crankyoldgit

eduardogi, 2017-03-18:
Sorry, I've forgotten response it. It didn't fix the problem . I've tried a lot of things but  in the end, I removed IR from my project.

Regards.

the0s, 2018-02-09:
try this: 
#define CAPTURE_BUFFER_SIZE 1024
#define TIMEOUT 15U 
#define RECV_PIN 12
IRrecv(RECV_PIN,CAPTURE_BUFFER_SIZE, TIMEOUT, true);

crankyoldgit, 2018-02-10:
I think just using a new (updated) version in the last year would probably also fix their issue(s).