crankyoldgit/IRremoteESP8266 #57: Receiving is not working after irrecv.disableIRIn() and irrecv.enableIRIn()

lucky62, 2016-12-10:
Receiving not working after disabling and re-enabling receiver..

I have attached both the IR decoder for receiving IR codes and IR LED to send IR codes.
When I send the code then it is also received back by receiver. To avoid this I am trying to disable receiver before the sending and enable receiver after sending. 
```c
    irrecv.disableIRIn();
    irsend.sendRC5(code, 12);
    irrecv.enableIRIn();
```
after this I cannot receive any IR code.

sticilface, 2017-01-19:
I have found this too.  

I'm trying to work out how the lib is working under the hood

enable calls this
```cpp
void IRrecv::enableIRIn() {
	
  // initialize state machine variables
  irparams.rcvstate = STATE_IDLE;
  irparams.rawlen = 0;

  // Initialize timer
  os_timer_disarm(&timer);
  os_timer_setfn(&timer, (os_timer_func_t *)read_timeout, &timer);
  
  // Attach Interrupt
  attachInterrupt(irparams.recvpin, gpio_intr, CHANGE);

}
```

disable calls this

```cpp
void IRrecv::disableIRIn() {
  //irReadTimer.stop();
  //os_timer_disarm(&irReadTimer);   
   ETS_INTR_LOCK();
   ETS_GPIO_INTR_DISABLE();
}
```

it does not make use of this 
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/core_esp8266_wiring_digital.c#L187

if i change the disable to this
```cpp
detachInterrupt(irparams.recvpin);
```

then enabling and disabling  works :)  can make a pull if desired?

markszabo, 2017-01-22:
Pull request would be nice :)

crankyoldgit, 2017-02-24:
@sticilface Could you please create and send a Pull Request?

PR #83 (2017-03-03): Fix disableIRIn disabling interrupts and enableIRIn not enabling
In enableIRIn we initialize a timer and attach an interrupt.
In disableIRIn we need to disable the timer and detach the interrupt.

Fixes #57
commit b0e72f1a (2017-03-03): Fix disableIRIn disabling interrupts and enableIRIn not enabling

In enableIRIn we initialize a timer and attach an interrupt.
In disableIRIn we need to disable the timer and detach the interrupt.

Fixes #57
closed 2017-03-03 by crankyoldgit