crankyoldgit/IRremoteESP8266 #58: IRGCTCPServer example questions

pallazola, 2016-12-12:
How is the IRGCTCPServer example supposed to work?
I'm a bit new to ESP8266 and I had a few questions about this example:

- Is the example missing an explicit command to connect to WiFi? (something like WiFi.begin?)
- How are the commands sent to the ESP8266 and How should they be formatted? 
  - Should a HTTP GET, PUT or POST request be made to port 4998???
- Is this example setup to create its own WiFi network?
  - should I be using that exclusively?
I'm very excited about this project!!! I am looking forward to being able to trigger IR commands from the terminal if that is possible. Thanks!!!
-Mike

crankyoldgit, 2017-02-24:
Hey Mike, the last time I played with that example, I think I had to issue the command 'telnet <ip_of_esp8266> 4998'. or 'nc <ip_of_esp8266> 4998'.
Looking at the source code, it appears it opens a TCP port on 4998 and listens for GC codes there.
e.g.
send '38000,1,1,170,170,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,63,20,63,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,20,20,20,20,20,20,20,20,20,20,63,20,20,20,63,20,63,20,63,20,63,20,63,20,63,20,1798\r\n' to tcp/4998 on the esp8266 in question.

closed 2017-02-24 by crankyoldgit

crankyoldgit, 2018-06-01:
Really fixed in #432