crankyoldgit/IRremoteESP8266 #59: Unknown encoding find in my TV remote.

Joypacharya, 2016-12-19:
Actually i am using your library.Its works great in decode some of Fan and AC remote.
Whenever i use TV remote.it all time display unknown encoding.Same remote i used decode my raw code via IRremote.h library in my arduino uno board.So, how to resolve this problem?
If i use Arduino uno board then result should be,
Decoded RC6: 1000C (20 bits)  <--- Power button pressed

If i use ESP8266 nodemcu then result should be,
Unknown encoding: F5AD0EC0 (32 bits) 

So, please help me resolve this problem.
Even i put double slash at  //case AIWA_RC_T501: Serial.print("AIWA_RC_T501");  break ;
file than i compare sketch.So, check this problem.

crankyoldgit, 2017-02-24:
You might want to try again with an up to date copy of this library. I've also noticed that decoding IR codes is a very challenging game and you get different results from different libraries because they are implemented differently.

The safest bet is to look at and use the 'raw' codes or look up the 'global cache' (GC) codes and use that interface to send a compatible signal or when trying to decipher input.

The AIWA thing should be fixed now.

crankyoldgit, 2017-03-20:
Can you please provide the _rawData_ from the ESP8266 device when using this library? That way we might be able to see what is going on and why it isn't decoding correctly.

crankyoldgit, 2017-03-28:
Ping. Without rawData output, I'm not sure I can help you.

crankyoldgit, 2017-04-08:
Closing this issue due to no response from the reporter. Feel free to reopen if/when you have the additional data requested.

closed 2017-04-08 by crankyoldgit