crankyoldgit/IRremoteESP8266 #68: Not decoding Daikin AC ARC479A5 Remote IR Codes.

kjhaSSI, 2017-02-07:
Tried "IRrecvDump" example for Daikin AC Remote ARC479A5. It returns following response. Surprisingly along with Unknown type, it returns decode type "7(Panasonic)" data as well.

Decode Type: -1
Unknown encoding: B3E2CB06 (32 bits)
Raw (12): 500 -450 450 -450 450 -450 450 -450 500 -400 500 
E400000C
Decode Type: 7
E400000C (48 bits)
Raw (100): 3500 -1750 500 -1300 500 -450 450 -450 500 -450 450 -1300 500 -450 500 -450 450 -400 500 -450 500 -1300 500 -350 550 -1300 500 -1300 500 -450 450 -1300 500 -1300 500 -1300 500 -1300 500 -1300 500 -450 450 -450 450 -1300 500 -450 500 -450 450 -450 500 -450 450 -400 550 -450 450 -450 450 -350 550 -450 450 -450 500 -450 500 -350 550 -450 500 -400 500 -450 450 -450 500 -450 450 -450 450 -450 450 -450 500 -400 500 -450 450 -1300 500 -1250 500 -400 500 -450 450

crankyoldgit, 2017-03-30:
I had a look at the decoding code. There is a routine for decoding Daikin A/C remotes, however it is not included in the default decoding list. I suspect it hasn't been added for a reason. A number of A/C IR code schemes split up the command into multiple separate-IR-like sequences. The way the library handles the decodes is for the typical use cases of simple IR commands. If you want it to work, you'll probably need to add the decodeDaikin() call in the appropriate place **and** play around with the routines that capture the incoming IR sequence so it can span across large gaps in transmission to collect the entire code at once. In the case of the Daikin protocol, there is a 29+msec gap in the middle of the protocol.

crankyoldgit, 2017-06-08:
Friendly ping. Any update/feedback/more info?

crankyoldgit, 2017-10-14:
Closing this due to inactivity. Please re-open if have an update/feedback/more info etc.

closed 2017-10-14 by crankyoldgit