crankyoldgit/IRremoteESP8266 #70: Support for Mitsubishi AC?

mihalski, 2017-02-19:
Hi,

Would it be possible to enhance this library with support for Mitsubishi air conditioning systems? Code is available here: https://github.com/r45635/HVAC-IR-Control

Regards,
Michal

crankyoldgit, 2017-02-24:
Hey Michal,
 I'll take a look at it, but I don't have the Mitsubishi unit to test/confirm it works with. Do you have one per chance? ie. Are you willing to do some testing?
Side note: I do have a Mitsubishi Heavy Industries A/C unit, but I don't think they are the same as a standard 'Mitsubishi' brand A/C.

mihalski, 2017-02-24:
Thank you!

I have 3 such units and I did test with a random IR led (that I don't have the specs to and had to use without a resistor to get any results) and it worked over a short distance.

Here's hoping once my Vishay leds come in I'll be able to get decent results. VERY happy to do any testing.

Regards,
Michal

crankyoldgit, 2017-02-24:
FWIW, make sure you are not driving the IR LED using the current from the output pin. The current is too low and doesn't give a good result.
i.e. Use the GPIO pin instead to turn on a transistor that drives the IR LED from Vcc (3.3 or 5v).

You can often drive the LEDs overcurrent and overvoltage for the short pulses used in IR signalling.

mihalski, 2017-02-24:
I tried that initially but I must have been doing something wrong as I was getting a lot less current than driving directly.. Not sure which GPIO I was using.. Will dig it out again tomorrow and measure everything and post how I've connected things..

I have two Broadlink RM mini 3 devices so I may strip one for parts while I wait for my components.

mihalski, 2017-02-25:
Basically I have a Wemos with driving an IR led via GPIO4 using a PN100 transistor connected to the 5V of the Wemos.

The specs for the LED MIGHT be as stated here:

https://www.jaycar.com.au/5mm-infrared-transmitting-led/p/ZD1945

Unclear since apparently they use various sources.

As is it works OK directly pointed at the AC at about 3 metres.. The original remote bounces f all the walls without a problem.

crankyoldgit, 2017-02-27:
There appears to be no explicit licence set for their code, and there is a copyright notice on the code. So without explicit permission or a change in their licence state, I'm afraid I can't use their code. I've logged an issue with them to ask them to address this.

Fingers crossed.
If we don't hear anything from them, I don't like your chances unless you want to work it out for yourself, or someone else gets around to it. I have no immediate need to access my Mitsub Heavy Industries unit, so it's low on my personal priorities.

crankyoldgit, 2017-02-27:
Update: Looks like they are going to update the license. Stay tuned.

mihalski, 2017-02-28:
That's fantastic news :)
Thank you for your effort.

Regards,
Michal

crankyoldgit, 2017-02-28:
@mihalski I've done an initial pass of adding Mitsubishi A/C controls to my [personal fork](https://github.com/crankyoldgit/IRremoteESP8266) of this library. Completely untested. Compiles. Not run it yet. Try the example TurnOnMitsubishiAC.ino. Let me know. I'm not expecting much.
I'm done for the night. Probably another day or two before I get back to it again.

Lets move the discussion of testing to my repo to keep the chatter off this distro while we work it out the bugs/issues.

mihalski, 2017-02-28:
@crankyoldgit It seems to work perfectly!

Now it's hard to tell for sure that all the signals are as they should, but the AC responds and seems to act in accordance with the provided parameters.

Thank you very much for your work thus far. I tried to make this reply in your repo, but you seem to be missing the Issues tab.

Refards,
Michal

crankyoldgit, 2017-03-01:
Closing this for now. We can add a new issue for any new issues with it.

closed 2017-03-01 by crankyoldgit