crankyoldgit/IRremoteESP8266 #71: Looking for maintaners/collaborators

markszabo, 2017-02-19:
Dear @crankyoldgit, @sebastienwarin, @sillyfrog, @pintomax, @roidayan, @bakrus, @renno-bih,

thank you for your great help with this project. Recently I have almost no time to work on this library, while more and more people are interested with PRs and issues more and more often. So if some of you have time and willing to help / take over the project, I would be happy to give you push access to this repo. Alternatively if you want to do it on your own github, I can deprecate mine and place a link to the top of the readme. However given the good number of people already using this repo, I would prefer the first solution.

Best regards,
Mark

crankyoldgit, 2017-02-19:
Happy to help out in any way.

markszabo, 2017-02-19:
@crankyoldgit: I have added you as collaborator. Could you accept it [here](https://github.com/markszabo/IRremoteESP8266/invitations)? :)

roidayan, 2017-02-19:
I'll be happy to help as well

crankyoldgit, 2017-02-20:
Done.

sillyfrog, 2017-02-20:
I would like to say I would, but I know I won't be able to be of much value practically. However I will continue to send through any PR that I do get to work on.
Thanks!

markszabo, 2017-02-20:
Okay, thank you for all of you, I have added you as collaborators as well.

pintomax, 2017-02-20:
I'll be happy to help as well.

Massimiliano

marcosamarinho, 2017-02-24:
Mark 
I can help too , if possible , guys  please validate the work I did at the pullrequest  on many days and nights that I merged with zt30 as requested by Mark , clean up the whole code  and fixed all issues I could find at that time , note that many of them are reported at the Issues yet  , this  allows all  my home devices working now  along EspEasy project  that I wish to share them a new  version. 

Marcos Marinho

crankyoldgit, 2017-02-25:
I'm in favor of @marcosamarinho being added. His PR is something I thought of doing to get this closer to the Arduino IRremote library. It's a beast of a change.

markszabo, 2017-02-25:
Sorry, that I forgot you, Marcos, I've added you as collaborator. Thank you for all your great work with this project :)

marcosamarinho, 2017-03-01:
Thanks , Mark I just want see my changes working along the project ! .

 , I received the invitation so  not managed to accept it . do you know if
is there something I  or you need to do ?

markszabo, 2017-03-01:
As far as I can see you have already accepted it, so nothing else to do with that. Thank you for all your help

closed 2017-03-18 by crankyoldgit