crankyoldgit/IRremoteESP8266 #75: IRrecvDumpV2 does not compile

witnessmenow, 2017-02-19:
case SANYO:        Serial.print("SANYO");         break ;
    case MITSUBISHI:   Serial.print("MITSUBISHI");    break ;
    case SAMSUNG:      Serial.print("SAMSUNG");       break ;
    case LG:           Serial.print("LG");            break ;
    case WHYNTER:      Serial.print("WHYNTER");       break ;
//    case AIWA_RC_T501: Serial.print("AIWA_RC_T501");  break ;

Sanyo and Aiwa_RC_T501 have the same value which throws a complication error

crankyoldgit, 2017-02-24:
Thanks for the report. I've submitted a fix for it. It compiles now for me. @witnessmenow Can you please confirm before I close this?

marcosamarinho, 2017-03-01:
Hi @ crankyoldgit, Just for let you know ,  I verified that AIWA_RC_T501 it is a so specific , hardcoded  implementation of SANYO and  this really can be removed from this project with no issues as it can be send/received as SANYO  protocol , at my pull request it is already commented on the definitions .

crankyoldgit, 2017-03-01:
Thanks for the confirmation.

closed 2017-03-01 by crankyoldgit