crankyoldgit/IRremoteESP8266 #88: Fujitsu ir codes

ericonetto, 2017-03-05:
Guys, did someone have ever tried to dump the IR codes and control any Fujitsu air conditioner? I couldn't manage to do that.

roidayan, 2017-03-05:
You didn't get anything dump? Or sending code didn't work?

ericonetto, 2017-03-05:
I could dump, but I think the array of codes are too large. When I send it
the air conditioner doesn't understand or they are at a frequency rate I
haven't tried yet.
Air conditioner remotes always try to sync the device to whatever is
showing in the IR remote screen. To do that they always send all the
settings again.

Btw, I've tried the lib with LG air conditioner and work super fine.

Thanks for your reply

roidayan, 2017-03-05:
You can try to increase rawbuf length from 100 to something bigger.
You can try to use this pull request https://github.com/markszabo/IRremoteESP8266/pull/86
Then using recv dump V2 will print if there is a need for a larger rawbuf.

crankyoldgit, 2017-03-23:
FWIW, I've started implementing the Fuji A/C protocol. It probably won't be ready for a few days at least.
It's odd. It has short commands (6 bytes long) and long (15 bytes).

Redferne, 2017-05-29:
Any news on Fujitsu? Great work on this library. Thank you for the effort!

crankyoldgit, 2017-05-30:
I started some work on it but a) version 2.0 work was higher priority, and
b) summer ended for me so my need to control my device naturally lowered.

rowdy15, 2017-07-01:
I just want to write that I am keen to see the fujitsu codes added as well, as I have the Fujitsu ASTA24LCCSM A/C, which I'm pretty confident will respond to the Fuji A/C protocol.

I see it is working on an actual arduino from [this](https://github.com/ToniA/arduino-heatpumpir) library. But I really like the idea of using just the esp8266 for wifi and logic.

I can currently control the air conditioner (in a limited way) through the RAW code ( the RAWBUFFER needs to be changed to be at least 259 in my case ( I changed my RAWBUFFER to be 500U)),  but I would love to be able use the fuji library.

jonnygraham, 2017-07-01:
I just finished code to add Fujitsu support and have submitted a pull request.

rowdy15, 2017-07-01:
Oh that was quick.  Thanks.

crankyoldgit, 2017-07-01:
Thanks @jonnygraham 
I was going to have another crack at this this week. You saved me a tonne of effort.

crankyoldgit, 2017-07-02:
FYI, @jonnygraham 's PR #259 is pretty much done/ready. I'll be merging it soon, and making a new release of the library. In the meantime, if some of you chaps/chapettes would care to pound on the code in real-life conditions, that would be awesome.

Expecting the merge/new version in a day or so.

crankyoldgit, 2017-07-03:
FYI [v2.1.0-RC0](https://github.com/markszabo/IRremoteESP8266/releases/tag/v2.1.0-RC0) release candidate is available. Please test etc.

crankyoldgit, 2017-07-04:
[v2.1.0](https://github.com/markszabo/IRremoteESP8266/tree/v2.1.0) has been officially released. This includes support for sending Fujitsu A/C codes.

Marking this closed. Please open a new issue if there are problems with it.

closed 2017-07-04 by crankyoldgit