crankyoldgit/IRremoteESP8266 #89: decode NEC issue?

mhoffma, 2017-03-05:
Not sure if this is my issue but in debugging the bit patterns coming back from a nec IR transmitter the state machines was not recognizing the codes.

[image]

To collect the capture and have the code correctly from the device I needed to make the following adjustments to the nec decoder which is probably wrong and is clearly just a hack.  1st I don't skip the 9ms (9000) its in the data and I'm not sure how this code base gets around that fact.  2nd the length has a small adjustment of 4 on the bounds of the received data and I'm not sure why we care for everything I tried it always caused the code to be rejected by your algorithm.

Do you want the two changes as a pull request or is it clear to just look at the diff with my debug telemetry?

@@ -556,12 +556,13 @@ int IRrecv::decode(decode_results *results) {                                                           
 // NECs have a repeat only 4 items long                                                                                      
 long IRrecv::decodeNEC(decode_results *results) {                                                                            
   long data = 0;                                                                                                             
-  int offset = 1; // Skip initial space                                                                                      
+  int offset = 0; // Skip initial space                                                                                      
   // Initial mark                                                                                                            
   if (!MATCH_MARK(results->rawbuf[offset], NEC_HDR_MARK)) {                                                                  
     return ERR;                                                                                                              
   }                                                                                                                          
   offset++;                                                                                                                  
:                                                                                                                             
+  Serial.println("got nec mark");                                                                                            
   // Check for repeat                                                                                                        
   if (irparams.rawlen == 4 &&                                                                                                
     MATCH_SPACE(results->rawbuf[offset], NEC_RPT_SPACE) &&                                                                   
@@ -571,9 +572,12 @@ long IRrecv::decodeNEC(decode_results *results) {     results->decode_type = NEC;                        
     return DECODED;                                                                                                          
   }                                                                                                                          
-  if (irparams.rawlen < 2 * NEC_BITS + 4) {                                                                                  
+  Serial.println(irparams.rawlen);                                                                                           
+  Serial.println(2*NEC_BITS+4);                                                                                              
+  if (irparams.rawlen < 2 * NEC_BITS ) { /* 4? */                                                                            
     return ERR;                                                                                                              
   }                                                                                                                          
+  Serial.println("$ got nec mark (len)");                                                                                    
   // Initial space                                                                                                           
   if (!MATCH_SPACE(results->rawbuf[offset], NEC_HDR_SPACE)) {                                                                
     return ERR;

crankyoldgit, 2017-03-05:
Pull requests are preferred.

mhoffma, 2017-03-05:
Quick question how is this tested other than usage?  Do you have a specific test setup?

This describes exactly my Ir setup:

https://learn.sparkfun.com/tutorials/ir-communication

The sequence matches perfectly with the expected codes with my two changes.  I'm just confused on why the original code skipped the initial signaling and imposes additional 4 samples in the buffer to make it a valid code.

crankyoldgit, 2017-03-06:
Can you please edit your initial comment and mark the diff/code as a code block ("insert code" in the web UI) so that it is properly formatted/readable?

As for testing, sadly, we don't have unit tests for this library. It's something I want this library to move too.
Thus, testing is mostly done by users, and the person submitting the change. :-/

crankyoldgit, 2017-04-10:
Per comments in your pull request, I can't find anything immediately wrong. I've looked and looked, and can't see any issues. Real world testing seems to indicate the decoding is working fine for NEC codes.

Feel free to re-open if you can provide a good example/test case etc.

closed 2017-04-11 by crankyoldgit