crankyoldgit/IRremoteESP8266 #91: Support RAWLEN of > 256

crankyoldgit, 2017-03-06:
There are several places where RAWLEN is assumed to be a uint8_t.
Some IR codes (*cough* Kelvinator *cough*) come perilously close to that limit. It'd be nice to support up to 512 or something.

crankyoldgit, 2017-03-06:
This is a low priority.

roidayan, 2017-03-06:
I played with this a little. increased rawbuf to 512 and rawlen to uint16. during dump prints of recv dump V2 sketch I got Soft WDT reset. dont know why yet. I stopped currently .

crankyoldgit, 2017-03-07:
That sounds like what was happening when I tried last time. I suspect we need to feed the wdt on these long decodes.

roidayan, 2017-03-07:
I did some testing with array size 512. just having a sketch going over 512 items in a loop to print it fires WDT. Adding yield() solves this. even limiting yield() to every 100 samples is fine.
So when we update to 512 samples we also need to update the examples with something like:

if (results->rawlen % 100)
  yield();

this should be added to every loop of results->rawbuf when rawbuf is big. (512 is considered big)

roidayan, 2017-03-08:
related #26

crankyoldgit, 2017-04-18:
This is pretty much done now. It will be fully resolved with the release of v2.0

crankyoldgit, 2017-05-20:
Appears to be working in the v2.0 tree.
Marking resolved

closed 2017-05-20 by crankyoldgit