crankyoldgit/IRremoteESP8266 #93: Multiple Receivers Support

kolsan, 2017-03-06:
Good morning
I have used the IrRemote succesfully, but due to my project requirements, i need to add several ir receivers. To got that, i have modified the included sample in this way:

`#include <IRremoteESP8266.h>
int RECV_PIN = 2; //an IR detector/demodulatord is connected to GPIO pin 2
int RECV_PIN2 = 3; //an IR detector/demodulatord is connected to GPIO pin 3
IRrecv irrecv(RECV_PIN);
IRrecv irrecv(RECV_PIN2);
decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
 irrecv2.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.print ("Ir Receiver 1: ");
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }

  if (irrecv2.decode(&results)) {
    Serial.print ("Ir Receiver 2: ");
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}`

´The program simply does not work, and not show any info. The receivers working fine if we setup the program with only one receiver.
It is possible to implement multiple receivers support?

Thanks in advance!

crankyoldgit, 2017-03-06:
I'm surprised it compiles. Try the following.

Your 5th line:
`IRrecv2 irrecv(RECV_PIN2);`
should probably be:
`IRrecv irrecv2(RECV_PIN2);`

crankyoldgit, 2017-03-06:
Have you also tried it with a single receiver separately on both pins?
i.e. That the pin you've requested works as expected.

kolsan, 2017-03-06:
Ok, thank you very much for your comments.  
 i will try this afternoon and i will post the results.

kolsan, 2017-03-07:
Hello, the modification provided by you in the code does not work (the same behaviour), but as you comment, using the same pin in several receivers works , so that´s all i need :)
Thank you very much for your support

crankyoldgit, 2017-03-07:
Okay, but just be aware that you could get mixed results (i.e. garbage) if each decoder tries to talk to the ESP during the same period, if you use a common input pin.

I consider the interrupt handler not working on multiple pins simultaneously a bug or at least something we should say isn't supported.

@roidayan Could you take a look at this as you are way more familiar with the interrupt/decoding code than I am?

crankyoldgit, 2017-03-11:
Thinking about this some more, I think we are invoking the class wrong in the examples.
e.g.
we should be using:
`IRrecv irrecv = IRrecv(RECV_PIN);`
to instantiate the object, thus it should then handle separate input pins. Correct?

crankyoldgit, 2017-04-11:
I think I found the issue. The interrupt routines are outside the object, and so is the storage & timer for the interrupts. So, currently, you can only have a single IR receiver.

On the plus side, I know what to do to fix it now. But it will be a wee while before I can release a version that does that.

crankyoldgit, 2017-07-14:
I've seriously looked into this. Given the nature of the Function Pointers need for returning from an Interrupt, and the use of a class object. I can't see any good way to solve this issue. i.e. It's beyond my C++ programming ability. We will just have to live with one receiver object per ESP device.

closed 2017-07-14 by crankyoldgit