crankyoldgit/IRremoteESP8266 #99: Arduino IDE update confusion...

timfield, 2017-03-10:
Arduino IDE informs me "Updates available for some of your libraries". I go to Manage Libraries -> Updatable and it tells me there is an update to IRremoteESP8266 available. I am using this (markszabo) library which is shown as v0.2.1. The supposed "update" is v1.0 of 2 yr old "sebastienwarin" fork from this library.

Am I doing something wrong? Is there some way to force Arduino IDE to only look for updates to this library and ignore the other? Thanks!!

crankyoldgit, 2017-03-10:
v0.2.1 is the most current version. Yeah, the version numbering is screwed up.
I'll bump the version number to something higher than v1.0 to clear up any confusion.

crankyoldgit, 2017-03-10:
I've updated the version number to be higher than the highest old one, and created a new release. Hopefully this will address the issue.
Not sure how long it will take to filter through to the Arduino IDE library management stuff.
I'm going to close this issue for now. Please feel free to re-open it if doesn't appear to have worked or filtered through within 24-48 hrs.

closed 2017-03-10 by crankyoldgit

timfield, 2017-03-10:
Thank you - that did it! Just been a low-key irritant for awhile! And thanks for all the work you are doing for this library!!