crankyoldgit/IRremoteESP8266 #107: sending different codes?

kiralikbeyin, 2017-03-14:
I get on and off with one of my nodemcu (by remote controller)
```
//Encoding  : NEC
//Code      : DFF807 (32 bits)
//Timing[67]: 
//     +9050, -4500     + 600, - 550     + 650, - 550     + 650, - 550
//     + 600, - 550     + 650, - 550     + 600, - 550     + 600, - 550
//     + 600, - 550     + 600, -1650     + 650, -1700     + 600, - 550
//     + 600, -1700     + 600, -1700     + 600, -1650     + 650, -1700
//     + 600, -1700     + 600, -1650     + 600, -1700     + 600, -1700
//     + 600, -1700     + 600, -1700     + 600, - 550     + 600, - 550
//     + 600, - 550     + 600, - 550     + 650, - 550     + 600, - 550
//     + 600, - 550     + 600, - 550     + 600, -1650     + 650, -1700
//     + 600, -1700     + 600
//unsigned int  rawData[67] = {9050,4500, 600,550, 650,550, 650,550, 600,550, 650,550, 600,550, 600,550, 600,550, 600,1650, 650,1700, 600,550, 600,1700, 600,1700, 600,1650, 650,1700, 600,1700, 600,1650, 600,1700, 600,1700, 600,1700, 600,1700, 600,550, 600,550, 600,550, 600,550, 650,550, 600,550, 600,550, 600,550, 600,1650, 650,1700, 600,1700, 600};  // NEC DFF807
//unsigned int  data = 0xDFF807;
//
//Encoding  : NEC
//Code      : DF7887 (32 bits)
//Timing[67]: 
//     +9050, -4500     + 600, - 550     + 600, - 550     + 600, - 550
//     + 600, - 550     + 600, - 550     + 600, - 550     + 650, - 550
//     + 600, - 550     + 600, -1700     + 600, -1650     + 600, - 550
//     + 600, -1650     + 650, -1700     + 600, -1700     + 600, -1650
//     + 600, -1700     + 600, - 550     + 650, -1700     + 600, -1700
//     + 600, -1650     + 600, -1700     + 600, - 550     + 650, - 550
//     + 600, - 550     + 600, -1700     + 600, - 550     + 600, - 550
//     + 600, - 550     + 600, - 550     + 600, -1700     + 600, -1700
//     + 600, -1700     + 600
//unsigned int  rawData[67] = {9050,4500, 600,550, 600,550, 600,550, 600,550, 600,550, 600,550, 650,550, 600,550, 600,1700, 600,1650, 600,550, 600,1650, 650,1700, 600,1700, 600,1650, 600,1700, 600,550, 650,1700, 600,1700, 600,1650, 600,1700, 600,550, 650,550, 600,550, 600,1700, 600,550, 600,550, 600,550, 600,550, 600,1700, 600,1700, 600,1700, 600};  // NEC DF7887
//unsigned int  data = 0xDF7887;
```

sender code (other nodemcu) :
```
#include <IRremoteESP8266.h>

IRsend irsend(0); //an IR led is connected to GPIO pin 0
unsigned int  ON[67] = {9100, 4450, 650, 500, 700, 500, 600, 550, 650, 500, 700, 500, 600, 550, 650, 500, 700, 500, 600, 1700, 650, 1650, 650, 500, 700, 1650, 650, 1650, 600, 1650, 700, 1650, 650, 1650, 600, 1650, 700, 1650, 650, 1650, 600, 1650, 700, 1650, 650, 500, 700, 500, 650, 550, 650, 500, 700, 500, 600, 550, 650, 500, 700, 500, 650, 1650, 700, 1650, 650, 1650, 600}; // NEC DFF807
unsigned int  OFF[67] = {9100, 4450, 600, 600, 550, 550, 650, 550, 650, 550, 600, 550, 600, 550, 600, 600, 550, 550, 650, 1700, 600, 1700, 600, 600, 550, 1700, 650, 1650, 600, 1700, 600, 1700, 650, 1650, 600, 600, 600, 1650, 650, 1700, 600, 1700, 600, 1650, 650, 600, 600, 550, 600, 550, 600, 1700, 600, 600, 600, 550, 650, 550, 600, 600, 600, 1700, 600, 1700, 650, 1650, 600}; // NEC DF7887

void setup()
{
  Serial.begin(9600);
  irsend.begin();

}

void loop() {
  Serial.println("NEC");
  // irsend.sendNEC(0xDFF807);tried
  // irsend.sendNEC(0xDFF807, 32);tried
  irsend.sendNEC(0x00DFF807, 32);
  delay(444);
  // irsend.sendRaw(ON, 67, 32);tried
  Serial.println("NEC");
  irsend.sendNEC(0xDF7887, 32);
  //  irsend.sendNEC(0x00DF7887,32);tried;
  delay(400);
  //  irsend.sendRaw(OFF, 67, 32); tried

}
```

this is what i get from ir sender nodemcu:

```
Encoding  : UNKNOWN
Code      : 37A24698 (32 bits)
Timing[51]: 
[... 178 lines cut ...]
     +9050, -4500     + 650, - 550     + 650, - 550     + 650, -1100
     + 650, - 550     + 650, -1150     + 650, - 550     + 650, -3400
     + 650, - 550     + 650, -1700     + 650, -1700     + 650, -3400
     + 650, -1700     + 650, -1700     + 650, -1700     + 650, -1700
     + 650, -1700     + 650, -1700     + 650, -1150     + 650, - 550
     + 650, -1100     + 650, -1150     + 650, - 550     + 650, -1700
     + 650, -3400     + 650
unsigned int  rawData[51] = {9050,4500, 650,550, 650,550, 650,1100, 650,550, 650,1150, 650,550, 650,3400, 650,550, 650,1700, 650,1700, 650,3400, 650,1700, 650,1700, 650,1700, 650,1700, 650,1700, 650,1700, 650,1150, 650,550, 650,1100, 650,1150, 650,550, 650,1700, 650,3400, 650};  // UNKNOWN C8BE069F

Encoding  : UNKNOWN
Code      : 13B999C4 (32 bits)
Timing[51]: 
     +9050, -4500     + 650, -1100     + 650, - 550     + 650, -1100
     + 650, - 550     + 650, -1100     + 650, -1700     + 650, -1700
     + 650, -2250     + 650, -1700     + 650, -1700     + 650, -1700
     + 650, -1700     + 650, -3400     + 650, -1700     + 650, -1700
     + 650, -1700     + 650, - 550     + 650, -1100     + 650, -1100
     + 650, - 550     + 650, -1100     + 650, -1700     + 650, -1700
     + 650, -1700     + 650
unsigned int  rawData[51] = {9050,4500, 650,1100, 650,550, 650,1100, 650,550, 650,1100, 650,1700, 650,1700, 650,2250, 650,1700, 650,1700, 650,1700, 650,1700, 650,3400, 650,1700, 650,1700, 650,1700, 650,550, 650,1100, 650,1100, 650,550, 650,1100, 650,1700, 650,1700, 650,1700, 650};  // UNKNOWN 13B999C4

Encoding  : UNKNOWN
....
...
```

crankyoldgit, 2017-03-14:
Thanks for the report. This looks very odd. You're getting a much shorter code than expected.
I'm assuming you're using the current (master) version of the library. Correct?

crankyoldgit, 2017-03-14:
Another observation, all of the spaces()s in the rawData you listed are mostly at odds with what should be being sent.
e.g.
#define NEC_ONE_SPACE	1690
#define NEC_ZERO_SPACE	560
and the spaces in the raw data are: 550 , 1100, 1150, 1700, 2250, 3400 etc.
i.e. 1100,1150, 2250 & 3400 are clearly wrong/bad.

kiralikbeyin, 2017-03-14:
Thanks. I installed latest release 1.0.1 and everthing is fine. 
I tried some ir libs a few months ago but this lib decodes very well.

closed 2017-03-14 by kiralikbeyin

crankyoldgit, 2017-03-14:
Phew. I was pulling my hair out trying to find out what was wrong, because I couldn't see it.

kiralikbeyin, 2017-03-14:
IT WORKS PERFECTLY! THANK YOU :)