crankyoldgit/IRremoteESP8266 #108: ir transmitter and receiver advise & apple tv

kiralikbeyin, 2017-03-15:
Which model ir transmitter and receiver do you use?

Did anyone try this library with apple tv remote controller?

crankyoldgit, 2017-03-16:
Most transmitters are done with a simple 940nm IR led. Try the typical places like Ebay/Aliexpress/etc.
Consult a datasheet if you have some specific requirements. e.g. current/voltage/angle requirements.

IR receivers depend on what you are trying to receive. Some are better at certain frequencies/wavelengths than others.
I've personally gone for a TSOP38238 as my devices are all 38kHz. See http://www.vishay.com/docs/82491/tsop382.pdf for an example of different modules for diff freqs.

But YMMV, do your own research.

I personally haven't tried the (old) apple tv IR remotes. Beware that newer ones are also bluetooth, I think. According to [wikipedia](https://en.wikipedia.org/wiki/Apple_Remote#Technical_details) it should work as they use the NEC protocol and the standard frequency (38kHz).

closed 2017-03-16 by crankyoldgit

kiralikbeyin, 2017-03-17:
I have TSOP38238 for decode no problem.

For transmitting gray-blue like in the picture works fine but white one didnt work.Maybe white one is for decoding?
[image]

crankyoldgit, 2017-03-17:
The physical appearance colour of the LED doesn't really matter. I've use clear cheap 940nm IR leds, and they seem to work for me. If the specs are correct, then they should work. A lot of this comes down to how you are driving them in a circuit too.