crankyoldgit/IRremoteESP8266 #118: sendSony problem

kiralikbeyin, 2017-03-17:
Decoded from sony remote
```
Encoding  : SONY
Code      : 540C (15 bits)
Timing[31]: 
     +2550, - 500     +1350, - 500     + 750, - 500     +1350, - 500
     + 750, - 500     +1350, - 500     + 750, - 500     + 750, - 500
     + 750, - 500     + 750, - 500     + 750, - 500     + 750, - 500
     +1350, - 500     +1350, - 500     + 750, - 500     + 750
unsigned int  rawData[31] = {2550,500, 1350,500, 750,500, 1350,500, 750,500, 1350,500, 750,500, 750,500, 750,500, 750,500, 750,500, 750,500, 1350,500, 1350,500, 750,500, 750};  // SONY 540C
unsigned int  data = 0x540C;

21516
```

Master lib nec,samsung or sony and also sendRaw didnt work.

1.0.1 release sends nec and samsung but sony doesnt work.

crankyoldgit, 2017-03-17:
Hey @kiralikbeyin, thanks for the report. I was able to find a subtle bug in some of the new timing code.
Can you please re-download the latest master and see if the issues are still current?

kiralikbeyin, 2017-03-17:
OK now samsung and nec works with master but sony still makes trouble

crankyoldgit, 2017-03-17:
Can you give us the code snippet you are using to invoke sendSony()? Try using just the constants so there is no other code issues etc. and/or distilling your code down to the bare minimum.
e.g.
```
// You need to send the code three times.
sendSony(0x540C, 15);
sendSony(0x540C, 15);
sendSony(0x540C, 15);
```

I tested the current 'master' revision of the library transmitting to an arduino with an IR receiver.
It decodes `sendSony(0x540C, 15);` correctly. All the timings appear to be correct to within +/- 50usecs of the [Sony spec](http://www.sbprojects.com/knowledge/ir/sirc.php). I don't have a physical Sony device to confirm with.

kiralikbeyin, 2017-03-18:
I was sending only one time. 3 times worked but i have a strange problem now
ON-OFF AND VOL- is working perfect but VOL+ is not working 

SONY AV SYSTEM RM-AAU023 Remote controller

```
ON-OFF
Encoding  : SONY
Code      : 540C (15 bits)
Timing[31]: 
     +2450, - 550     +1300, - 550     + 700, - 550     +1250, - 550
     + 650, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1250, - 550     +1250, - 550     + 650, - 550     + 650
unsigned int  rawData[31] = {2450,550, 1300,550, 700,550, 1250,550, 650,550, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1250,550, 1250,550, 650,550, 650};  // SONY 540C
unsigned int  data = 0x540C;

21516

VOL-
Encoding  : SONY
Code      : 640C (15 bits)
Timing[31]: 
     +2500, - 550     +1300, - 550     +1300, - 550     + 700, - 550
     + 700, - 550     +1250, - 550     + 650, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1300, - 550     +1300, - 550     + 700, - 550     + 700
unsigned int  rawData[31] = {2500,550, 1300,550, 1300,550, 700,550, 700,550, 1250,550, 650,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 700,550, 700};  // SONY 640C
unsigned int  data = 0x640C;

25612

VOL +
Encoding  : SONY
Code      : 240C (15 bits)
Timing[31]: 
     +2450, - 550     + 700, - 550     +1250, - 550     + 700, - 550
     + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
     + 700, - 550     + 700, - 550     + 700, - 550     + 700, - 550
     +1250, - 550     +1300, - 550     + 700, - 550     + 700
unsigned int  rawData[31] = {2450,550, 700,550, 1250,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 700,550, 700,550, 700,550, 1250,550, 1300,550, 700,550, 700};  // SONY 240C
unsigned int  data = 0x240C;

9228

```

crankyoldgit, 2017-03-18:
In principle there is no difference in sending one code vs. the next with respect to how sendSony() operates. So, I don't think there is any problem with the library code.

How sure are you that the value 0x240C is VOL+? Have you tried stripping down your code to just sending that hex value 3 times to eliminate any other mistakes in your own code?

I looked up on globalcache.com for Sony Stereos and TVs and they don't appear to match the raw bit pattern in your capture above.

kiralikbeyin, 2017-03-18:
I tried 
```
irsend.sendSony(0x240C , 15);
irsend.sendSony(0x240C , 15);
irsend.sendSony(0x240C , 15);
````
and also
````
   unsigned int  rawData[31] = {2450, 550, 700, 550, 1250, 550, 700, 550, 700, 550, 1300, 550, 700, 550, 700, 550, 700, 550, 700, 550, 700, 550, 700, 550, 1250, 550, 1300, 550, 700, 550, 700}; // SONY 240C
      irsend.sendRaw(rawData, 31 , 15);
      irsend.sendRaw(rawData, 31 , 15);
      irsend.sendRaw(rawData, 31 , 15);
`````

I know it is 0x240C because i pressed VOL+ and saw decoded values. VOL- or ONOFF buttons are decoded and send in the same way.

I will try sending different values for VOL+ (near 0x640C) because i think wrong decoding is making this problem.

crankyoldgit, 2017-03-18:
You are using sendRaw incorrectly. The last argument is the frequency. I.e you want 40kHz for Sony.

You'll also need to add the delays when sending raw, as it doesn't know about them unlike sendSony().

kiralikbeyin, 2017-03-18:
ok i tried it again as you said

```` 
unsigned int  rawData[31] = {2450, 550, 700, 550, 1250, 550, 700, 550, 700, 550, 1300, 550, 700, 550, 700, 550, 700, 550, 700, 550, 700, 550, 700, 550, 1250, 550, 1300, 550, 700, 550, 700}; // SONY 240C
    irsend.sendRaw(rawData, 31 , 40);
    delay(50);
    irsend.sendRaw(rawData, 31 , 40);
    delay(50);
    irsend.sendRaw(rawData, 31 , 40);
````
didnt work..

I decoded some of other buttons and send them.. %50 working %50 not working

8-10 years passed since i bought this sony device, is there any difference between new ones?
+i put new batteries to remote controller nothing change.

kiralikbeyin, 2017-03-18:
OK Decoding is correct, i used this https://github.com/cyborg5/IRLib2/blob/master/IRLib2/examples/record/record.ino with a arduino nano 
Received Sony Value:0x240C
And it sends correctly VOL+ and also other buttons worked.

So there is a problem with sending sony..

crankyoldgit, 2017-03-18:
Are you able to decode what the transmitting sendSony() ESP8266 is sending on another ESP8266?

crankyoldgit, 2017-03-18:
On my test setup, sending from master IRremoteESP8266 sendSony() to an arduino yields:
```
Decoded Sony(2): Value:240C Adrs:0 (15 bits) 
Raw samples(32): Gap:28472
  Head: m2350  s650
0:m600 s600	1:m1200 s650		 2:m600 s600	3:m600 s650		 
4:m1150 s650	5:m600 s650		 6:m550 s650	7:m600 s650		 
8:m550 s650	9:m600 s650		 10:m550 s650	11:m1200 s650		 
12:m1150 s650	13:m600 s600		 14:m600
Extent=23100
Mark  min:550	 max:1200
Space min:600	 max:650
```

You might want to try adjusting the value of:
```
#define SONY_ONE_MARK	1200
#define SONY_ZERO_MARK	600
```
to
```
#define SONY_ONE_MARK	1250  // or 1300
#define SONY_ZERO_MARK	650  // or 700
```
in IRremoteInt.h to see if that helps as that may be closer to what your physical remote is sending.

kiralikbeyin, 2017-03-18:
Is this setup ok?
int RECV_PIN = 2; //D4-2
IRrecv irrecv(RECV_PIN);
IRsend irsend(4); //D2-4

When i send sony from esp to nano, my nano serial never ends (only for sendSony)...
Sending nec and samsung from esp to nano, nano decode was ok.

```
Send a code from your remote and we will record it.
Type any character and press enter. We will send the recorded code.
Type 'r' special repeat sequence.
Received Unknown saving raw data.
Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:8802
  Head: m2350  s650
0:m700 s600	1:m1250 s650		 2:m650 s600	3:m750 s600		 
4:m1250 s600	5:m750 s600		 6:m700 s600	7:m700 s600		 
8:m700 s600	9:m700 s600		 10:m700 s600	11:m1300 s600		 
12:m1300 s600	13:m700 s600		 14:m700 s1450	15:m2400 s600		 

16:m700 s600	17:m1300 s600		 18:m700 s600	19:m700 s600		 
20:m1300 s600	21:m700 s600		 22:m700 s600	23:m700 s600		 
24:m700 s600	25:m700 s600		 26:m700 s600	27:m1250 s600		 
28:m1300 s600	29:m700 s600		 30:m700 s1500	31:m2400 s600		 

32:m700 s600	33:m1300 s600		 34:m700 s600	35:m700 s600		 
36:m1250 s650	37:m650 s650		 38:m650 s650	39:m650 s650		 
40:m650 s650	41:m700 s600		 42:m650 s650	43:m1250 s650		 
44:m1250 s650	45:m650 s650		 46:m700 s1500	47:m2400 s600		 

48:m700
Extent=81000
Mark  min:650	 max:2400
Space min:600	 max:1500

Received Unknown saving raw data.
Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:46934
  Head: m51914  s32550
0:m-30586 s30050	1:m-3086 s32550		 2:m32450 s30050	3:m-28086 s30050		 
4:m-3086 s30050	5:m-28086 s30050		 6:m-30586 s30050	7:m-30586 s30050		 
8:m-30586 s30050	9:m-30586 s30050		 10:m-30586 s30050	11:m-586 s30050		 
12:m-586 s30050	13:m-30586 s30050		 14:m-30586 s7014	15:m-11122 s30050		 

16:m-30586 s30050	17:m-586 s30050		 18:m-30586 s30050	19:m-30586 s30050		 
20:m-586 s30050	21:m-30586 s30050		 22:m-30586 s30050	23:m-30586 s30050		 
24:m-30586 s30050	25:m-30586 s30050		 26:m-30586 s30050	27:m-3086 s30050		 
28:m-586 s30050	29:m-30586 s30050		 30:m-30586 s9514	31:m-11122 s30050		 

32:m-30586 s30050	33:m-586 s30050		 34:m-30586 s30050	35:m-30586 s30050		 
36:m-3086 s32550	37:m32450 s32550		 38:m32450 s32550	39:m32450 s32550		 
40:m32450 s32550	41:m-30586 s30050		 42:m32450 s32550	43:m-3086 s32550		 
44:m-3086 s32550	45:m32450 s32550		 46:m-30586 s9514	47:m-11122 s30050		 

48:m-30586
Extent=773150
Mark  min:-30586	 max:32450
Space min:7014	 max:32550

Received Unknown saving raw data.
Decoded Unknown(0): Value:0 Adrs:0 (0 bits) 
Raw samples(100): Gap:52990
  Head: m39746  s54686
0:m-22022 s-4778	1:m-23278 s-10850		 2:m-15950 s-4778	3:m-28094 s-4778		 
4:m-23278 s-4778	5:m-28094 s-4778		 6:m-22022 s-4778	7:m-22022 s-4778		 
8:m-22022 s-4778	9:m-22022 s-4778		 10:m-22022 s-4778	11:m-29350 s-4778		 
12:m-29350 s-4778	13:m-22022 s-4778		 14:m-22022 s23070	15:m-31862 s-4778		 

16:m-22022 s-4778	17:m-29350 s-4778		 18:m-22022 s-4778	19:m-22022 s-4778		 
20:m-29350 s-4778	21:m-22022 s-4778		 22:m-22022 s-4778	23:m-22022 s-4778		 
24:m-22022 s-4778	25:m-22022 s-4778		 26:m-22022 s-4778	27:m-23278 s-4778		 
28:m-29350 s-4778	29:m-22022 s-4778		 30:m-22022 s16998	31:m-31862 s-4778		 

32:m-22022 s-4778	33:m-29350 s-4778		 34:m-22022 s-4778	35:m-22022 s-4778		 
36:m-23278 s-10850	37:m-15950 s-10850		 38:m-15950 s-10850	39:m-15950 s-10850		 
40:m-15950 s-10850	41:m-22022 s-4778		 42:m-15950 s-10850	43:m-23278 s-10850		 
44:m-23278 s-10850	45:m-15950 s-10850		 46:m-22022 s16998	47:m-31862 s-4778		 
endless...

```

kiralikbeyin, 2017-03-18:
```
#define SONY_ONE_MARK	1250  // or 1300
#define SONY_ZERO_MARK	650  // or 700
```
Thats it Sony sends VOL+ now!

Thank you!

PR #121 (2017-03-23): Adjust Sony mark() timings based on experimental evidence.
See issue for more details.
And more reading at http://www.righto.com/2010/03/understanding-sony-ir-remote-codes-lirc.html backs the increase in the mark() timings based on LIRC data.
Fixes #118
closed 2017-03-18 by kiralikbeyin