crankyoldgit/IRremoteESP8266 #120: Get same code and send back this code

Joypacharya, 2017-03-17:
I want to try to send same code which i did get.
Due to some reason my module has been restarted .

here the exception decoder result,

Decoding 12 results
0x40105a51: ets_timer_disarm at ?? line ?
0x4020f1b4: pp_tx_idle_timeout at ?? line ?
0x401020f3: ppCalFrameTimes at ?? line ?
0x4021226e: pm_onBcnRx at ?? line ?
0x40212e70: pm_sleep_for at ?? line ?
0x4020f490: DefFreqCalTimerCB at ?? line ?
0x4020e734: pp_disable_idle_timer at ?? line ?
0x4020f059: pp_tx_idle_timeout at ?? line ?
0x4020ea1b: ppPeocessRxPktHdr at ?? line ?
0x4020ca3b: loop_task at C:\Users\JOY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 57

So, help me resolve this issue.I want complete this project as soon as possible.
further information you need than tell me.

crankyoldgit, 2017-03-17:
Can you provide us a link to your code?
What version of the IRremoteESP8266 library are you running? e.g. 1.0.1 or master?

Joypacharya, 2017-03-17:
master version .

crankyoldgit, 2017-03-19:
Thanks. What about a link to your code?

Joypacharya, 2017-03-20:
Can you provide any example for  my code ?

crankyoldgit, 2017-03-20:
Sorry if I wasn't clear. I'm asking for some way for us to see your code, so we can check how you are using your library. e.g. Put it up on pastebin or a link to it in Google Drive/Github, or just cut the code down to a workable size that still has the problem, then post it as a code quote in a comment on this issue.

That way we can try to help you better.

Joypacharya, 2017-03-20:
Actually i am working on this prototype. I found some error while i was combining.
If you provide any example than it's better for understand.

crankyoldgit, 2017-03-20:
From what you've put in the issue/report so far, I can't figure out what is going wrong. i.e. which part of our library might be causing the crash. With out more concrete details I don't think we can help you.

If your project is to big/sensitive/confidential/etc that you don't want to share it, perhaps you can make a simple example where you can reproduce the crash/issue and you feel happy enough to share.

If we can't get enough details to point us in a direction to look, or a way to reproduce the problem, or to see if you are using the library in some way we were not expecting, then we really can't help you.

TL;DR: We need more information and data to be able to help you. You have not given us enough as it stands now.

crankyoldgit, 2017-03-20:
FYI, it might help both of us if you read [this](http://www.chiark.greenend.org.uk/~sgtatham/bugs.html), it might help you to be able to help me help you better.

Joypacharya, 2017-03-20:
Actually i am using irsend.sendNEC(0x00000,xx) for transmit IR code.
and IR receiver,

void  loop ( )
{
  decode_results  results;        // Somewhere to store the results

  if (irrecv.decode(&results)) 
{  // Grab an IR code
    dumpCode(&results);           // Output the results as source code
    Serial.println("");           // Blank line between entries
    irrecv.resume();              // Prepare for the next value
  }
}  

So, i got data from this code and send same data via irsend.

crankyoldgit, 2017-03-20:
Where and what is your initialisation & setup of irrecv? I.e. where you define which pin etc.

Are you sending and receiving on the same board? If not, which one is crashing? If the same, what pins are you using?

Joypacharya, 2017-03-20:
as per example i sat down everything. IR  at 5 and Tx at 14.
Sender and receiver both put on same esp board.
even i did shut down interrupt at Tx time and on at Rx time. Even my hardware crash down.

crankyoldgit, 2017-03-20:
Do you know which part of your code is crashing? e.g. The sending part or the receiving part, or something else? Have you put in some debug Serial.println()s and monitored the serial port to work out where the program is getting to before it crashes? That is to find out what is the last successful line of code before it crashes.

My _guess_ is, if the library is the cause of the crash, then it will be in the receive part, as the sending part really doesn't have anything that should cause a crash.

Looking at your last code snippet, I'm guessing you've based this on [IRrecvDumpV2.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino). Have you tried running that exact code, modified to match your desired pins of course, on your existing hardware setup? Does that run without error/crash?

crankyoldgit, 2017-03-28:
Just chasing this up. Any luck in finding more information to help us debug your issue?

darshkpatel, 2017-04-05:
I noticed you don't have any delay in between, 
If you don't have a delay between and just want to resend the same code , just like a ir range extender,  then why not just setup a circuit to replay the same code ? Instead of decoding it with a esp and transmitting it back again 
Something like this
 http://panofish.net/infrared-ir-remote-extender/

Joypacharya, 2017-04-08:
Ok.i am trying to do that.

crankyoldgit, 2017-04-12:
Just a thought/potential issue. If you are receiving and then re-transmitting, you probably want to _not_ resume() until you are finished the re-transmission.
If you don't, you will potentially overwrite the the receive buffer if the IR receiver gets any new signals. That would probably lead to corruption of what you are re-transmitting. Also, your IR re-transmission might be what is causing the corruption/interference, hence don't 'resume()' until it is safe to do so.

I have an incoming patch which will stop the corruption by making a copy of the receive buffer, if you want to try that, but I still recommend disabling IR receiving while you are transmitting.

crankyoldgit, 2017-04-26:
Another thing to note, You can't tell the modulation frequency of the incoming signal from behind a _typical_ IR demodulator, so you can't just repeat the 'signal' from the demodulator  directly to an IR LED. You'd also need to implement a duty cycle/modulation on top of the demodulator output if you are going to do it all in hardware.

If you are going to capture, and them send raw, you also have the problem of not knowing what the inbound frequency was/is. You'd have to guess.

If you want to use a _common/typical_ IR demodulator, your best bet is to decode the protocol, and then send using the appropriate sendFoo() function for that protocol. That's not going to work for everything obviously.

If you want to make a true repeater, you'll need an IR receiver that doesn't demodulate, but literally detects every little on/off of the incoming signal, or passes on duty-cycle and frequency info some how. Maybe via a I2C interface.

rahuljawale, 2017-04-27:
Just to add to what @crankyoldgit said, you might end up repeating unwanted noise/interference as well.

crankyoldgit, 2017-06-08:
Friendly ping. Is this still an issue, or can we close this bug?

crankyoldgit, 2017-06-19:
Last ping before I close this issue due to inactivity.

Joypacharya, 2017-06-19:
Sorry for late response.you may close this issue.thanks for all kind help.

closed 2017-06-19 by crankyoldgit