crankyoldgit/IRremoteESP8266 #122: [RC6] seems not sending the right signal?

mafflerbach, 2017-03-18:
from the level It's not a newbie question. But I'm a newbie in embedded programming (i'm coming from the web development side). 

I have an ai Thinker NodeMCU dev board connected with an Ir diode.

i collected with my raspberry and irrecord the specific key codes. My tests to controll my tv with these code over irsend is functioning (e.g. irsend SEND_ONCE philips KEY_MUTE)

*KEY_MUTE stands for the signal 0x0FFFF2 btw for signal 1048562 

But i have problem to send the signal over the dev board. A roundtrip test (rerecording the data from esp to raspi) have a complete other hex lis. i.g. 

KEY_MUTE 0x0FFFF2 => 1048562  
KEY_MUTE 0x10000D => 1048589 

here is my code *striped some lines: 
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <IRremoteESP8266.h>

const char* ssid = "*";
const char* password = "*";
MDNSResponder mdns;

ESP8266WebServer server(80);

IRsend irsend(13);

void handleRoot() {
 server.send(200, "text/html", "<html><body><a href=\"ir?code=0x0FFFF2&key=KEY_MUTE&int=1048562\">");
}

void handleIr(){
  for (uint8_t i=0; i<server.args(); i++){

    if(server.argName(i) == "int")
    {
      unsigned long code = server.arg(i).toInt();
      int foo = server.arg(i).toInt();
		// testing for the roundtrip
      for(int i = 0; i < 10; i++) {
        irsend.sendRC6(foo, 21);
        delay(10);
        Serial.println(foo);
      }
    }
  }
  handleRoot();
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup(void){
  irsend.begin();

  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (mdns.begin("esp8266", WiFi.localIP())) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);
  server.on("/ir", handleIr);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void){
  server.handleClient();
}

```

the base bit was switch from 21 to 22

So, on this point totally clueless how to debug this shi*** .
so maybe the folk from here have some ideas.

esptest lirc.conf

```
begin remote

  name  esptest
  bits           22
  flags RC6|CONST_LENGTH
  eps            30
  aeps          100

  header       2665   915
  one           437   470
  zero          437   470
  gap          34528
  min_repeat      9
#  suppress_repeat 9
#  uncomment to suppress unwanted repeats
  toggle_bit_mask 0x3
  rc6_mask    0x20000

      begin codes
          KEY_POWER                0x11000C
          KEY_MUTE                 0x10000D
          KEY_ARCHIVE              0x1100CC
          KEY_INFO                 0x10000F
          KEY_1                    0x110001
          KEY_2                    0x110002
          KEY_3                    0x100003
          KEY_4                    0x110004
          KEY_5                    0x110005
          KEY_6                    0x100006
          KEY_7                    0x110007
          KEY_8                    0x100008
          KEY_9                    0x110009
          KEY_0                    0x110000
      end codes

end remote

```

functioning lirc.conf

```
begin remote

  name  philips
  bits           21
  flags RC6
  eps            30
  aeps          100

  header       2658   911
  one           431   457
  zero          431   457
  gap          83814
  min_repeat      1
#  suppress_repeat 1
#  uncomment to suppress unwanted repeats
  toggle_bit_mask 0x0
  rc6_mask    0x10000

      begin codes
KEY_0                    0x0EFFFF
KEY_1                    0x0EFFFE
KEY_2                    0x0EFFFD
KEY_3                    0x0FFFFC
KEY_4                    0x0EFFFB
KEY_5                    0x0EFFFA
KEY_6                    0x0FFFF9
KEY_7                    0x0EFFF8
KEY_8                    0x0FFFF7
KEY_9                    0x0EFFF6
KEY_MUTE                 0x0FFFF2
KEY_ARCHIVE              0x0EFF33
KEY_INFO                 0x0FFFF0
KEY_BACK                 0x0EFFF5
KEY_MENU                 0x0FFFAB
KEY_UP                   0x0EFFA7
KEY_DOWN                 0x0FFFA6
KEY_LEFT                 0x0EFFA4
KEY_RIGHT                0x0FFFA5
KEY_VOLUMEDOWN             0x0EFFEE
KEY_VOLUMEUP                       0x0EFFEF
KEY_CHANNELDOWN            0x0FFFDE
KEY_CHANNELUP                      0x0FFFDF
KEY_OK                   0x0FFFA3
KEY_RED                  0x0EFF92
KEY_BLUE                 0x0FFF8F
KEY_GREEN                0x0EFF91
KEY_YELLOW               0x0FFF90
KEY_OPEN                 0x0FFFC7
KEY_POWER                0x0EFFF3
end codes
end remote
```

crankyoldgit, 2017-03-18:
Having a look at your code, a few things stand out as potential issues:

```
      unsigned long code = server.arg(i).toInt();
      int foo = server.arg(i).toInt();
		// testing for the roundtrip
      for(int i = 0; i < 10; i++) {
        irsend.sendRC6(foo, 21);
        delay(10);
        Serial.println(foo);
```

FYI, int on the ESP8266 & arduino's is only 16 bits. long is 32 bit. This is unlike linux which typically has a 32 bit int. So change all instances of 'foo' to 'code' is a good first start. The codes you are trying to send are all bigger than a uint16_t (0xFFFF).
Next, I'm not 100% sure .toInt() handles longs. It probably does, but worth checking/confirming.

You are printing 'foo' (code) to serial. What is the value it is displaying. i.e. Does it match what you're sending as the '&int=' html argument?

> So, on this point totally clueless how to debug this shi*** .

Lots of Serial.println()s. ;-) You'll need to connect a serial monitor/terminal/etc to the ESP8266 and set it to 115200 baud. We also typically use our own library (or similar) on another ESP8266 or arduino device and capture the _raw_ timing values, so we know exactly what we are looking at. See examples directory.

Now, my confessions. I'm not familiar with LIRC that much, so you may need to explain what you are seeing a bit more to me. Next, I haven't really worked with or have any RC6 devices, so I can't easily test with 100% assurance.

A thing I find unsettling is, if I am understanding the LIRC stuff correctly, is that the ESP is transmitting 22 bits, when you've told it 21. Btw, where are you getting the '21' value in your `irsend.sendRC6(foo, 21);` call? 21 seems to be an odd (pun intended) value for the number of bits. The [RC6 protocol](http://www.sbprojects.com/knowledge/ir/rc6.php) calls for a Stop Bit after the initial header. This isn't counted in the IRremote library in the nr. of bits. Maybe LIRC is seeing that and counting it in its count of bits. So something to try is use '20' instead of '21'. e.g. `irsend.sendRC6(code, 20);`

Let us know what debug values you're getting and what of the above you've tried.

mafflerbach, 2017-03-19:
Most of the code (okay eeeevrything ;) ) comes from examples/IRServer.

in case of 21 to 22 bit:
I thought the bits param from the lirc.conf is the right bit counting. But maybe i missunderstood the meaning of "bits" in the conf and "bits" as param in the methode.

```
//extending only the handleIr methode 
void handleIr(){
  for (uint8_t i=0; i<server.args(); i++){
    //Serial.println("arg: " + server.argName(i));

    if(server.argName(i) == "int")
    {
      Serial.println("val: " + server.arg(i));
      unsigned long code = server.arg(i).toInt();
      int code2 = server.arg(i).toInt();
      //Serial.print("Send: " + code);
      // don't print to SM - but why ?
      Serial.printf("code: %u\n", code);
      // print the expected value

      for(int i = 0; i < 7; i++) {
        irsend.sendRC6(code, 20);
        delay(10);
      Serial.printf("code: %u\n", code);
      
	  }

      /**
      Complete Output in SM:

      Connected to "Networkname"
      IP address: "IP Address"
      MDNS responder started
      HTTP server started
// after click on the link:
      val: 1048562
      code: 1048562

      */
    }
  }
  handleRoot();
}

```
So at this point maybe i have 2 problems... The wrong bit (i have teste with 20 and 22 bit, but both of them could not fix the problem), and the second is, that the codes from irrecord from linux is not compatible with the lib from IRremoteESP8266.

My next step is, to record the codes with the esp, and testing with this side.

crankyoldgit, 2017-03-19:
> My next step is, to record the codes with the esp, and testing with this side.
Taking a step back. What is it you are trying to achieve?
I'm assuming you have a device which receives an RC6 code, and a remote that produces an RC6 code and you want to replicate the latter on the ESP.
You should then look at dumping what the actual remote sends, on the ESP using https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino or similar.
Numeric 'codes' can often differ from one library implementation to another for the same physical button press on an actual remote. e.g. Some store the incoming transmitted bit pattern starting at the LSB and some in the MSB.

Also, the _raw_ dump will include the timing values of each mark(led on) and each space(led off) period, and will be useful to debug if we are producing an incorrect pattern with sendRC6().

Thanks for the extra debugging info so far. I'm glad to know .toInt() handles longs also on the platform.

BTW, what device & remote are you trying to emulate? e.g. XBox 360 and some MCE stuff uses 36 bit codes which we are trying to work out support for, as they are larger than the 32bit longs currently used.

crankyoldgit, 2017-03-19:
Oh, reading up a bit on LIRC, I found this: http://www.righto.com/2010/12/64-bit-rc6-codes-arduino-and-xbox.html

> To use these RC6 code with the Arduino takes a bit of work. First, the codes have been split into 21 bits of pre-data, followed by 16 bits of data. So if you want the OnOff code, you need to concatenate the bits together, to get 37 bits: 0x37ff08bf3. The second factor is the Arduino library provides the first start bit automatically, so you only need to use 36 bits. The third issue is the LIRC files inconveniently have 0 and 1 bits swapped, so you'll need to invert the code. The result is the 36-bit code 0xc800f740c that can be used with the IRremote library.
The rc6_mask specifies which bit is the double-wide "trailer" bit, which is the fourth bit sent (both in 20 and 36 bit RC6 codes). The library handles this bit automatically, so you can ignore it.
> 
> The toggle_bit_mask is important, as it indicates which position needs to be toggled every other transmission. For example, to transmit OnOff you will send 0xc800f740c the first time, but the next time you will need to transmit 0xc800ff40c.

So, you can't just take the code from LIRC and use it in an IRremote-esque based library.
Using a 20 bit length in our library is the equiv of 21 bit in LIRC for this protocol (see quote).

That also reminds me, **you** need to handle the toggle bit when sending multiple codes with RC6. This library leaves it up to the caller to handle that aspect.

For 36bit codes, there is an experimental RC6 branch of the library, where I've been working on supporting that bit code length. It might be worth a shot, but you'll need to convert your IR code args to unsigned long long's to use it.

mafflerbach, 2017-03-20:
> BTW, what device & remote are you trying to emulate? e.g. XBox 360 and some MCE stuff uses 36 bit codes which we are trying to work out support for, as they are larger than the 32bit longs currently used.

 It's a orginal philips RC 4492 01b. 

Ohh well, really interesting. I thougt i have a rc6 remote controll. But the dumping says other things (see below).

So, i understand it right, that my only option by these incomming data is, sending raw data ?

Here is the output if someone needs it in the future:

```
on/off

Encoding  : UNKNOWN
Code      : B4CB49F4 (32 bits)
Timing[40]:
     + 950, - 500     + 850, - 500     + 400, - 500     + 400, - 500
     + 850, - 950     + 450, - 500     + 400, - 500     + 400, - 500
     + 400, - 500     + 400, - 500     + 400, - 500     + 400, - 500

     + 400, - 500     + 400, - 500     + 400, - 500     + 400, - 500
     + 400, - 950     + 450, - 500     + 850, - 500     + 400, - 500
 
unsigned int  rawData[40] = {950,500, 850,500, 400,500, 400,500, 850,950, 450,500, 400,500, 400,500, 400,5
00, 400,500, 400,500, 400,500, 400,500, 400,500, 400,500, 400,500, 400,950, 450,500, 850,500, 400,500 };
// UNKNOWN B4CB49F4
 

mute

Encoding  : UNKNOWN
Code      : 11C1CC36 (32 bits)
Timing[42]:
     + 950, - 500     + 900, - 500     + 450, - 500     + 450, - 500
     + 900, - 900     + 500, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500
unsigned int  rawData[42] = {950,500, 900,500, 450,500, 450,500, 900,900, 500,500, 450,500, 450,500, 450,5
00, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500 };  
// UNKNOWN 11C1CC36

1
Encoding  : UNKNOWN
Code      : 8075A6C5 (32 bits)
Timing[38]:
     + 950, - 500     + 900, - 500     + 450, - 500     + 450, -1300
     +1400, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 950
unsigned int  rawData[38] = {950,500, 900,500, 450,500, 450,1300, 1400,500, 450,500, 450,500, 450,500, 450
,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,950 };  
// UNKNOWN 8075A6C5

2
Encoding  : UNKNOWN
Code      : 36F039E3 (32 bits)
Timing[40]:
     + 950, - 500     + 850, - 500     + 400, - 500     + 400, - 500
     + 850, - 950     + 450, - 500     + 400, - 500     + 450, - 500
     + 400, - 500     + 450, - 500     + 400, - 500     + 450, - 500
     + 450, - 500     + 400, - 500     + 450, - 500     + 450, - 500
     + 400, - 500     + 450, - 500     + 450, - 950     + 900, - 500
 
unsigned int  rawData[40] = {950,500, 850,500, 400,500, 400,500, 850,950, 450,500, 400,500, 450,500, 400,5
00, 450,500, 400,500, 450,500, 450,500, 400,500, 450,500, 450,500, 400,500, 450,500, 450,950, 900,500 };
// UNKNOWN 36F039E3

[... 82 lines cut ...]
Timing[38]:
     + 950, - 500     + 900, - 500     + 450, - 500     + 450, - 500
     + 900, - 900     + 500, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 900     + 950, - 500     + 450, - 900
unsigned int  rawData[38] = {950,500, 900,500, 450,500, 450,500, 900,900, 500,500, 450,500, 450,500, 450,5
00, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,900, 950,500, 450,900 };  
// UNKNOWN A0873985

0
Encoding  : UNKNOWN
Code      : 11C1CC36 (32 bits)
Timing[42]:
     + 950, - 500     + 900, - 500     + 450, - 500     + 450, - 500
     + 900, - 900     + 500, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500     + 450, - 500     + 450, - 500     + 450, - 500
     + 450, - 500
unsigned int  rawData[42] = {950,500, 900,500, 450,500, 450,500, 900,900, 500,500, 450,500, 450,500, 450,5
00, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450,500, 450
,500 };  // UNKNOWN 11C1CC36

```

crankyoldgit, 2017-03-21:
Thanks for that. In the short term, yes, raw format is what you'll need to use to (re)send those codes.

The codes you have provided are a little odd. The don't seem to have a clear header mark & space like most protocols. Next, they seem to be of different lengths which is even more odd.
I don't recognise the protocol off the top of my head, but I'll have a look around later.

crankyoldgit, 2017-03-21:
You might want to try globalcache's IRDB to see if you can find the codes for what ever the recipient device is.

mafflerbach, 2017-03-21:
After hours to reproduce send and resend codes in raw format, i can't get running a simple functioning call.... ohh what frustrating. 
hmph. Could i have damaged my receiver so it get oddish results?

crankyoldgit, 2017-03-21:
Unlikely.
I'd suggest trying getting a single code correct first. As I said earlier, the received results above don't have the right feel to them. They are odd. I'd expect long(er) pulses/timings for the first pair in the rawData, and a constant length to the rawData array for typical remote codes.

crankyoldgit, 2017-03-28:
Still need any help?

mafflerbach, 2017-03-28:
sorry in the last days i had not time to check the codes. Today i recorded my remote control and in this time i get totally other results. here in example the mute button: 

Encoding  : RC6
Code      : 1000D (20 bits)
Timing[37]:
     +2650, - 950     + 500, - 900     + 500, - 450     + 500, - 450
     +1300, -1400     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 500, - 450     + 500, - 450     + 500, - 450     + 500, - 450
     + 950, - 400     + 500, - 900     + 950
unsigned int  rawData[37] = {2650,950, 500,900, 500,450, 500,450, 1300,1400, 500,450, 500,450, 500,450, 500,450, 500,450, 500,450, 50
0,450, 500,450, 500,450, 500,450, 500,450, 950,400, 500,900, 950};  // RC6 1000D
unsigned int  data = 0x1000D;

oh, and watch its RC6 remote control -_- really annoying. But this result came after holding the button for a second or two.

So maybe this is the point, to get valid codes

crankyoldgit, 2017-03-28:
My _guess_ is the IR receiver you've got connected is a _typical_ 38kHz variety, and there might be some issues with the 36kHz signal from the remote. If you want to receive the IR signal to do actions on, then that's probably an issue. However, sending really doesn't require anything special other than a typical 940nm IR LED. If that's all you need to do, then you're probably fine.

Something to note, the example dumping code doesn't play well with repeated signals. The time it takes to print out the decoded info etc, it's already missed/mangled the next incoming code. Often you only get a good clean sample on the very first code.

It's good that its a standard RC6 protocol remote. Now all we need to work out is why the sendRC6() code isn't giving you a good enough signal for your device.

We seem to be using timing values of the following for RC6:
#define RC6_HDR_MARK	2666
#define RC6_HDR_SPACE	889
#define RC6_T1		444
#define RC6_RPT_LENGTH	46000

That pretty much is what we are seeing (+/- 50) in the raw dump and in the RC6 spec, except the 4th bit which is double length, seems to be greater than expected in your dump. Interesting.

Have you tried sending with `sendRC6(0x1000D, 20);` yet to see if it controls the device or if your LIRC device decodes it correctly as an RC6 signal?

crankyoldgit, 2017-03-29:
Oh, and remember, you have to manage the toggle bit yourself between transmissions for RC6.

darshkpatel, 2017-03-30:
Looks like 
https://github.com/markszabo/IRremoteESP8266/issues/21 
?

crankyoldgit, 2017-04-11:
FYI, something to try is the reporter's suggestion in #62

mafflerbach, 2017-04-11:
What the heck.... Okay Today i have a litlebit time to testing the "new" recorded codes, *Drum whirl* aaaaannnnndd it's work... I don't know what, i don't know how, and why, but I must have done something else. I wish I could better describe what was different, or why it did not work at the first time.

Nevertheless many thank you for the great support!

crankyoldgit, 2017-04-12:
Good to hear. If you do isolate what the cause of your issue was, I'd love to know so we can add it to the [Troubleshooting Guide](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide).

Otherwise we can chalk it up to ["Unvalidated, synchronous memory variance"](http://www.strauss.za.com/sla/support.asp).

crankyoldgit, 2017-04-17:
Marking this resolved now. If you think it isn't, please feel free to re-open it.

closed 2017-04-17 by crankyoldgit