crankyoldgit/IRremoteESP8266 #123: Troubles combining IRsend and IRrecv

Joypacharya, 2017-03-21:
Actually i am using both function in same hardware(12E).
So, how to write any program?
How to enable and disable receiver function ?
If possible than put any example for combine function.

crankyoldgit, 2017-03-21:
You can have a look at this code, it uses both the send and receive in the same code. They should work together fine.
https://github.com/1technophile/OpenMQTTGateway/blob/master/OpenMQTTGateway.ino

Joypacharya, 2017-03-22:
can you provide me more example for develop my code?
I check this code but they did not on and off interrupt or enable/disable function ir receiver at  transmit time.
so, help me for that one. and thanks for quick reply.

crankyoldgit, 2017-03-22:
http://lmgtfy.com/?q=ESP8266+disableIRIn()

There is at least one result on that page that has a link to some example code that seems to enable and disable the interrupts.

closed 2017-03-22 by crankyoldgit