crankyoldgit/IRremoteESP8266 #124: send.Sony still requires triple sending

wolass, 2017-03-21:
I am using the master lib because the release 1.0.1 even got the nec codes wrong...
The most problematic send.Sony still requires triple code sending (with 40 ms delay).

It was not working but after changing this 

```
   if (topic.lastIndexOf("IR_Sony")!=-1)
      irsend.sendSony(data, 12);
      delay(40);
      irsend.sendSony(data, 12);
      delay(40);
      irsend.sendSony(data, 12);
```

It works like a charm.

crankyoldgit, 2017-03-21:
See the comments [here](https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.h#L167) & [here](https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.cpp#L263). Yes, sendSony() needs to send the code _atleast_ 3 times for a sony device to acknowledge it.
Your use of delay()s is redundant when using the master branch.

Out of curiosity, what was broken with NEC in release 1.0.1? I want to know in case I need to push another release.

crankyoldgit, 2017-03-22:
I've merged my pull request in the master tree.
You should be able to replace your code with:
```
  if (topic.lastIndexOf("IR_Sony")!=-1)
      irsend.sendSony(data, 12, 2);
```

wolass, 2017-03-22:
Great I'll check later today.
I was not able to recognize NEC codes sent with the IRsendDemo example from release 1.0.1. Neither Sony nor NEC worked here. After upgrading to master NEC codes worked flawlessly, and Sony still nothing. 

I implemented the said triple sending and it worked. 

So now I hope I can use it without my prothesis. 

Great work guys!

crankyoldgit, 2017-03-22:
Yes, there was a bug in IRsendDemo. It was calling sendNEC with the wrong bit length. No actual bug in sendNEC() however.

closed 2017-03-22 by crankyoldgit