crankyoldgit/IRremoteESP8266 #136: Wrong decode espeasy with IRremoteesp8266

salimku, 2017-03-29:
Hi,
I am Salim. Just newbie in esp and arduino.
When I used Arduino (use irremote) to send IR to my NEC TV, is working properly.
But no if I used ESP-12 (use IRremoteESP8266, as plugin in easyesp).
looks like wrong code if I send IR thru espeasy (web: IP/control?cmd=irsend,NEC,20DF10EF,32

Please help me what should I do with my espeasy.
thanks
Salim

crankyoldgit, 2017-03-29:
Hey, I think I've found the problem. If I understand the code etc at easyesp correctly, they are not using our library, but are using an old unmaintained fork.
I've [reported](https://github.com/letscontrolit/ESPEasy/issues/188) it to them, and sent them a [fix](https://github.com/letscontrolit/ESPEasy/pull/189). Hopefully, that will resolve your problem.

In the meantime, should probably download/install/clone our library (master tree or the latest release) and rebuild easyesp. If you get easyesp to use our current library, it should send that code fine.

salimku, 2017-03-30:
Hey,
I used mega branch 2.0.0 dev 7 and change irremoteesp8266 lib from your the latest library.
For LG, I could send even two or more time repeat to make correct decode, but no luck with NEC always with wrong type and wrong decode.

Maybe there is wrong step to flash my esp?

Thank yous

crankyoldgit, 2017-03-30:
I can't speak for EasyESP, but I just tested the current 'master' head of this library, and sendNEC() works completely as expected for me sending from an ESP8266 device. Commands sent are accepted by a NEC-code using device. They decode correctly for me on an arduino which has a 38kHz IR receiver on it. I'm fairly confident that our libraries' sendNEC() is sending a correct signal when used correctly.

Note: I haven't tested decoding on the ESP8266 recently, so I'm not vouching for that at present.

Can you give us some more idea of what you are trying to do, and how you've got things hooked up? Please give us lots more details.

Have you tried one of our very basic send [example](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino#L20)? You'll need to change the code in the example to use what ever pin and IR code you want to use.

Are you driving the IR LED with sufficient voltage and current? (take a look at the wiki for some example hook ups. Basically, if you are powering the LED from the GPIO pin, it can't produce enough current for a good signal.)

For decoding, what part (number) are you using for receiving the IR signal? Is it a 38kHz part etc.?

salimku, 2017-03-31:
I need to use esp-01 or sonoff to remotely control my possible electronic devices such as lamp and AC. I used espeasy because this very-very powerful, from my perspective, controller: openhab2 and mosquitto for mqtt broker. I have installed and tested openhab2 and mosquitto on my RPI 3. Working like charm for Lamp controlled. Next I will use it also to control my LG aircon, turn on/off my LG TV (but its remote use NEC format, I don't know how come?). Tested IRremote lib on arduino with expected result to control my LG (but with NEC decode).

First, when I found wrong decode for NEC on esp-01, I think, maybe, voltage and current as same as your suggestion, so searched on some places about current and driver IR led with transistor if I use esp-01. both simple transistor and also constant current driver. but no luck.
Yes, I have flashed esp-01 with above example but incorrect decode for NEC but LG.
I flash IRrecvDumpV2 on my arduino nano for receiver.

Hardware:
esp-01 and esp-12, IR LED 940nm, IR receiver with some type, I think those are work ok (because it worked if I use arduino), transistor BC337 for both mode, simple gain and constant current.
Simple gain: GPIO-0 -330R - basis, IR anode to 3.3V/5V, cathode to collector, emitor to ground.
constant current: GPIO-0 - 330R basis. seri in4148 diode from basis to ground, IR anode to 3.3/5v, cathode to collector and emitor seri 2R7 to ground.

meanwhile I will try from fresh to flash esp-01, with constant current IR driver.
hope you can find missing step that I have done.

thank you
Salim

crankyoldgit, 2017-03-31:
FYI, a lot of devices & manufacturers use the NEC code format. It's very popular and expected.

It sounds like you've got the right kind of circuit to drive the IR LED.

I _think_ the issue might be with using an esp-01 module. They have a lot of limitations. (see: http://www.instructables.com/id/How-to-use-the-ESP8266-01-pins/?ALLSTEPS)
Maybe try using the esp-12 as the transmitter and a different GPIO? GPIO-0 is not the best choice due to it being used to control boot/programming mode etc I gave up on using ESP-1's ages ago.

Can you give us the rawData dump from the arduino of what you have the ESP-12 send? and tell us what sendNEC() code line you are using for each rawData dump?

Some things to try: send an NEC code with a small value. e.g. 0, 1, 1234 etc
`sendNEC(0x1, 32);`
Then try sending a larger than 16bit value. e.g. `sendNEC(0xDEADBEEF, 32);`

See how they decode on the arduino. I suggest this as old versions of this library and some other code that uses our library have problems with unsigned longs. That might expose if you're somehow including the wrong library when you compiled. That, or put some Serial.println()s in the library to confirm your using the correct library. 

I don't have a BC337, but I do have some 2N3904 which seem the same. If you can send me the circuit diagram I can try it on a ESP-01, I use a setup similar to the wiki example, except I'm using D4 to switch the transistor in my test setup.

crankyoldgit, 2017-03-31:
Also, this might be an issue (long shot): http://www.esp8266.com/viewtopic.php?p=42900

salimku, 2017-04-03:
Finally, I got everything work as charm.
thank for your guidance.
What I did:
1. Do from "fresh" and put correct and update IRremoteESP8266 lib
2. Use esp-12 instead esp-01.
3. Use constant current for IR driver (see https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)
4. Try with small value and then 16bit. both work as expected. Same raw data as using arduino.

Again, Thank for you help.

regards
Salim

closed 2017-04-03 by salimku

crankyoldgit, 2017-04-03:
Good to know. Thanks for the feedback. I'd personally like to know if the esp-01 worked or not in the same setup. So we can let others know etc.