crankyoldgit/IRremoteESP8266 #141: IRserver.ino doesn't handle 32bit unsigned values correctly.

crankyoldgit, 2017-04-07:
@carlostorrez pointed out an issue with the IRserver example.
It looks like when converting the url string value to a number, use of .toInt() handles _long_ value, but not _unsigned long_ values. We should use strtoul() instead of toInt().

carlostorrez, 2017-04-07:
Dear David Conran
The solution you propose is correct, however I do not understand why with the SendNEC instruction if it works without problems.

Finally I am used to programming in VS.Net with tooltip which helps a lot and in the Arduino environment does not exist. Can you tell me where I can find the Arduino instructions description?

Thanks in advance
regards

crankyoldgit, 2017-04-08:
@carlostorrez The values used in the sendNEC examples are 0xFFE01F & 0xFAB123, if they were larger, they wouldn't have worked either.
The both comfortably fit in to a _signed_ long. The maximum positive value for a _signed_ long on this platform is (2^31)-1 = 0x7FFFFFFF, as opposed to (2^32)-1 = 0xFFFFFFFF for an _unsigned_ long.
The values you were trying to send were >= 0x80000000. This tipped me off that .toInt() probably doesn't handle _unsigned_ longs. Further investigation digging through the arduino esp8266 code showed that .toInt() uses strtol() to convert the text to a number. We needed to handle unsigned values (larger maximum) to fully fill the 32bits of the command.
I hope that answers the 'why' part of your query.

As for where to find Arduino instruction descriptions, I don't know of a good resource other than googling (e.g. https://www.google.com.au/search?q=arduino+strtoul) or digging down into the source code via https://github.com/esp8266/Arduino. If you find a good resource/IDE etc, I'm interested. Currently I use Atom/PlatformIO and it doesn't provide that sort of thing that I know of.

PR #142 (2017-04-08): Handle unsigned long values in the URL
toInt() only handles long correctly, not unsigned long. Use strtoul() instead.

Fixes #141 

FYI @carlostorrez
closed 2017-04-08 by crankyoldgit