crankyoldgit/IRremoteESP8266 #144: Problem Sending for Sony DVD

scropion86, 2017-04-08:
Hi

below is sample IR signal collected using "IRrecvDumpV2"

```
               power

               IR code too long. Edit IRremoteInt.h and increase RAWBUF
               Timing[99]: 
                    +2500, - 550     +1300, - 550     + 700, - 550     +1300, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     +1300, - 550
                    +1300, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     + 700, - 550     + 700, - 550     +1300, - 550
                    + 700, -12950     +2500, - 550     +1300, - 550     + 700, - 550
                    +1300, - 550     + 700, - 550     +1300, - 550     + 700, - 550
                    + 700, - 550     + 700, - 550     +1300, - 550     + 700, - 550
                    +1300, - 550     +1300, - 550     +1300, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     + 700, -12950     +2500, - 550     +1300, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     +1300, - 550
                    + 700, - 550     + 700
               unsigned int  rawData[99] = {2500,550, 1300,550, 700,550, 1300,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 1300,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,12950, 2500,550, 1300,550, 700,550, 1300,550, 700,550, 1300,550, 700,550, 700,550, 700,550, 1300,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,12950, 2500,550, 1300,550, 700,550, 1300,550, 700,550, 1300,550, 700,550, 700};  // SONY A8B92
               unsigned int  data = 0xA8B92;

               play

               IR code too long. Edit IRremoteInt.h and increase RAWBUF
               Timing[99]: 
                    +2500, - 600     + 700, - 550     +1250, - 600     + 650, - 600
                    + 650, - 600     +1250, - 600     +1250, - 600     + 700, - 550
                    + 650, - 600     +1250, - 600     + 700, - 550     +1300, - 550
                    +1300, - 550     +1300, - 550     + 700, - 550     + 700, - 600
                    +1300, - 550     + 700, - 550     + 650, - 600     +1250, - 600
                    + 700, -12950     +2500, - 550     + 700, - 550     +1300, - 550
                    + 700, - 550     + 700, - 550     +1300, - 550     +1300, - 550
                    + 700, - 550     + 700, - 550     +1300, - 550     + 700, - 550
                    +1300, - 550     +1300, - 550     +1300, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     + 700, -12950     +2500, - 550     + 700, - 550
                    +1300, - 550     + 700, - 550     + 700, - 550     +1300, - 550
                    +1300, - 550     + 700
               unsigned int  rawData[99] = {2500,600, 700,550, 1250,600, 650,600, 650,600, 1250,600, 1250,600, 700,550, 650,600, 1250,600, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,600, 1300,550, 700,550, 650,600, 1250,600, 700,12950, 2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,12950, 2500,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 1300,550, 700};  // SONY 4CB92
               unsigned int  data = 0x4CB92;

               stop

               IR code too long. Edit IRremoteInt.h and increase RAWBUF
               Timing[99]: 
                    +2500, - 550     + 700, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     +1300, - 550     +1300, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     +1300, - 550
                    +1300, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     + 700, - 550     + 700, - 550     +1300, - 550
                    + 700, -12950     +2500, - 550     + 700, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     +1300, - 550     +1300, - 550
                    + 700, - 550     + 700, - 550     +1300, - 550     + 700, - 550
                    +1300, - 550     +1300, - 550     +1300, - 550     + 700, - 550
                    + 700, - 550     +1300, - 550     + 700, - 550     + 700, - 550
                    +1300, - 550     + 700, -12950     +2500, - 550     + 700, - 550
                    + 700, - 550     + 700, - 550     +1300, - 550     +1300, - 550
                    +1300, - 550     + 700
               unsigned int  rawData[99] = {2500,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,12950, 2500,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 1300,550, 1300,550, 1300,550, 700,550, 700,550, 1300,550, 700,550, 700,550, 1300,550, 700,12950, 2500,550, 700,550, 700,550, 700,550, 1300,550, 1300,550, 1300,550, 700};  // SONY 1CB92
               unsigned int  data = 0x1CB92;

```

- i have noted the suggestion of " IR code too long. Edit IRremoteInt.h and increase RAWBUF" but i cant find how to change it in  IRremoteInt.h.

- i have tried to send sendSony(0x1CB92, 99,3) and with 3 repeats as suggested for sony devices , but i have no success that DVD react to my signal.
while same sketch is sending normally a working signal to Samsung TV using sendSamsung(0XCODE,32) no repeats.

please help if there is any thing i can change to make it working.

thanks

crankyoldgit, 2017-04-08:
Interesting. The decode is not matching it as a Sony protocol correctly. That aside, you are most certainly calling the function wrong.

For starters:
`sendSony(0x1CB92, 99,3)`
* The code (0x1CB92) is possibly wrong. (But checking by hand below, seems like they are okay.)
* 99 is certainly wrong. That parameter is the number of bits in the _code_ being sent. typically 12, 15, or 20 for the Sony protocol(s). Looking at the rawdata, I can see that your remote is producing a 20 bit code.
* For the 3, the repeats you typically need is 2. The code needs to be sent 3 times. i.e. code + 2 repeats = 3 codes sent. (make sure you are using the latest version of the library.)

The Sony code actually ends at the `-12950` entry. That's one of the gaps between the three codes.

The line in IRremoteESP8266.cpp: `os_timer_arm(&timer, 15, ONCE);` the _15_ is 15ms (15000us). That's longer than the gap of 12950. I suggest you change it to 10, and try again. That might help it match the sony signal. Let me know how it goes, please.

If you want to decode by hand, here is how:
 +2500, - 550  is the header. i.e. the start of a new code.
Each +1300, - 550 pair is a '1', & each + 700, - 550 pair is a '0'. Or to put it another way, +1300 == 1, +700 = 0
The most significant bit comes first.
Your first 'power' example decodes to:
(binary) 10101000101110010010 == (hex) a8b92 == (dec) 691090

e.g. Try:` sendSony(0xa8b92, 20, 2);  // Sony Power button per your decode`

crankyoldgit, 2017-04-08:
Oh, fyi, RAWBUF is defined in IRremote8266.h, it will be moving to IRremoteInt.h soon.

scropion86, 2017-04-08:
thanks for your reply, but why is the example sketch not showing the received code correctly. and is there any way to get it correctly as calculating and decoding such numbers as of of my Home humble little knowledge☺️?

scropion86, 2017-04-08:
i will try the suggest code? and i will update you asap. 

also if i changes the RAWBUF which values should i change it to? and alos that solving my IR signal problem?

crankyoldgit, 2017-04-09:
I'll be looking into improving the decoding soon. I agree, it should decode it correctly. You're the first person to report an issue with it for the Sony protocol.

I look forward to hear if the changes I suggested yield better results for you.

Technically, you only need a RAWBUF of 2+(20*2) == 46 for a 20bit sony code. As the remote is sending at least three codes at a time, you'd need at least 3 * 46 == 138, so I'd suggest a value of 200 to be safe.That'd cover 4 codes. The maximum is 255 at present.

scropion86, 2017-04-09:
Thanks @crankyoldgit is working fine after increasing `RAWBUF ` to 200 and sending the signal in 20bits with two repeats. 👍

closed 2017-04-09 by scropion86

crankyoldgit, 2017-04-09:
Hey, Can you please give me the new rawData dump? I'll use that to debug why it isn't matching the Sony protocol neatly.