crankyoldgit/IRremoteESP8266 #146: No more connection after a reboot

Timm27, 2017-04-08:
Hello,
I use IRremote ESP8266: IRServer and it works well, thank you very much.
The only problem is that my WeMos-D1-mini with two IR-LEDs and transistor loses after a reboot the WLAN connection and is no longer available.
I tried to enter the IP address permanently:

```
IPAddress ip (172,16,1,104);
IPAddress gateway (172,16,1,1);
IPAddress subnet (255,255,255.0);
```
`WiFi.config (ip, gateway, subnet);`

But that did not help.
what am I doing wrong ?
I would be very grateful for any help!
/Tim

crankyoldgit, 2017-04-08:
I've not tried to use a static network config on an ESP yet, so I can't comment on that.
However, looking at your code snippet, you seem to have '255.0' instead of '255,0'. Maybe thats it?

There should be no difference between boots of the device. Each time it starts a fresh and everything should be the same, unless you have something very odd with your circuit. That leaves external things like: Your wireless network; The power supply; A faulty WeMos.

Does anything come up on the serial monitor on the boot that fails?

Timm27, 2017-04-08:
Thank you for your answer.
with the .255 was unfortunately a little typo, sorry
The problem is solved, I have taken another port for the transmitter and now it works perfectly
Irsend irsend (4);

Thank you
/Tim

closed 2017-04-08 by Timm27