crankyoldgit/IRremoteESP8266 #147: Wrong decode or wrong encode?

takigama, 2017-04-08:
I've recently started working with this library (my own adhoc hardware receiver/sender). For the most part it works, but i've run into an issue with a pioneer amp, and im not sure whether its a case of its not decoding correctly or im not sending correctly.

When the IR receiver reads the code, i get this:
```
NEC,32,a55a58a7
unsigned int  rawData[67] = {8550,4200, 600,1550, 600,500, 600,1600, 600,500, 600,500, 600,1550, 600,500, 600,1550, 600,500, 600,1550, 600,500, 600,1600, 600,1600, 550,500, 600,1550, 600,500, 600,500, 600,1600, 600,500, 600,1600, 600,1550, 600,500, 600,500, 600,500, 600,1550, 600,500, 600,1550, 600,500, 600,500, 600,1550, 600,1550, 600,1600, 600};
```
When i send that back to the amp using sendNEC or sendRaw, nothing interesting happens.

According to the pioneer site it says this:
[image]

Though im not entirely sure what that means, plus im not entirely sure what frequency I should send with sendraw (fairly new to sending/receiving IR with arduino/esp at this point) and would just love some pointers about what may be going wrong here.

crankyoldgit, 2017-04-09:
NEC uses 38kHz. If you have a look at the sendNEC() procedure in [IRremoteESP8266.cpp](https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.cpp#L198), you'll see where it sets it frequency.

Can you post the code (or a link to the code) you are using? and a description/circuit diagram? to help us work out what you are doing exactly.

In short, I'd strongly suggest you use the code & setup at [IRsendDemo.ino](https://github.com/markszabo/IRremoteESP8266/blob/master/examples/IRsendDemo/IRsendDemo.ino)
I just updated it with helpful comments for new folk, so please grab the latest master release.

Based on the decode results, the command you want to use (replace in the example) is:
`sendNEC(0xa55a58a7UL, 32);
`That should transmit the code you received.

takigama, 2017-04-09:
Howdy, yeah the board and code are here:
https://github.com/takigama/HomeAutomationExperiments/tree/master/IRRemote

I did try the send demo ino as well without much success

I also tried with sendGC using what global cache says the remote code is and this code (didnt change the variable name, but its pioneer power toggle in this case):

> function: POWER TOGGLE
> 
> code1: sendir,1:1,1,40000,1,1,360,180,22,67,22,22,22,67,22,22,22,22,22,67,22,22,22,67,22,22,22,67,22,22,22,67,22,67,22,22,22,67,22,22,22,22,22,22,22,67,22,67,22,67,22,22,22,22,22,22,22,67,22,67,22,22,22,22,22,22,22,67,22,67,22,67,22,989
> 

```
unsigned int Samsung_power_toggle[71] = {40000,1,1,360,180,22,67,22,22,22,67,22,22,22,22,22,67,22,22,22,67,22,22,22,67,22,22,22,67,22,67,22,22,22,67,22,22,22,22,22,22,22,67,22,67,22,67,22,22,22,22,22,22,22,67,22,67,22,22,22,22,22,22,22,67,22,67,22,67,22,989};

IRsend irsend(5); //an IR emitter led is connected to GPIO pin 4

void setup()
{
  irsend.begin();
  Serial.begin(115200);
}

void loop() {
  Serial.println("Toggling power");
  irsend.sendGC(Samsung_power_toggle, 71 );
  delay(5000);
}
```

crankyoldgit, 2017-04-09:
(Sorry, my browser or github ate my earlier response it seems)

Hi fellow local. ;-)

Yes, 40kHz is correct according to the GC codes. A passing glance over those values, and they are all pretty much the same, some with a repeated code in them. e.g. the 1753 in the second code is a 'gap' between two codes. (360, 180) being the header etc.

Your code looks good, so I fear the problem is elsewhere.
e.g. If you are using right code for the device. etc.

I've _just_ written up a troubleshooting guide on the wiki, you might want to run through [it](https://github.com/markszabo/IRremoteESP8266/wiki/Troubleshooting-Guide). I fear/suspect you've got a hardware problem somewhere. Unlikely if you're comfortable designing your own board, but hey, something to verify/check. e.g. Is the IR LED actually illuminating etc?

Let me know when you've tried those suggestions. I'm out of ideas (except for those)

takigama, 2017-04-09:
Fellow diver as well!

Fairly comfortable that the hardware should be able to cope, but i've been testing with generic hardware as well. Arduino Uno (using a different receiver with that one, ds188 or something, and the upstream arduino irremote library). Also testing with the node mcu. With those im using the generic 2n2222 tranny for driving the led, With the nodemcu im using the same receiver as i am with my own hardware (tl1838). But on my own board, im using a different tranny, though its rated for 300mhz so should be ok.

The device does actually work with the code base in my repo for several other devices (xbox, two different tv's and an air con) just not the pioneer vsx522/523 amp... I know with the IR receiver im using on my hardware currently, it says it a 38khz receiver (the tl1838) but the one i have on the arduino is supposed to be universal. The IR led doesn't list a frequency at all, but i was under the impression they weren't specific to one frequency in any case?

crankyoldgit, 2017-04-10:
Correct. Pretty much all IR LEDs should work with whatever pulse frequency you send it. The physical wavelength is fixed at 940nm.

If you have your (custom) hardware working with other devices, then it's probably not the hardware.
Can you control the other devices with that setup/module/example code? i.e. to isolate we are just dealing with an issue with the pioneer code/amp/etc and/or the IR code word you are trying to send.

Are you able to get the pioneer device to respond with IRremote etc on your Uno? (That would at least confirm if issue is the IR code value or not, and eliminate any library difference or hardware difference)

Have you tried using the IR receiver on you Uno to see what your custom board is sending? i.e. compare the results to what you sent. If so, what were the results.

crankyoldgit, 2017-04-18:
Friendly ping. How are things going?

crankyoldgit, 2017-05-20:
Friendly ping.

takigama, 2017-05-20:
Howdy :)

Sorry, i havent had a chance to work on my IR project due to work, but will be getting back to it this weekend and this week, so expect something soon... I did spend a bit of time trying to figure out exactly where it should go cause its a bit of a special case.

crankyoldgit, 2017-06-08:
Friendly ping.

crankyoldgit, 2017-07-14:
Marking closed due to inactivity. Feel free to reopen if needed.

closed 2017-07-14 by crankyoldgit