crankyoldgit/IRremoteESP8266 #152: home/commands/IR_LG not working, NodeMCU dies

fbsdmon, 2017-04-12:
Hi,

The IR_LG command is not working, it completely blocks my NodeMCU v1.0.

Serial output from LG TV Power button
```
Hey I got a callback 
Receiving data by MQTT
home/commands/IR_LG
Callback value
551489775
Converted value to unsigned long
551489775
```
If I try the same command with IR_Sony or IR_NEC I get `Acknowedgement of reception published` at the end and the OpenMQTTGateway continues to work. But with IR_LG I don't get the acknowledgement and the mqtt gateway no longer responds.

I have the latest Arduino IDE, latest code, latest everything.

fbsdmon, 2017-04-12:
Here is what I get on `mosquitto_sub -t home/# -v` when I press the power on my remote
```
Client mosqsub/9657-ubox received PUBLISH (d0, q0, r0, m0, 'home/IRtoMQTTAdvanced', ... (33 bytes))
home/IRtoMQTTAdvanced Value 551489775 Bit 32 Protocol 3
Client mosqsub/9657-ubox received PUBLISH (d0, q0, r0, m0, 'home/sensors/ir', ... (9 bytes))
home/sensors/ir 551489775
```

crankyoldgit, 2017-04-12:
Technically, your NodeMCU wasn't crashing, it was just delay()ing for a very long time. It would have responded again after about ~70 mins.
It should be fixed now. Thanks for the report.

commit 6d37ad45 (2017-04-12): Incorrect use of LG_RPT_LENGTH

LG_RPT_LENGTH - usecs.elapsed() was incorrect. LG_RPT_LENGTH was meant to be the gap between repeats, not the gap + length of the command.

Fixes #152
closed 2017-04-12 by crankyoldgit