crankyoldgit/IRremoteESP8266 #166: Template for adding new Manufacturers

rahuljawale, 2017-04-17:
Is it possible to create a template for adding new Manufacturers? I, for instance, would like to add Hitachi AC codes that I have been working on. But call me busy or call me lazy, I have been postponing the exercise. 

I believe by using a template, contributors , existing and new alike, could speed up the process of developing/supporting new brands.

crankyoldgit, 2017-04-17:
When I created the Kelvinator ones, I copied the Daikin code and used that as my template.
Before you stare into the abyss that is A/C remote codes, I'd strongly recommend trying to find prior work or examples, for down that path lies madness. I swear [Cthulhu](https://en.wikipedia.org/wiki/Cthulhu) or even [Hastur](https://en.wikipedia.org/wiki/Hastur) are involved with the engineering of A/C remotes.

It's a much simpler task to just capture and replay a few commands, then to actually reverse engineer how a _code_ is generated. e.g. I lost a lot of sleep and hair trying to working the checksum alg. for Kelvinator. I had to capture close to 100 different codes to work it out fully. Even then I didn't work out how the timer bits worked. Much voodoo there.

To save your sanity, try starting [here](https://perhof.wordpress.com/2015/03/29/reverse-engineering-hitachi-air-conditioner-infrared-remote-commands/), [here](https://github.com/jackbell16/AC-Progetto-Arduino), and [here](https://github.com/ToniA/arduino-heatpumpir). The latter probably being the best place to start.

crankyoldgit, 2017-07-02:
With the code split up from a monolithic file, to a file per protocol. I think this is now a lot easier/more obvious. Do you think we still need a specific example/template file(s)?

crankyoldgit, 2017-07-14:
Marking closed due to inactivity, and I think it's kind-of been addressed now.

closed 2017-07-14 by crankyoldgit

darshkpatel, 2017-07-14:
@rahuljawale plus one on adding Hitachi remote codes